Romkabouter/ESP32-Rhasspy-Satellite #3: Question on the code

Lumy88, 2019-01-14:
Hello Rom,
Thanks for sharing such an amazing work. I'm new in programming, and have a silly question on your code. ( line 331-334&337 of MatrixVoiceAudioServer.ino)

```
for (int i=0;i<message_count;i++) {
        for (uint32_t s = CHUNK*i; s < CHUNK * (i+1); s++) {
            voicebuffer[s-(CHUNK*i)] = mics.Beam(s);
        }      
```

As far as I understanding, voicebuffer only has 256 elements, and the first voicebuffer ( when i =0 ) will be overlapped after running this part of code.

` memcpy(voicemapped,voicebuffer,CHUNK*WIDTH)`

Thus how to copy the voicebuffer to voicemapped from 0-511?

Thank you!

Romkabouter, 2019-01-14:
voicebuffer has 256 16bit elements, so 512 bytes.
voicemapped is uint_16, while voicebuffer is uint_8

For correct audio, you need to keep the bytes in synch. So 16bit aligned.
Each 16bit sample needs to be copied to 2bytes, that is exactly what the memcpy does.

Lumy88, 2019-01-23:
Dear Rom,
Thank you very much. Finally, I totally understand your code.
Then I can go on to the next step!

Great Work!

closed 2019-01-23 by Lumy88