Romkabouter/ESP32-Rhasspy-Satellite #6: wishbone_bus.h: No sush File or directory

LeTengu, 2019-05-17:
Hello Romkabouter, congratulations for your work.
I'm a newbie, first time I use IDE Arduino…
I do step by step the Readme guide.
When I compile, I obtain this error:

Arduino : 1.8.9 (Windows Store 1.8.21.0) (Windows 10), Carte : "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 115200, None"

MatrixVoiceAudioServer:49:26: error: wishbone_bus.h: No such file or directory

Plusieurs bibliothèque trouvées pour "WiFi.h"
compilation terminated.

Utilisé : C:\Users\leten\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.2\libraries\WiFi
Non utilisé : C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\libraries\WiFi
exit status 1
wishbone_bus.h: No such file or directory

Ce rapport pourrait être plus détaillé avec
l'option "Afficher les résultats détaillés de la compilation"
activée dans Fichier -> Préférences.
"
Please, could you help me to find an issue ?
Thank you in advance.

Romkabouter, 2019-05-17:
Sound like you did not copy the hal folder to your Arduino Library folder?

LeTengu, 2019-05-18:
You’re absolutely right, I’m sorry...and I’ve done it several times.
I would still need your help, now when I compile I get the error:

Arduino : 1.8.9 (Windows Store 1.8.21.0) (Windows 10), Carte : "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 115200, None"

C:\Users\leten\Documents\Arduino\libraries\hal\microphone_array.cpp: In member function 'void matrix_hal::MicrophoneArray::CalculateDelays(float, float, float, float)':

C:\Users\leten\Documents\Arduino\libraries\hal\microphone_array.cpp:138:17: error: 'round' is not a member of 'std'

     int delay = std::round((it->first - min_distance) * sampling_frequency_ /

                 ^

C:\Users\leten\Documents\Arduino\libraries\hal\microphone_array.cpp:138:17: note: suggested alternative:

In file included from c:\users\leten\documents\arduinodata\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\cmath:44:0,

                 from C:\Users\leten\Documents\Arduino\libraries\hal\microphone_array.cpp:21:

C:\Users\leten\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.2/tools/sdk/include/newlib/math.h:278:15: note:   'round'

 extern double round _PARAMS((double));

               ^

Plusieurs bibliothèque trouvées pour "WiFi.h"
Utilisé : C:\Users\leten\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.2\libraries\WiFi
Non utilisé : C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\libraries\WiFi
exit status 1
Erreur de compilation pour la carte ESP32 Dev Module

Ce rapport pourrait être plus détaillé avec
l'option "Afficher les résultats détaillés de la compilation"
activée dans Fichier -> Préférences.

Could you help me ?
Thank you for your patience.
Tengu.

Romkabouter, 2019-05-18:
I do not have a windows machine, so I'm sorry I cannot help you with that.
Also, the error is in microphone_array.cpp, which is software from Matrix company. Maybe some help there?

LeTengu, 2019-05-18:
Thank you very much, I managed to compile by modifying the file "microphone_array.cpp" as follows:

(a) change line 138:
int delay = round(((it->first - min_distance) * sampling_frequency_ /

(b) add to line 80:
int round(float a) {
return (int)(std::ceil(a-0.5) == std::ceil(a)? std:ceil(a): std:floor(a));
}

I hadn’t written a C++ in years, but it works. If it helps someone....

I would still need your help for the rest (I never used Arduino).
You write the following:
“in the Deploy.sh file, change to IP address to the IP your Pi is on.”
I know it’s stupid, but I can’t find this file…: -/

And then:
Open a terminal (Linux and Mac should work, I use Mac), change to the folder where this code is and run: sh Deploy.sh.
Being on Windows, do I have to do it on Raspberry?
Second, I don’t understand how to flash to Matrix Voice the first time without connecting it to the USB PC…

I realize that these are novice questions, but you would be very kind to accompany me.

Thank you very much.

Romkabouter, 2019-05-19:
First flash should indeed be with the Voice attached to a raspberry pi.
I see I have renamed deploy.sh to deploy_single and deploy_ota. Best to use deploy_ota
You do not have to attach the Voice to the PC, but to a raspberry.
Then execute the script from your PC (via commandprompt), after you change the IP. You might have to change more since you are on windows. I do not think is will run with sh as extension, but might run if you change it to .bat.
This I did not test at all, tar is possibly not available on windows.

I suggest creating a virtual machine with linux for this purpose

closed 2019-05-19 by Romkabouter