Romkabouter/ESP32-Rhasspy-Satellite #8: No response from the ESP

waldhari81, 2019-08-04:
Hi,
I followed your tutorial to the letter, I think.
The ESP flashed and all looks good.  However the last part doesn't work for me:
"Make a change (or not) and do a Sketch -> Upload. The leds will turn WHITE
Sometimes uploading fails, just retry until it succeeds."
I thought at first it would be the IP address, but that checked out ok, I believe.
I tried a dozen times, but keep getting the "No response from the ESP" error from Arduino, after I enter the correct password. Also started over numerous times.
The leds do look white though and the device connected to the wifi on its own just fine. (I pinged it and checked the router)
Can someone help me with this issue?

Romkabouter, 2019-08-05:
Hmm, that is strange. Did you already try unplug the Voice and retry?
In most cases, that worked for me when I had problems.

waldhari81, 2019-08-11:
Same thing, at first "authenticating... FAIL" ; then  "no answer to our authentication", after yet another retry: "No response from the ESP". Even connecting Rhasspy via Hassio to the ESP doesnt work.
I will retry the entire process compiling & flashing again tomorrow, I must be doing something wrong. Thx anyway

Romkabouter, 2019-08-11:
ok, I could not reproduce it thisfar. You can always reflash the same way as the first time :)

waldhari81, 2019-08-11:
I found the error.  I got a Pubsubclient error during the compilation ('PubSubClient' does not name a type), I thought I fixed this with updating the Pubsub lib in Arduino, but it must have overwritten MQTT_MAX_PACKET_SIZE in PubSubClient.h back to its standard and thus the compilation succeeded with wrong values. I'm now looking for a way to get rid of the error and compile it with the correct packet size (2000).

waldhari81, 2019-08-12:
I have the same problem with other programs too, it has nothing to do with Matrix-Voice-ESP32-MQTT-Audio-Streamer. I will close this. Thx anyway

closed 2019-08-12 by waldhari81