Romkabouter/ESP32-Rhasspy-Satellite #9: No audio / noise output on MATRIX ESP32 (input detection and bing sound works)

df1gr, 2019-08-16:
Hi all,

I have flashed my Matrix Voice ES32 with the newest version of the audio streamer. The MQTT based connection with snips works well, also the hotrod detection and the intent detection. 

Unfortunately, I cannot hear any reply (e. g. current time), except some noise.

The snips server is installed with default settings in an Ubuntu VM running on a FreeNAS.

Changing the sample rate and channels (e. g. to "32000" and "2" or "44100" and "2") did not result in working speech output.

Do you have any ideas what I can do to get the audio output via Matrix working?

Many thanks,
  Roland

Romkabouter, 2019-08-16:
How did you change the sample rate? I think the audio sounds coming form snips are fixed samplerate/channels. I do not know this for sure.
Also, there are different formats coming from snips, I believe there is 1 44100 wave sound. This is when snips does not understand your command.

Maybe ask around on the Snips community on if and how the audio output can be configured?

df1gr, 2019-08-16:
We changed the sample rate in your program.
And we found in MQTT output that different sample rates are coming from snips e.g. 22050 for beeps and other output with 44100 sample rate. 
Wonder, if audio sound  produced correctly output with your MATRIX ESP32 equipment? What is your experience regarding voice output? 
Do you know which sample rate MATRIX ESP32 hardware is able to convert?

Thank you,

Roland

Romkabouter, 2019-08-16:
The Matrix Voice will only play 44100 / 2 correctly. This is a hardware/firmware issue.
Anything else should be converted on the fly, there is functionality to change mono to stereo but that is about it. Mono 16000 could also work, see the latest PR from lonely-skywalker.

I currently use Rhasspy with the voice and configured that with GoogleWaveNet set to 44100. My code converts that to stereo and I can hear the audio fine.

The hardware of the ESP32 can convert any samplerate, but you have to write code for it.
It also should be fast enough the do it on the fly (with the ringbuffer), or there will be way to much delay between the audio coming in from MQTT and the audio output.
If you can write code to convert any rate / channel on the fly to 44100/2 that would be awesome, but I could not get that working sadly to this point.

df1gr, 2019-08-17:
Thanks a lot for the information - this could help me further.
Kind Regards,
Roland 

---

closed 2019-08-17 by Romkabouter

lonely-skywalker, 2019-08-17:
If you need to convert from a lower sample rate to a higher one, eg from 22050 to 44100, consider the max buffer the matrixio API accepts. If I remember right it was 1024 bytes.
Exceeding this limit leads to crashes and funny noise :)