Romkabouter/ESP32-Rhasspy-Satellite #13: Matrix Keeps rebooting

thomasfals, 2019-12-19:
I have flashed the Matrix Voice. Only parameters changed in platformio.ini are:

`DWIFI_SSID`
`DWIFI_PASS`
`DMQTT_IP`
`DMQTT_HOST`

Build was succesfull and so was the deploy. After powering up the Voice standalone nothing happens apart from blue led underneath. The voice does not connect to the wifi so I plugged it back into the Pi and ran:

`sudo minicom -D /dev/ttyS0`

With the result repeating:

```
I (12) boot: ESP-IDF v4.0-dev-459-gba1ff1692 2nd stage bootloader               
I (12) boot: compile time 11:07:17                                              
I (12) boot: Enabling RNG early entropy source...                               
I (17) boot: SPI Speed      : 40MHz                                             
I (21) boot: SPI Mode       : DIO                                               
I (25) boot: SPI Flash Size : 4MB                                               
I (29) boot: Partition Table:                                                   
I (33) boot: ## Label            Usage          Type ST Offset   Length         
I (40) boot:  0 nvs              WiFi data        01 02 00009000 00004000       
I (47) boot:  1 otadata          OTA data         01 00 0000d000 00002000       
I (55) boot:  2 phy_init         RF data          01 01 0000f000 00001000       
I (62) boot:  3 factory          factory app      00 00 00010000 00100000       
I (70) boot:  4 ota_0            OTA app          00 10 00110000 00100000       
I (77) boot:  5 ota_1            OTA app          00 11 00210000 00100000       
I (85) boot: End of partition table                                             
I (89) boot: Defaulting to factory image                                        
I (94) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x3449c (21p
I (178) esp_image: segment 1: paddr=0x000444c4 vaddr=0x3ffbdb60 size=0x035f8 ( d
I (183) esp_image: segment 2: paddr=0x00047ac4 vaddr=0x40080000 size=0x00400 ( d
I (185) esp_image: segment 3: paddr=0x00047ecc vaddr=0x40080400 size=0x08144 ( d
I (207) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0xa8214 (6p
I (448) esp_image: segment 5: paddr=0x000f8234 vaddr=0x40088544 size=0x0bef0 ( d
I (481) boot: Loaded app from partition at offset 0x10000                       
I (481) boot: Disabling RNG early entropy source...                             
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandl.
Core 0 register dump:                                                           
PC      : 0x401750af  PS      : 0x00060930  A0      : 0x80139eab  A1      : 0x3 
A2      : 0x00000000  A3      : 0x000000fe  A4      : 0x00000001  A5      : 0x0 
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x801399d1  A9      : 0x3 
A10     : 0x00000020  A11     : 0x4013958c  A12     : 0x3ffe3a78  A13     : 0x0 
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x0 
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x0 
                                                                                
Backtrace: 0x401750af:0x3ffe3a90 0x40139ea8:0x3ffe3ab0 0x400d492d:0x3ffe3ae0 0x0
                                                                                
Rebooting... 
```
Disclaimer I'm a bit of a novice in the this area and I have not used the Voice for anything else yet apart from running everloop demo.

The Pi is running a fresh install of Raspbian Stretch followed by setup steps from here: [https://www.hackster.io/matrix-labs/program-matrix-voice-esp32-with-vs-code-using-platformio-3dd498](https://www.hackster.io/matrix-labs/program-matrix-voice-esp32-with-vs-code-using-platformio-3dd498)

thomasfals, 2019-12-19:
Followed answer from @Romkabouter here [https://community.matrix.one/t/matrix-voice-esp32-constant-rebooting/2752/7](https://community.matrix.one/t/matrix-voice-esp32-constant-rebooting/2752/7) and now I see the Matrix on the network.

closed 2019-12-19 by thomasfals

Romkabouter, 2019-12-19:
Thanks, I was just typing to verify the versionnumber 👍