Romkabouter/ESP32-Rhasspy-Satellite #14: Unable to install

thomasfals, 2020-01-04:
I can build the latest version but the folder structure seems to have changed quite a bit. There is no lon ger an ota folder and no ./install.sh ?

Basically I got no clue on how to install the new build.

Romkabouter, 2020-01-05:
Folder structure is the same as previous release (since I added plaformIO)
Did you do a git clone?

thomasfals, 2020-01-05:
> Folder structure is the same as previous release (since I added plaformIO)
> Did you do a git clone?
> Also, when I download the zip source file, everything is there.
> In any case, if you can build the latest version using plaformIO, you can update via OTA (use the small upload button)

Hi. Yes I did a clone git. However, now I have started over completely and reset the voice and set it up for OTA.

When I try to upload OTA with the latest build I get:

`Building in release mode
Retrieving maximum program size .pio/build/esp32dev/firmware.elf
Checking size .pio/build/esp32dev/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  19.5% (used 63856 bytes from 327680 bytes)
Flash: [======    ]  55.6% (used 1093542 bytes from 1966080 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = espota
Uploading .pio/build/esp32dev/firmware.bin
20:51:01 [DEBUG]: Options: {'esp_ip': '192.168.1.93', 'host_ip': '0.0.0.0', 'esp_port': 3232, 'host_port': 57000, 'auth': 'voice', 'image': '.pio/build/esp32dev/firmware.bin', 'spiffs': False, 'debug': True, 'progress': True, 'timeout': 10}
20:51:01 [INFO]: Starting on 0.0.0.0:57000
20:51:01 [INFO]: Upload size: 1093664
Sending invitation to 192.168.1.93 
Authenticating...OK
20:51:01 [INFO]: Waiting for device...
20:51:11 [ERROR]: No response from device
*** [upload] Error 1
`
Not sure where I can get additional info for bug tracking.

Romkabouter, 2020-01-09:
Ok, I get that same error ever so often but with the old version as well.
May I ask how you started completely over? If it was using the latest release, you already running it ;)

If not, maybe try to unplug the power and plug it back in. I will see if I get the upload error more often now.
The new release has some extra code in it, but the ondevice hotword detection is off by default

thomasfals, 2020-01-10:
> Ok, I get that same error ever so often but with the old version as well.
> May I ask how you started completely over? If it was using the latest release, you already running it ;)
>
> If not, maybe try [...]

I did a complete reset of the Matrix and set it up for OTA. Then I made a git clone of the project, configured project file, made a build and tried to upload.

Romkabouter, 2020-01-10:
Yes, but did you use my repo for complete reset and OTA setup or the guide from hackster?
That guide from Hackster was based on my code, but different settings for the partitions.

I think it is best to use this repo to reset the Matrix and setup for OTA. That way you can still run the latest release.
I will see if reverting my latest changes has had an effect on the OTA performance

thomasfals, 2020-01-13:
> Yes, but did you use my repo for complete reset and OTA setup or the guide from hackster?
> That guide from Hackster was based on my code, but different settings for the partitions.
>
> I think it is [...]

I'm not sure how I go about doing a reset from this repro. Perhaps an updated guide could be useful?

Romkabouter, 2020-01-13:
Apparently the Getting Started part is not clear enough.
In the 5th bulllit, you can use this repo instead of the Hackster guide. Folder is not esp32-platformio as in the guide, but just PlatformIO.
I'll update the readme to make that more clear.

Romkabouter, 2020-01-14:
I have updated the readme to make some steps more clear. Please follow the get started.
Please try that if the OTA continues to fails. I get the device error once in a while as well, but I do not know why. That is why it is mentioned in the Know Issues at the bottom of the readme.

closed 2020-01-14 by Romkabouter