Romkabouter/ESP32-Rhasspy-Satellite #21: WebSockets support

sskorol, 2020-03-02:
Hi @Romkabouter,

Already asked it at a Matrix forum. Just curios if you had a chance to take a look at WebSockets support?

I have a https://github.com/alphacep/vosk-api installed on RPi. And there's a simple WS server which accepts an audio stream and returns JSON transcribe: https://github.com/alphacep/vosk-server/blob/master/websocket/asr_server.py

From a laptop I'm using a nodejs WS client which streams chunks captured from the integrated microphone via rec tool:
```
const RECORDER_OPTIONS = {
  program: 'rec',
  bits: 16,
  channels: 1,
  encoding: 'signed-integer',
  rate: 8000,
  type: 'wav',
  silence: 0
}
```
I also found some EPS32 WS samples like this one: https://github.com/krdlab/esp32-websocket/blob/master/examples/Client/Client.ino

As far as I can see from your code, the main audio processing happens in
`void Audiostream(void *p)`, where you send chunks via
`audioServer.publish(audioFrameTopic.c_str(), (uint8_t *)payload, sizeof(payload));`

Can you point me to where I can adjust audio settings to meet the above requirements (16b/8k/1ch/wav) before actual sending? I want to replace mqtt call with WS to test it with Kaldi.

Romkabouter, 2020-03-02:
Thanks for the pointers, I will check it very soon indeed!
There is a call in the code: mics.SetSamplingRate(RATE)

This sets the sampling rate, but I have never used any other rate then 16000.
For the rest the settings are ok (16bit, mono);

The chucks are send in two messages, because 1 mics.Read() will read 512 samples, which is in turn read in the loop mics.Beam(s) (around line 650..670)
For the websocket, that loop is not necessary anymore, because it will be a continues stream.

sskorol, 2020-03-03:
@Romkabouter there're [two loops](https://github.com/Romkabouter/Matrix-Voice-ESP32-MQTT-Audio-Streamer/blob/master/PlatformIO/src/MatrixVoiceAudioServer.cpp#L648-L651) around `misc.Beam`. Did I get it correctly: we don't need both, right? Or we still need to split it into 2 messages? Seems like it's required for Snips only?

Romkabouter, 2020-03-03:
Yes, there are two loops. The top loop is for the messages, that is no longer needed.
The innerloop fills the voicebuffer with the mics.Beam(s), that is still needed.

The default settings for Snips was 256 buffers per frame, so that is why there are two messages needed for sending the first 256 from mics.Beam and then the second set of 256.
If you would have changed the framesize in snips to 512, you could change it in The streamer as well to get 512 buffers per audioframe (=wavefile send to the MQTT broker)

sskorol, 2020-03-03:
But in case of WebSockets, I will send a stream directly to Kaldi server. I don't believe Snips is required at all. So I guess the code could be simplified to the following. Correct?
```cpp
            if (xEventGroupGetBits(audioGroup) != PLAY) {
                mics.Read();

                // Sound buffers
                uint16_t voicebuffer[CHUNK];
                uint8_t payload[sizeof(header) + CHUNK];

                for (uint32_t i = 0; i < CHUNK; i++) {
                    voicebuffer[i] = mics.Beam(i);
                }
                memcpy(payload, &header, sizeof(header));
                memcpy(&payload[sizeof(header)], voicebuffer, sizeof(voicebuffer));
                // send payload
            }
```

Romkabouter, 2020-03-03:
yes, I think that is correct

sskorol, 2020-03-05:
Unfortunately, I couldn't make native [espidf sockets](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/protocols/esp_websocket_client.html) work with your code. Maybe you will have more luck. What I did manage to do is mqtt streaming to my server, and redirecting chunks to web sockets server. But I don't really like this approach.

In general, I did the following updates:
```cpp
            if (xEventGroupGetBits(audioGroup) != PLAY) {
                mics.Read();

                if (!hotword_detected && localHotwordDetection) {
                    int16_t voicebuffer_wk[CHUNK];
                    for (uint32_t i = 0; i < CHUNK; i++) {
                        voicebuffer_wk[i] = mics.Beam(i);
                    }

                    int result = wakenet->detect(model_data, voicebuffer_wk);
                    if (result > 0) {
                        xEventGroupSetBits(everloopGroup, EVERLOOP);
                        detectMsg =  std::string("{\"siteId\":\"") + SITEID + std::string("\", \"modelId\":\"") + MODELID + std::string("\"}");
                        asyncClient.publish(hotwordDetectedTopic.c_str(), 0, false, detectMsg.c_str());
                        hotword_detected = true;
                        publishDebug("Hotword Detected");
                    }
                    //simulate message for leds
                    streamMessageCount++;
                }

                if (hotword_detected || !localHotwordDetection) {
                    uint16_t voicebuffer[CHUNK];
                    uint8_t payload[CHUNK];

                    for (uint32_t i = 0; i < CHUNK; i++) {
                        voicebuffer[i] = mics.Beam(i);
                    }

                    // memcpy(payload, &header, sizeof(header));
                    memcpy(payload, voicebuffer, sizeof(voicebuffer));
                    audioServer.publish(audioFrameTopic.c_str(),(uint8_t *)payload, sizeof(payload));

                    //simulate message for leds
                    streamMessageCount++;
                }
            }
```
I also increased chunks size:
```cpp
int CHUNK = 1024;
```
And decreased the amount of messages:
```cpp
if (streamMessageCount < 50) {
```
With such a huge buffer size, I had to increase MQTT packets as well:
```
'-DMQTT_MAX_PACKET_SIZE=6000'
```
These modifications perfectly work with Kaldi server with 1 minor change. I increased rate to 16k as well.

Anyway, that's just a temporary solution. If you find a way to send sockets stream directly, would be greatly appreciated!

sskorol, 2020-05-23:
Hi @Romkabouter, did you have a chance to take a look at it? :)

Romkabouter, 2020-05-24:
Yes, I actually have and it is WIP.
Does not process very quickly however, sorry about that.

sskorol, 2020-05-25:
No worries, thanks for your time and effort!

Romkabouter, 2020-07-14:
Hi sskorol, I have not yet had the time to make further code but this support is on my goals :)

Romkabouter, 2020-07-30:
I have created a branch to start working on this:
https://github.com/Romkabouter/Matrix-Voice-ESP32-MQTT-Audio-Streamer/tree/websockets

closed 2020-07-30 by Romkabouter

Romkabouter, 2020-07-31:
Apparently, there is also a MQTT server:
https://github.com/alphacep/vosk-server/tree/master/mqtt

sskorol, 2020-07-31:
It was added by me, as there was no WS alternative. However, I believe that for voice streaming mqtt is not the best option.