Romkabouter/ESP32-Rhasspy-Satellite #22: Stuck trying to do initial deploy

carlos-sarmiento, 2020-05-23:
I followed the instructions on the README but I'm stuck trying to run `deploy.sh` but getting an error. I've checked the steps and I don't know what I'm doing wrong.

When I try to run `deploy.sh` this is the output
```
$ ./deploy.sh
cp: cannot stat '../OTABuilder/build/bootloader/bootloader.bin': No such file or directory
Loading firmware: .pio/build/esp32dev/firmware.bin

-----------------------------------
esptool.py wrapper for MATRIX Voice
-----------------------------------
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt]
                           [--ignore-flash-encryption-efuse-setting]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: '/tmp/bootloader.bin'
done

[SUCCESS] Please disconnect your MatrixVoice from the RaspberryPi and reconnect it alone for future OTA updates.
```

Also tried to to run make on the OTABuilder folder, but it also failed:

```
$ make
Makefile:6: /make/project.mk: No such file or directory
make: *** No rule to make target '/make/project.mk'.  Stop.
```

What could I be doing wrong?

Romkabouter, 2020-05-23:
What is your $(IDF_PATH) ?

carlos-sarmiento, 2020-05-24:
It's not set. Don't know how to et it

```
$ echo $IDF_PATH
$
```

carlos-sarmiento, 2020-05-24:
Checked on Platformio and saw the Espressif 32 framework is installed (seemed like correctly) on VS Code

Romkabouter, 2020-05-24:
I see, for the OTAbuilder to actually build you need to install the esp32 framework:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/#installation-step-by-step

If you follow that guide, you will setup your IDF_PATH. 
The OTABuilder is not platformIO project, I will update some documentation on that.
Let me know if it works out

closed 2020-07-14 by Romkabouter

carlos-sarmiento, 2020-07-15:
Forgot to post here, but that solved the problem