Romkabouter/ESP32-Rhasspy-Satellite #23: Would it work on an ESP32-Audio-Kit?

ei23-fxg, 2020-07-14:
Hi.
As you may know, there are cheap standalone ESP32 Audio Kits out there.
They have some buttons, microphones and speaker output etc.
Would an adapted MQTT-Streamer for Rhasspy also work on them?

Romkabouter, 2020-07-14:
I currently do not know of those kits, but if they are standard ESP32 I would assume that this is possible

ei23-fxg, 2020-07-14:
I have one laying around for testing. They look like this:
[image]

https://de.aliexpress.com/item/4001097459996.html

Romkabouter, 2020-07-14:
Well, in the  Audiostream, the way the samples are read have to be changed.
If you fork my repo and work things out, I will be happy to merge a PR, but it should be a PR which still supports the Matrix Voice obviously ;)

Romkabouter, 2020-07-30:
I currently have no plans for buying this board, also due to other projects.
I will keep it in mind though, so if I can get my hands on one I will certainly see if it can be supported.

sehraf, 2020-11-10:
Hey there,
I'm currently working on porting the code to the audio kit. Since the calls to the hardware specific parts (eg matrix hal) are rather sparse, there are only a few lines that need to be replaced.

My biggest problem at the moment is the lack of "drop in" libraries. For the audio playback the AC101 and ESP8266audio library do work, though the later would only require to set up I2S and I think I can do that on my own since this project handles the WAV header already. 
The recording part will be "fun" though. I haven't seen any example for the AC101 chip yet. Though it should be straight I2S, right?

Don't expect any pr soon, I'd just like to let you know that I am on it in the case that somebody else is doing the same.

Romkabouter, 2020-11-10:
Nice, I have recently ordered the M5 Atom echo.
When it arrives, I will start with the code working on there as well.
I have also some plans on rewriting the code, because it is now a bit messy. 
The statemachine you can find in this video might be a good approach: https://www.youtube.com/watch?v=re-dSV_a0tM&ab_channel=atomic14

I think the recording can be I2S indeed, as long as you can find a way to publish the bytes as small wav files, it will be ok 👍 
Good luck and you can always PM me.

sehraf, 2020-12-18:
Quick update:
I've audio playback (tested with rhasspy) running fine half of the time. Often the begin is distorted/messed up. 
I2S is quirkier than expected, to get it running I've basically ESP8266-Audio ....

I'm now struggling with the mics / the AC101 configuration. I've found [this](https://github.com/respeaker/seeed-voicecard) but at the moment i'm just reading `...FF7FFF7FFF7FFF7FFF7F...`. Regarding the WAV part of the recording: I'm planing to (again :wink: ) copy ESP8266-Audio.

I find your code structure ok, every functionality has its own function. Maybe add some abstraction layer in the future. e.g. the whole speex stuff is (currently) not needed for the audio kit.

Romkabouter, 2020-12-18:
Please check my rewrite branch, I hope you'll like it.

It moves almost all code to a state machine. It is now written (and working) for the M5Stack Atom Echo. You can find the device here:

https://m5stack.com/collections/m5-atom/products/atom-echo-smart-speaker-dev-kit

It is still WIP, but I made the ringbuffer much bigger. I still get distorted messages, but that is mostly on higher sample rates (44100 16bit stereo). I am pulling my hair as to why, I am 100% sure the buffer does not get overwritten.
I have debugged a lot of the incoming audio message but I cannot seem to find the issue.

sehraf, 2020-12-18:
> Please check my rewrite branch, I hope you'll like it.

**NO! I HATE IT!**
I've simply changed the i2s pins, added a few lines for the AC101 and guess what?! "_It just works_" (Todd Howard)
Seriously it took me ages to copy and paste my previous solution together and now audio playback works out of the box. 

(The **bold** part was obvious irony :wink: )

sehraf, 2020-12-18:
Regarding the sound distortion (i have them too), I've added the size of the buffered data to the serial (at the beginning and the end, not in the middle) and it looks like it gets completely filled up
```
1311
1312
Enter PlayAudio
59738
59739
59740
```
Does MQTT buffer the received message? I mean if it would discard some, there should be a (noticeable) jump in the audio output but no distortion, right?

Romkabouter, 2020-12-19:
It streams audio to your MQTT broker, the same as the master branch, so it can act as remote microphone for Rhasspy )or snips, but I do not use that anymore)

Indeed, the buffer gets filled and when I send an audio message which fits in the (large) buffer completely there is no distortion.
The problem starts when the message is too big.

The player just pops form the audioData buffer,  and add zero's when the buffer is empty.
So indeed, I would assume there was some stuttering when the buffer gets an underrun, not distortion.

The async MQTT client pushes the bytes whenever they arrive and there is room.  It get around 1450 bytes per partial message. You can see that when you put in a Serial with len in the partial message block.
The I2S task is running on core 1 and the MQTT task (used in the async code) has been set to core 0.

The onMessage parts does a while audioData.isFull, so it blocks the core 0 thread as long as there is no room in the buffer.
The I2S task pops that buffer on another core, therefore they are not blocking each other and I am sure the whole incoming audio is going through.

In the I2S task it tries to play 256 bytes, but when the audioData buffer is empty (aka pop fails), it buts 0 in the buffer which to my believe should result in the typical audio gaps you hear when there is a buffer underrun.

I cannot figure this one out yet, but great news it works for you out of the box!
I will add some settings to be able to support multiple hardware, so it should be easy then to add this device as well.
Maybe I'll buy one when the M5Stack and MatrixVoice are supported. I might do some kind of easy switching of devices but that is something I have to think about :)

sehraf, 2020-12-19:
> It streams audio to your MQTT broker, the same as the master branch, so it can act as remote microphone for Rhasspy )or snips, but I do not use that anymore)

Do you know an easy way to listed to what the ESP streams? My rhasspy did not recognize anything and i assume that the AC101 is not configured properly. The data at least looks like it is changing. Yesterday I've tried to copy some WAV data from the MQTT messages and stitch them together but ffmpeg said that it cannot find any valid data... (could have been a copy and paste problem though)

> so it blocks the core 0 thread as long as there is no room in the buffer.

How does the MQTT lib handle this situation? Have you tried to simply drop a frame when it does not fit into the buffer?

 > I cannot figure this one out yet, but great news it works for you out of the box!

You have probably saved me some headaches! I'll see if i can find any reason for the distortion.

Romkabouter, 2020-12-19:
I do not have a way to listed yet, but it is possible and probably a good idea to have. The stream consist of a huge amount of small wave messages (256bytes + 44 byte header) I had a small python script saving a couple of seconds to disk but I do know if I still have it. If I can find it, I will add it to the repo as well.

Yes, the MQTT lib handles this. When you do not set CONFIG_ASYNC_TCP_RUNNING_CORE it takes a random core.
This can lead to a blocking situation (I encountered it so that is why I found it)

You can skip the pushing of the data by removing the while and just push. It will then silently fail.
Maybe a good idead to give that a try too, just the check results.

sehraf, 2020-12-19:
Quick update: i was actually able to record some sound, so the mic patch should (rudimentary) work.

sehraf, 2020-12-23:
using this small python script, i was able to "listen" to what the ESP sends, the mics work in general, but the audio is roughly 2x the expected speed. Using 0.5 playback speed sounds normal. 

Any Idea what could cause this or how i can fix it?

```python
cat dumpStream.py 
#!/usr/bin/python

import time
import sys
import os
import paho.mqtt.client as paho
broker = "<IP>"

folder = '/tmp/foo/'
file_name = folder + 'wav_'

if not os.path.exists(folder):
    os.makedirs(folder)
os.system('rm ' + folder + '/*')

def on_message(client, userdata, message: paho.MQTTMessage):
    # print(message.payload)
    # time.sleep(1)
    # print("received message =", str(message.payload.decode("utf-8")))

    millis = int(round(time.time() * 1000))

    f = open(file_name + str(millis) + '.wav', 'wb')
    f.write(message.payload)
    f.close()

client = paho.Client("client-001")
client.on_message = on_message

#####
print("connecting to broker ", broker)
client.username_pw_set('<USER>', '<PASS>')
client.connect(broker)  # connect
client.loop_start()  # start loop to process received messages

print("subscribing ")
client.subscribe("hermes/audioServer/default/audioFrame")  # subscribe
time.sleep(5)
client.disconnect()  # disconnect
client.loop_stop()  # stop loop

print("for f in ./*.wav; do echo \"file '$f'\" >> mylist.txt; done")
print("ffmpeg -f concat -safe 0 -i mylist.txt -c copy output.wav")
```

sehraf, 2020-12-23:
### I've successfully controlled a light with Homeassistant and rhasspy using my AudioKit thanks to your great work! :rocket: 

For the beginning I've attached my `device` implementation. A few things needed to be sorted out though:
- tell platformio which board to build
  - load different libraries
- make the code build the correct board (and ignore everything else)
- enable the use of buttons to trigger the recording / make rhasspy listen
- maybe add hot word detection?
- actually send `finishedMsg` :wink:

last but not least: reopen this issue! :smiley: 

```c++
#include "device.h"
#include <driver/i2s.h>
#include <AC101.h>

// I2S pins
#define IIS_SCLK 27
#define IIS_WSLC 26
#define IIS_DSIN 25
#define IIS_DSOU 35

#define CONFIG_I2S_BCK_PIN IIS_SCLK
#define CONFIG_I2S_LRCK_PIN IIS_WSLC
#define CONFIG_I2S_DATA_PIN IIS_DSIN
#define CONFIG_I2S_DATA_IN_PIN IIS_DSOU

// AC101 I2C pins
#define IIC_CLK 32
#define IIC_DATA 33

// amplifier enable pin
#define GPIO_PA_EN GPIO_NUM_21
#define GPIO_SEL_PA_EN GPIO_SEL_21

// LEDs
#define LED_D4 GPIO_NUM_19
#define LED_D5 GPIO_NUM_22
#define LED_WIFI LED_D5
#define LED_STREAM LED_D4

#define SPEAKER_I2S_NUMBER I2S_NUM_0

class AudioKit : public Device
{
public:
    AudioKit();
    void init();
    void updateLeds(int colors);
    void setReadMode();
    void setWriteMode(int sampleRate, int bitDepth, int numChannels);
    void writeAudio(uint8_t *data, size_t size, size_t *bytes_written);
    void readAudio(uint8_t *data, size_t size);

private:
    void InitI2SSpeakerOrMic(int mode);
    AC101 ac;
};

AudioKit::AudioKit(){};

void AudioKit::init()
{
    Serial.printf("Connect to AC101 codec... ");
    while (!ac.begin(IIC_DATA, IIC_CLK))
    {
        Serial.printf("Failed!\n");
        delay(1000);
    }
    ac.SetMode(AC101::MODE_ADC_DAC);

    uint8_t volume = 40;
    ac.SetVolumeSpeaker(volume);
    ac.SetVolumeHeadphone(volume);

    // LEDs
    pinMode(LED_STREAM, OUTPUT); // active low
    pinMode(LED_WIFI, OUTPUT);   // active low
    digitalWrite(LED_STREAM, HIGH);
    digitalWrite(LED_WIFI, HIGH);

    // Enable amplifier
    pinMode(GPIO_PA_EN, OUTPUT);
    digitalWrite(GPIO_PA_EN, HIGH);
};

void AudioKit::updateLeds(int colors)
{
    digitalWrite(LED_STREAM, HIGH);
    digitalWrite(LED_WIFI, HIGH);

    switch (colors)
    {
    case COLORS_HOTWORD:
        break;
    case COLORS_WIFI_CONNECTED:
        digitalWrite(LED_WIFI, HIGH);
        break;
    case COLORS_STREAM:
        digitalWrite(LED_STREAM, LOW);
        break;
    case COLORS_WIFI_DISCONNECTED:
        digitalWrite(LED_WIFI, LOW);
        break;
    }
};

void AudioKit::InitI2SSpeakerOrMic(int mode)
{
    esp_err_t err = ESP_OK;

    i2s_driver_uninstall(SPEAKER_I2S_NUMBER);
    i2s_config_t i2s_config = {
        .mode = (i2s_mode_t)(I2S_MODE_MASTER),
        .sample_rate = 16000,
        .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT, // is fixed at 12bit, stereo, MSB
        .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
        .communication_format = i2s_comm_format_t(I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB),
        .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,
        .dma_buf_count = 8,
        .dma_buf_len = 60,
    };
    if (mode == MODE_MIC)
    {
        i2s_config.mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX);
    }
    else
    {
        i2s_config.mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_TX);
        // i2s_config.use_apll = false;
        i2s_config.tx_desc_auto_clear = true;
    }

    err += i2s_driver_install(SPEAKER_I2S_NUMBER, &i2s_config, 0, NULL);
    i2s_pin_config_t tx_pin_config;

    tx_pin_config.bck_io_num = CONFIG_I2S_BCK_PIN;
    tx_pin_config.ws_io_num = CONFIG_I2S_LRCK_PIN;
    tx_pin_config.data_out_num = CONFIG_I2S_DATA_PIN;
    tx_pin_config.data_in_num = CONFIG_I2S_DATA_IN_PIN;

    err += i2s_set_pin(SPEAKER_I2S_NUMBER, &tx_pin_config);
    err += i2s_set_clk(SPEAKER_I2S_NUMBER, 16000, I2S_BITS_PER_SAMPLE_16BIT, I2S_CHANNEL_MONO);

    return;
}

void AudioKit::setWriteMode(int sampleRate, int bitDepth, int numChannels)
{
    if (mode != MODE_SPK)
    {
        InitI2SSpeakerOrMic(MODE_SPK);
        mode = MODE_SPK;
    }
    if (sampleRate > 0)
    {
        i2s_set_clk(SPEAKER_I2S_NUMBER, sampleRate, static_cast<i2s_bits_per_sample_t>(bitDepth), static_cast<i2s_channel_t>(numChannels));
    }
}

void AudioKit::setReadMode()
{
    if (mode != MODE_MIC)
    {
        InitI2SSpeakerOrMic(MODE_MIC);
        mode = MODE_MIC;
    }
}

void AudioKit::writeAudio(uint8_t *data, size_t size, size_t *bytes_written)
{
    i2s_write(SPEAKER_I2S_NUMBER, data, size, bytes_written, portMAX_DELAY);
}

void AudioKit::readAudio(uint8_t *data, size_t size)
{
    size_t byte_read;
    i2s_read(SPEAKER_I2S_NUMBER, data, size, &byte_read, (100 / portTICK_RATE_MS));
}
```

Romkabouter, 2020-12-23:
Great to hear it worked :)
I am currently changing a lot of code on this branch indeed, sorry to break stuff for you.
Excellent work on the device, there are a couple more methods coming that way.

With hotword detection you mean on device right? That is indeed the plan, I want the rewrite to have all the features the master branch has. Rhasspy will do the hotword detection if not on device (should already work, it does in my setup)
Finished message is not needed for rhasspy, and since snips is out of order it does not have my focus now (just as support for streamingBytes has not)
I need to find out if finishedMessage is used in Rhasspy. I'll reopen this issue for others to see and discuss

sehraf, 2020-12-23:
> sorry to break stuff for you.

No problem, that was kind of expected.

> With hotword detection you mean on device right? 

yes, my small ARM board dies after some time when running Rhasspy with active hot word detection :smile: I think for my personal use a simple button (on the device) will be enough. Though would be nice to have when the device could do hotword detection.

> I need to find out if finishedMessage is used in Rhasspy. 

i got some timeouts when (re)playing messages in the webui. Though i have no idea if these are the result of the missing finish message or something completely different.

Romkabouter, 2020-12-23:
I could add a method in the Device class to be overridden per device. A can add an event to change to the hotworddected state. The M5Atom had such a button on the device, so I can test it as well :)

There is the matter of audio coming in as well.  When the incoming audio has a higher sample rate, it sometimes gets the infamous hissing sound. It might be needed to set a maximum on the samplerate, because I never get these issues with 16000 as sample rate. I need more testing for that.

Thank you for your efforts in any case!

Romkabouter, 2020-12-27:
I have updated the code even more and added a hardware switch.
The code you have provided is slightly altered, but will be included in my next commit.
I have to create code for the Matrix Voice and update the readme a bit an will then push new code :)

Romkabouter, 2020-12-30:
I have just pushed a fairly large commit. I can't get the matrix voice running yet, but the M5 is running fine.
Your code has been slightly modified, but I did a copy&paste in general.

Can you check if you device builds? I have added resampler code, but that is for the Matrix Voice since that can only playback 44100 wav. It is not used in other devices, receiving 44100 files still has issues

sehraf, 2020-12-31:
Works fine! I've wired up a few things, like the hotword button: #28 

A few things I've noticed so far:
- when `isHotwordDetected` returns `true` the LEDs are not updated
- on my board the wifi/host name gets not set, so ota does not work (no idea what the actual problem is here, though others seem to have it, [too](https://github.com/espressif/arduino-esp32/issues/806))
- i was unable to get the button/hotword working. (I have enabled Dialogue Manager in Rhasspy, is there anything else?)

Romkabouter, 2020-12-31:
The hostname is not needed, only the correct IP address. It does print it in the Serial.
I have not severly tested the hotword button yet, I will look into it :)

Romkabouter, 2020-12-31:
Can you confirm if the isHotwordDetected function on your device is actually returning true?
A message should be published to hermes/dialogueManager/startSession in that case.
Works with the M5

Romkabouter, 2020-12-31:
Just pushed a fix for the hotword, it should only publish one message

sehraf, 2020-12-31:
> Can you confirm if the isHotwordDetected function on your device is actually returning true?

Yes, that part worked perfectly.

Romkabouter, 2021-01-01:
I suggest to close this issue and open a new one for the hotword issue.
Be sure to check the latest code, it should publish a message to hermes/dialogueManager/startSession which triggers the hotword.
Then, it should enter the HotwordState and that should be printed to the serial

Romkabouter, 2021-01-03:
Closing this, since first support is done

closed 2021-01-03 by Romkabouter