Romkabouter/ESP32-Rhasspy-Satellite #24: fatal error: ESPAsyncWebServer.h: No such file or directory

gfawkes66, 2020-08-08:
Hi,
I am new to this but the instructions seems pretty straightforward.
When i build however, i get an error of a missing library - ESPAsyncWebServer.h - output below

I have checked my sums but i wonder if i have missed a step or whether one of the dependency repositories has secretly changed.....

I have also had a quick search on github and only find the file here:-

luqmanhakimpens/esp32_sht31_oled
lib/ESPAsyncWebServer/src/ESPAsyncWebServer.h 

afonsus1997/Pixels-Camp-PCB-Badge
CODE/ESP/ESPAsyncWebServer.h

Thanks.!

error details:- 
Building in release mode
Compiling .pio\build\esp32dev\src\MatrixVoiceAudioServer.cpp.o
Compiling .pio\build\esp32dev\lib2b1\WiFi\WiFiMulti.cpp.o
Compiling .pio\build\esp32dev\lib2b1\WiFi\WiFiSTA.cpp.o
Compiling .pio\build\esp32dev\lib2b1\WiFi\WiFiScan.cpp.o
Compiling .pio\build\esp32dev\lib2b1\WiFi\WiFiServer.cpp.o
Compiling .pio\build\esp32dev\lib2b1\WiFi\WiFiUdp.cpp.o
src\MatrixVoiceAudioServer.cpp:104:31: fatal error: ESPAsyncWebServer.h: No such file or directory

***************************************************************************
* Looking for ESPAsyncWebServer.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESPAsyncWebServer.h"
* Web  > https://platformio.org/lib/search?query=header:ESPAsyncWebServer.h
*
***************************************************************************

compilation terminated.
Compiling .pio\build\esp32dev\lib2c2\ESPmDNS\ESPmDNS.cpp.o
*** [.pio\build\esp32dev\src\MatrixVoiceAudioServer.cpp.o] Error 1

Romkabouter, 2020-08-09:
This should be on the bottom of your platformio.ini:

```
lib_deps =
   https://github.com/matrix-io/matrixio_hal_esp32.git
   https://github.com/matrix-io/esp32-arduino-ota.git
   https://github.com/Locoduino/RingBuffer.git
   https://github.com/marvinroger/async-mqtt-client.git
   https://github.com/me-no-dev/AsyncTCP.git
   https://github.com/knolleary/pubsubclient.git
   https://github.com/bblanchon/ArduinoJson.git
```
Can you check that?

gfawkes66, 2020-08-09:
Hi,

Thanks for looking at this.
Yes, it looks the same to me.

lib_deps =
   https://github.com/matrix-io/matrixio_hal_esp32.git
   https://github.com/matrix-io/esp32-arduino-ota.git
   https://github.com/Locoduino/RingBuffer.git
   https://github.com/marvinroger/async-mqtt-client.git
   https://github.com/me-no-dev/AsyncTCP.git
   https://github.com/knolleary/pubsubclient.git
   https://github.com/bblanchon/ArduinoJson.git

Romkabouter, 2020-08-09:
Ok, thanks. I think I have installed it om my local machine without adding it to the libdeps.
Try adding: 
https://github.com/me-no-dev/ESPAsyncWebServer.git
to the lib_deps, I will check it out as well

gfawkes66, 2020-08-09:
yes thanks, that was the missing dependency.
I have successfully compiled and uploaded to my Matrix.
now i just have to figure out how it works!
thanks again.

closed 2020-08-09 by gfawkes66