Romkabouter/ESP32-Rhasspy-Satellite #25: Sound Playback quality is poor

gfawkes66, 2020-08-24:
Hello again,

I have this working quite well now, but have an issue with the audio playback.

issue:-
The audio playback is hissy/crackly and of extremely poor quality. (nothing approaching the 'Alexa demo' video i have seen of yours from a while back).
Sometimes i just get a hiss when something is attempted to be spoken.
It sounds like there is an issue with the volume or the frequency, but i understand your software corrects the formatting/ freq etc.
Mostly i hear the correct response, it just sounds awful.

I have tried:-
I have turned the volume right down and i understand that your software changes the sound to the correct format.
I wondered if it was latency, so i connected my wifi router directly to the 1gb switch which has the rhasppy / Homeassistant pi's.
its exactly the same if i use the 3.5mm jack or the 2 speaker connectors.
I have tried a couple of different speakers and also a headset. all the same.
I have tried a couple of different power sources.

setup:-
I think my setup is pretty much as you have given in examples in the past.
Responses are called to Rhasspy by homeassistant using the rest api http://server.local:12101/api/text-to-speech?siteId={{ site }} which works fine for my pi0 satellites.
I am using Mary TTS / Prudence voice.
I have one pi4 which has Mary TTS and Rhasspy.
Another pi4 with Homeassistant using the integrated MQTT.

I very much appreciate the assistance you have given so far.
Thanks again.

Romkabouter, 2020-08-26:
I also get the hissing sometimes, that seems to be a software issue.
Most probably related to playing two audio buffer at the same time, but I cannot seem to find a good solution.

Do the feedback sounds sound bas as well?

gfawkes66, 2020-08-29:
Hi,
yes the feedback sounds also sound bad, but not every time. I have tried the Hi/Lo beeps supplied with Rhasspy as well as a couple of phrases set at 44100 from Mary. both the same.

i have been testing with different feedback sounds, but this makes no difference either.
The Hissing is not consistent, although it sounds 'ok' sometimes.
Sometimes i get 2/3 response cycles with no major hissing.
Sometimes, i just get a bit of a click at the end of a feedback sound or the response.
Sometimes the feedback sound is replaced by a loud Hiss.
Sometimes the spoken word is replaced by a loud hiss.
It never goes 5 requests without a loud Hiss.

I have been using the same speaker attached to a Pi0 with a 2mic Seeed hat and that always sounds fine.

I am quite happy to try stuff if it helps and am in no rush!

Romkabouter, 2020-11-22:
I am currently trying to rewrite the code, because I think this issue is caused by interfering messages. Just so you know :)

gfawkes66, 2020-11-23:
Thanks for letting me know!

Hopefully you’ll have some success without too much pain. In the meantime, Ill keep an eye on github

Thanks again.

Romkabouter, 2021-01-03:
You can check out my rewrite branch, I never get it anymore, as long as the samplerate is not higher than 22050. The code will resample, which still needs some work because the quality is not very good but no hissing.
I still get the same issue with high samplerates (44100), but I am not focussing on that issue.
Please try the code from the rewrite branch and let me know

gfawkes66, 2021-01-03:
Hi,

I had a go but can’t get it to compile.
I must admit I am a bit of an occasional coder so this is not one of my strong points, but anyway, here is the terminal output.
If you see anything obvious, let me know ad I’ll try again.
regards

PS C:\Users\uasername\Matrix-Voice-ESP32-MQTT-Audio-Streamer-rewrite\PlatformIO> pio run
Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option

warning: Ignoring attempt to set reserved variable `$TARGETS'
File "C:\Users\username\Matrix-Voice-ESP32-MQTT-Audio-Streamer-rewrite\PlatformIO\load_settings.py", line 36, in <module>
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (2.1.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
- framework-arduinoespressif32 3.10004.201016 (1.0.4)
- tool-esptoolpy 1.30000.201119 (3.0.0)
- toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 39 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <MATRIXVoiceESP32HAL> 0.77.0+sha.320c897
|-- <RingBuffer> 1.0.3+sha.06e8bf1
|-- <AsyncMqttClient> 0.8.2+sha.7f1ba48
|   |-- <AsyncTCP> 1.1.1
|-- <AsyncTCP> 1.1.1+sha.ca8ac5f
|-- <PubSubClient> 2.8.0+sha.2d228f2
|-- <ArduinoJson> 6.17.2+sha.1360b6a
|-- <ESP Async WebServer> 1.2.3
|   |-- <AsyncTCP> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <ArduinoJson> 6.17.2+sha.1360b6a
|-- <M5Atom> 0.0.1
|   |-- <FastLED> 3.4.0
|   |   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|-- <FastLED> 3.4.0
|   |-- <SPI> 1.0
|-- <AC101> 0.0.1
|   |-- <Wire> 1.0.1
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
|-- <WiFi> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <Update> 1.0
|   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|-- <libspeex>
Building in release mode
Compiling .pio\build\esp32dev\src\Satellite.cpp.o
Compiling .pio\build\esp32dev\libea1\ESP Async WebServer\WebServer.cpp.o
Compiling .pio\build\esp32dev\libb28\SPI\SPI.cpp.o
Compiling .pio\build\esp32dev\lib46f\Wire\Wire.cpp.o
Archiving .pio\build\esp32dev\libb28\libSPI.a
Compiling .pio\build\esp32dev\lib73c\M5Atom\M5Atom.cpp.o
In file included from src\Satellite.cpp:97:0:
src/General.hpp:23:3: error: stray '\' in program
   IPAddress mqtt_host = MQTT_IP;
   ^
src/General.hpp:23:3: error: stray '\' in program
In file included from .pio/libdeps/esp32dev/M5Atom/src/M5Atom.h:52:0,
                 from src\devices/M5AtomEcho.hpp:1,
                 from src\Satellite.cpp:87:
[... 23 lines cut ...]
Compiling .pio\build\esp32dev\lib73c\M5Atom\utility\LED_DisPlay.cpp.o
*** [.pio\build\esp32dev\src\Satellite.cpp.o] Error 1
In file included from .pio\libdeps\esp32dev\M5Atom\src\utility\LED_Display.h:4:0,
                 from .pio\libdeps\esp32dev\M5Atom\src\utility\LED_Display.cpp:1:
.pio\libdeps\esp32dev\FastLED\src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000
#    pragma message "FastLED version 3.004.000"
                     ^
In file included from .pio\libdeps\esp32dev\FastLED\src/FastLED.h:65:0,
                 from .pio\libdeps\esp32dev\M5Atom\src\utility\LED_Display.h:4,
                 from .pio\libdeps\esp32dev\M5Atom\src\utility\LED_Display.cpp:1:
.pio\libdeps\esp32dev\FastLED\src/fastspi.h:135:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
#      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
In file included from .pio\libdeps\esp32dev\FastLED\src/controller.h:9:0,
                 from .pio\libdeps\esp32dev\FastLED\src/FastLED.h:47,
                 from .pio\libdeps\esp32dev\M5Atom\src\utility\LED_Display.h:4,
                 from .pio\libdeps\esp32dev\M5Atom\src\utility\LED_Display.cpp:1:
.pio\libdeps\esp32dev\FastLED\src/pixeltypes.h: In member function 'virtual void LED_Display::run(void*)':
.pio\libdeps\esp32dev\FastLED\src/pixeltypes.h:163:37: warning: iteration 25u invokes undefined behavior [-Waggressive-loop-optimizations]
         r = (colorcode >> 16) & 0xFF;
                                     ^
.pio\libdeps\esp32dev\M5Atom\src\utility\LED_Display.cpp:19:27: note: containing loop
     for (int num = 0; num < 26; num++)
                           ^
===================================================================== [FAILED] Took 19.67 seconds

Romkabouter, 2021-01-03:
That is because you are on windows. Change the devices/ to devices\

Romkabouter, 2021-01-13:
As can be seen here, the build fail is due to windows:

https://github.com/Romkabouter/Matrix-Voice-ESP32-MQTT-Audio-Streamer/issues/32#issuecomment-759529742

Closing this

closed 2021-01-13 by Romkabouter