Romkabouter/ESP32-Rhasspy-Satellite #27: won't connect to mosquitto, socket error

wzbfyb, 2020-12-17:
Hi,

when Matrix Voice boots up, the leds change blue/red all the time and in mosquito log I see:
`
1608233147: New connection from 192.168.132.27 on port 1883.
1608233147: Socket error on client <unknown>, disconnecting.
`

any hints whats going on there?

wzbfyb, 2020-12-18:
noticed something: when I setup mosquito to allow anonymous logins, and remove user/pw it works.
seems to be an issue with my password:

username=default
password="W6o4wUc(/76"

I had to use " " for the password, otherwise the ( in the pw would cause a compilation error.

Romkabouter, 2020-12-18:
Might be a password problem, please do not use a slash as special character

closed 2020-12-18 by Romkabouter

wzbfyb, 2020-12-18:
most likely, thanks.
Will setup a dedicated user for matrix voice and avoid nasty chars in the password 👍