Romkabouter/ESP32-Rhasspy-Satellite #29: No wifi connection

thegnuu, 2021-01-02:
I am currently trying to setup this streamer on my matrix voice, unfortunately it seems like the device is not able to connect to my wifi, the LED is red at the beginning, turns to blue after a few seconds but I cannot find the device in the clients-list of my firewall and there is no connection to my MQTT broker.

This is my current config, flashing it works like without any problem.
Of course the ssid/mqtt settings are changed, but i triple-checked all params and passwords.

Has anyone an idea what I am doing wrong?

```
[Matrix]
hostname=10.10.10.30
siteId=matrixlivingroom

[Wifi]
ssid=myWifi
password=my.super!password?

[OTA]
password=OTA
port=3232

[MQTT]
ip=10.10.10.11
hostname=homeassistant
port=1883
username=myUser
password=mySuperMqttPassword
maxPacketSize=2000
```

Romkabouter, 2021-01-02:
In an upcoming version, the hostname will be removed. Can your try with hostname in MQTT set as 10.10.10.11?
If the leds are blue, there should be a wifi connection.
If you can, install minicom on the pi and monitor the serial

andersonsmith, 2021-01-10:
Hi!  I was having the same issue with 6.0.  Trying to flash with the rewrite branch, but I'm getting a strange error in VSCode/PIO.  It's complaining about a stray \ from this line:

`IPAddress mqtt_host = MQTT_IP;`

I'm assuming it's something I've done in VSCode, but I can't figure it out for the life of me.  When I run the code with what I have in settings in a separate python instance, I get `'IPAddress\\(10,0,1,20\\)'` which I assume is the desired format.

Romkabouter, 2021-01-10:
What is the error message you get? 
Can you also state your operation system?
Also, the setting [Matrix] is renamed as [General] in rewrite branch.
You should change that in your settings, it is best that you use the settings example (again) and then re-add your own settings in that.

thegnuu, 2021-01-12:
> In an upcoming version, the hostname under mqtt will be removed. Can your try with hostname in MQTT set as 10.10.10.11?
> If the leds are blue, there should be a wifi connection.
> If you can, install minicom on the pi and monitor the serial
> 
> You could also try the rewrite branch. Playback is not 100%, but I think that branch is more stable.

@Romkabouter thank you for your response and sorry it took me so long to check it out.

It looks like the device is now connected to my network, which is nice :) Unfortunately I still receive `No valid IP address for MQTT` when I debug with minicom. But this is unfortunately not true ;) I used exact the same settings in MQTT Explorer and it could connect without any issues.

I also realised that `Failed to read file, using default configuration` is printed somewhere, but the wifi connection seems to work.

Romkabouter, 2021-01-12:
Yeah, the default config message is ok. Will pop up the first time you flash, the config is saved on SPIFFS.
When there is no config file, it should take over the MQTT_IP setting.

If not, you can browse to the ip adress of the device (in your browser) and you can change the MQTT ip from there :)

thegnuu, 2021-01-12:
Thank you for your help! I didn't even now there is a config web ui :)
I was now able to set the MQTT ip (it was set to 0.0.0.0) and now I am receiving hermes notifications via MQTT!
Now I just need to figure out how to configure rhasspy, but I guess I will find a way to do this.

Thanks a lot and I am happy to help you test the new version as soon as it is ready, will setup the 3 remaining matrix now!

Romkabouter, 2021-01-12:
Great, strange that it did not pick the MQTT_IP with the first flash though, I'll check into that.
Closing this for now

closed 2021-01-12 by Romkabouter

Romkabouter, 2021-01-12:
Can you tell me which operation system you are building?

thegnuu, 2021-01-12:
So far the stream to rhasspy works, only the wake word detection has issues, but this might be due to the config...

I am using MacOS 11.0.1.

Romkabouter, 2021-01-13:
Ok, strange that you get that error because I am on MacOS as well.
Although this issue is closed, I might be able to help with your settings. 
If your server has a different siteId than your Matrix, you should add your Matrix siteId in every "Satellite siteIds:" setting, under all config sections.
Also Audio Recording and Audio Playing should be Hermes MQTT, but you probably have that. The rest should NOT be set to Hermes MQTT. I have set the following:
- mqtt: external
- audio recording: hermes
- wakeword: porcupine
- speech to text: kaldi
- intent recognition: fsticuffs
- text to speech: wavenet
- audio playing: hermes
- dialogue management: rhasspy
- intent handling: home assistant

thegnuu, 2021-01-13:
This configuration is working on my old Rhasspy installation, you are right! I think it has nothing to do with the streamer itself.

I was switching my installation yesterday to a new docker installation and I guess there is an issue with the wake word itself within Rhasspy.

Thank you anyway :)

Romkabouter, 2021-01-13:
I am using the addon in Home Assistant as Rhasspy server

thegnuu, 2021-01-13:
Yeah in this case it is totally an issue with my installation, I am just not able to configure the wake word file properly...

I added it to my profile but Rhasspy seems not to be able to find it....

Romkabouter, 2021-01-13:
Ok, I suggest using a default one for start. 
If that works, you can add your own. I find snowboy a good one.

thegnuu, 2021-01-13:
Tried all of them, none seems to work. Anyway, I will try to clean the complete installation and reinstall it, maybe there is something wrong with my entire setup :)

Thank you for your help!