Romkabouter/ESP32-Rhasspy-Satellite #31: Question: Rhasspy with satellites

thegnuu, 2021-01-12:
I was able to set up everything properly working with Rhasspy at least if I set the global siteId of my Rhasspy instance to the same the matrix has.

My original idea was to use one Rhasspy instance and add multiple matrix voice as satellites, unfortunately, if i configure it this way, the LED is flashing green for a short time on wake word, turns back to blue and then to green again and the commands are mostly not recognised (1 out of 10 is recognised or so).

I am not sure, this might be of course an issue with Rhasspy itself, but maybe you have an idea on how to setup satellites the proper way?

This configuration is working great:
- siteId: your-matrix-id
- no satellites configured

This configuration is not working:
- siteId: default
- satelliteId your-matrix-is configured on
  - Wakeword
  - Speech to Text
  - Intent Recognition
  - Dialogue Management

Romkabouter, 2021-01-12:
Assuming you are using the master branch:
- the client is set hardcoded to MatrixVoiceAudio, that is causing connection issues and might well be your issue (although I'd need more info)

This has been changed in the rewrite branch, but if you want to use the master branch:
- search for: audioServer.connect("MatrixVoiceAudio", MQTT_USER, MQTT_PASS) and change the clientID per device
- search for: asyncClient.setClientId("MatrixVoice"); and change the clientID per device

My suggestion is to use the rewrite branch :)

Keep me posted on progress please

thegnuu, 2021-01-12:
Thanks for the info, in this case I will switch to the rewrite branch in the next few days.
But I don't think that this causes my issue at the moment, since the problem occurs as well even if only one Matrix is online.

I will try to debug the MQTT messages Rhasspy sends, but I guess I will not find much time until Sunday to do this.

Romkabouter, 2021-01-12:
Maybe I misread your config.
This works for me:
- MatrixVoice => siteId "office"
- Rhasspy server => siteId "server", "office" in all satellite Id's
settings:
[image]

thegnuu, 2021-01-13:
Looks exactly like my config, except I have not enabled a few settings.
Mine looks like this:

[image]

Which Rhasspy version are you using?

Romkabouter, 2021-01-13:
2.5.8, I have the addon installed in Home Assistant

Romkabouter, 2021-01-13:
rewrite branch has been merged to master

thegnuu, 2021-01-14:
Thanks, I will update this weekend. :)

closed 2021-01-15 by Romkabouter