Romkabouter/ESP32-Rhasspy-Satellite #32: Build failed

wengtad, 2021-01-13:
Hi, first of all, thanks for your great project. I'm trying to build using rewrite branch but i had encountered some issues which i don't know what is going on here. 

I'm using Windows 10 and the tool is PlatformIO on VSCode. The `settings.ini` value is same as the `settings.ini.example`.
The reason to use rewrite branch because it support atom echo.

Some output logs:
```
...
Found 40 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <MATRIXVoiceESP32HAL> 0.77.0+sha.320c897
|-- <RingBuffer> 1.0.3+sha.06e8bf1
|-- <AsyncMqttClient> 0.8.2+sha.7f1ba48
|   |-- <AsyncTCP> 1.1.1
|-- <AsyncTCP> 1.1.1+sha.ca8ac5f
|-- <PubSubClient> 2.8.0+sha.2d228f2
|-- <ArduinoJson> 6.17.2+sha.1360b6a
|-- <ESP Async WebServer> 1.2.3
|   |-- <AsyncTCP> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <ArduinoJson> 6.17.2+sha.1360b6a
|-- <M5Atom> 0.0.1
|   |-- <FastLED> 3.4.0
|   |   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|-- <FastLED> 3.4.0
|   |-- <SPI> 1.0
|-- <AC101> 0.0.1
|   |-- <Wire> 1.0.1
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
|-- <WiFi> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <Update> 1.0
|   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|-- <libspeex>
Building in release mode
Compiling .pio\build\esp32dev\libefb\FastLED\FastLED.cpp.o
Compiling .pio\build\esp32dev\libefb\FastLED\bitswap.cpp.o
Compiling .pio\build\esp32dev\libefb\FastLED\colorpalettes.cpp.o
Compiling .pio\build\esp32dev\libefb\FastLED\colorutils.cpp.o
Compiling .pio\build\esp32dev\libefb\FastLED\hsv2rgb.cpp.o
Compiling .pio\build\esp32dev\libefb\FastLED\lib8tion.cpp.o
Compiling .pio\build\esp32dev\libefb\FastLED\noise.cpp.o
Compiling .pio\build\esp32dev\libefb\FastLED\platforms.cpp.o
Compiling .pio\build\esp32dev\libefb\FastLED\platforms\esp\32\clockless_rmt_esp32.cpp.o
.pio\libdeps\esp32dev\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp: In static member function 'static void ESP32RMTController::interruptHandler(void*)':
.pio\libdeps\esp32dev\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:318:10: warning: unused variable 'stuff_to_do' [-Wunused-variable]
     bool stuff_to_do = false;
          ^
Compiling .pio\build\esp32dev\libefb\FastLED\power_mgt.cpp.o
Compiling .pio\build\esp32dev\libefb\FastLED\wiring.cpp.o
Compiling .pio\build\esp32dev\src\Satellite.cpp.o
Generating partitions .pio\build\esp32dev\partitions.bin
Compiling .pio\build\esp32dev\liba6c\MATRIXVoiceESP32HAL\everloop.cpp.o
Compiling .pio\build\esp32dev\liba6c\MATRIXVoiceESP32HAL\matrix_driver.cpp.o
Compiling .pio\build\esp32dev\liba6c\MATRIXVoiceESP32HAL\microphone_array.cpp.o
Compiling .pio\build\esp32dev\liba6c\MATRIXVoiceESP32HAL\microphone_core.cpp.o
Compiling .pio\build\esp32dev\liba6c\MATRIXVoiceESP32HAL\wishbone_bus.cpp.o
Archiving .pio\build\esp32dev\lib21b\libRingBuffer.a
Compiling .pio\build\esp32dev\libe65\AsyncTCP\AsyncTCP.cpp.o
Compiling .pio\build\esp32dev\lib18a\AsyncMqttClient\AsyncMqttClient.cpp.o
Compiling .pio\build\esp32dev\lib18a\AsyncMqttClient\AsyncMqttClient\Packets\ConnAckPacket.cpp.o
Compiling .pio\build\esp32dev\lib18a\AsyncMqttClient\AsyncMqttClient\Packets\PingRespPacket.cpp.o
In file included from src\Satellite.cpp:97:0:
src/General.hpp:23:3: error: stray '\' in program
   IPAddress mqtt_host = MQTT_IP;
   ^
src/General.hpp:23:3: error: stray '\' in program
In file included from .pio\libdeps\esp32dev\MATRIXVoiceESP32HAL\components\hal\microphone_core.cpp:31:0:
.pio\libdeps\esp32dev\MATRIXVoiceESP32HAL\components\hal\./microphone_array_location.h:33:14: warning: 'matrix_hal::micarray_location' defined but not used [-Wunused-variable]
 static float micarray_location[8][2] = {
              ^
In file included from .pio/libdeps/esp32dev/M5Atom/src/M5Atom.h:52:0,
                 from src\devices/M5AtomEcho.hpp:1,
                 from src\Satellite.cpp:87:
.pio/libdeps/esp32dev/FastLED/src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000
 #    pragma message "FastLED version 3.004.000"
                     ^
Compiling .pio\build\esp32dev\lib18a\AsyncMqttClient\AsyncMqttClient\Packets\PubAckPacket.cpp.o
In file included from .pio/libdeps/esp32dev/FastLED/src/FastLED.h:65:0,
                 from .pio/libdeps/esp32dev/M5Atom/src/M5Atom.h:52,
                 from src\devices/M5AtomEcho.hpp:1,
                 from src\Satellite.cpp:87:
Compiling .pio\build\esp32dev\lib18a\AsyncMqttClient\AsyncMqttClient\Packets\PubCompPacket.cpp.o
.pio/libdeps/esp32dev/FastLED/src/fastspi.h:135:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
Archiving .pio\build\esp32dev\liba6c\libMATRIXVoiceESP32HAL.a
Compiling .pio\build\esp32dev\lib18a\AsyncMqttClient\AsyncMqttClient\Packets\PubRecPacket.cpp.o
Compiling .pio\build\esp32dev\lib18a\AsyncMqttClient\AsyncMqttClient\Packets\PubRelPacket.cpp.o
Compiling .pio\build\esp32dev\lib18a\AsyncMqttClient\AsyncMqttClient\Packets\PublishPacket.cpp.o
Archiving .pio\build\esp32dev\libe65\libAsyncTCP.a
Compiling .pio\build\esp32dev\lib18a\AsyncMqttClient\AsyncMqttClient\Packets\SubAckPacket.cpp.o
Compiling .pio\build\esp32dev\lib18a\AsyncMqttClient\AsyncMqttClient\Packets\UnsubAckPacket.cpp.o
Compiling .pio\build\esp32dev\lib349\AsyncTCP@src-7fb2940bccb78b8d2de6915ae328b7fc\AsyncTCP.cpp.o
*** [.pio\build\esp32dev\src\Satellite.cpp.o] Error 1
============================================ [FAILED] Took 25.91 seconds ============================================
...
```

Romkabouter, 2021-01-13:
Probably due to the difference of \ and /  in windows.
I cannot test that, but you might try and change the IPadress\ in the load_setting.py and in Satellite.cpp the devices/ (around line 80)

wengtad, 2021-01-13:
Thanks for you help. I'm able to build it success by changing in the load_setting.py (around line 25).

```py
("MQTT_IP", "IPAddress(" + config[sectionMqtt]["ip"].replace(".", ",") + ")"),
```

Not test on atom echo yet. Once have the result will update here again. Thanks.

wengtad, 2021-01-13:
Yes it is working on atom echo. I'm closing this issue, thanks.

Romkabouter, 2021-01-13:
I can use a windows machine now and conclude that this is indeed an usse.

Reopening :)

Romkabouter, 2021-01-13:
Should be fixed with
https://github.com/Romkabouter/Matrix-Voice-ESP32-MQTT-Audio-Streamer/commit/7dffc9d37342d1e64d7459d56d8e7b42dd755b39

closed 2021-01-13 by Romkabouter