Romkabouter/ESP32-Rhasspy-Satellite #35: Question towards endianness conversions

maxbachmann, 2021-01-29:
I am slightly confused by the following code:
```cpp
// Convert 4 byte little-endian to a long,
#define longword(bfr, ofs) (bfr[ofs + 3] << 24 | bfr[ofs + 2] << 16 | bfr[ofs + 1] << 8 | bfr[ofs + 0])
#define shortword(bfr, ofs) (bfr[ofs + 1] << 8 | bfr[ofs + 0])
```
according to the comment this converts from little endian to long. Since the endianness of long is not stated I assumed, that it means the host endianness. However since the esp32 is little endian aswell this does not make much sense. It would probably help to use a naming convention similar to e.g. Linux https://linux.die.net/man/3/endian

Romkabouter, 2021-01-30:
To be honest I copied that code :D
The reason is the waveheader format: http://soundfile.sapp.org/doc/WaveFormat/
That is a mix of little and big endian. To correctly extract the samplerate etc, that needs to be converted.

Romkabouter, 2021-02-17:
@maxbachmann do you have some sparetime left to correct this?
Otherwise I will try and see if I can :)

maxbachmann, 2021-02-17:
I was apparently not thinking straight when I wrote this :disappointed_relieved:
The macro simply takes a little-endian value out of memory by reading it in bytes and combining them to a little-endian value

closed 2021-02-17 by maxbachmann