Romkabouter/ESP32-Rhasspy-Satellite #39: Rebooting every few seconds?

hpmueller1971, 2021-02-14:
Hi,

i successfully installed on an Atom Echo, it seems to work (ping, webui reachable), but it keeps rebooting every few seconds in a loop.

`ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
BootiM5Atom initializing...Loading configuration
{
  "mqtt_host": "10.123.1.2",
  "mqtt_port": 1883,
  "mqtt_user": "",
  "mqtt_pass": "",
  "mqtt_valid": true,
  "mute_input": false,
  "mute_output": false,
  "amp_output": 0,
  "brightness": 30,
  "hotword_brightness": 100,
  "hotword_detection": 0,
  "volume": 100,                                                                                        
  "gain": 5                                                                                             
}                                                                                                       
Enter WifiDisconnected                                                                                  
Total heap: 270760                                                                                      
Free heap: 196232
Enter WifiConnected
Connected to Wifi with IP: 10.123.1.33
Enter MQTTDisconnected
Enter MQTTConnected
Connected as satellite
Enter Idle

Backtrace: 0x4008ca58:0x3ffbe170 0x4008cc89:0x3ffbe190 0x400ec4b3:0x3ffbe1b0 0x40084ab9:0x3ffbe1d0 0x4000bfed:0x3ffeb550 0x4008a181:0x3ffeb560 0x400896e5:0x3ffeb580 0x400d2e75:0x3ffeb5a0 0x40089015:0x3ffeb820

Rebooting...
ets Jun  8 2016 00:22:57
`

The Backtrace looks the same every time. I've tried with dhcp and static ip.

Any ideas?

regards,

/hp

sp33dsk8, 2021-02-14:
I'm not working on the project but it appears since [Satellite.cpp v7.0 hotword has been disabled locally](https://github.com/Romkabouter/ESP32-Rhasspy-Satellite/blob/362b795adaa065e1e91387f2f7e82469ffa22cfc/PlatformIO/src/Satellite.cpp#L74)

Romkabouter, 2021-02-14:
Local hotword detection does not work on the latest version due to the fact that the newest esp library does not like the WakeNet library used.

This should not cause the reboot however, so I will look into this.

hpmueller1971, 2021-02-14:
Ah, thanks guys, i missed that in the commit-log! So i play with the remote-wakeword detection for the time being... Does the fact, that you didnt remove the web-option mean, that there is hope for the WakeNet integration (this is also from espressif, right?), or is WakeNet dead for the ESP32?

Romkabouter, 2021-02-14:
WakeNet development seems to have stopped. I was working in release 6, but the network stability was less good and also the M5 is not supported with that version (actually only the Matrix Voice is and that is the whole reason for the rewrite)

I am hoping on either a new WakeNet library or a Porcupine library. Untill then.. no local hotword ;(
Maybe it is best to remove the setting for now indeed. App still should not crash though.

uc-david, 2021-02-16:
Hello,
I had the same problem since last Friday. Now i found the Problem. The actual mosquitto broker doesnt work! I tried the  eclipse-mosquitto:1.6.12-- then the Atom Echo doesent reboot!

flatsiedatsie, 2021-02-19:
I'm out of my depth, but I think the issue might be related to `MQTTDisconnected` being called both by the state machine and by the run section:

```
  void run(void) override {
    //Serial.print("in run");
    if (audioServer.connected() && asyncClient.connected()) {
      transit<MQTTConnected>();
    } else {
      currentMillis = millis();
      if (currentMillis - startMillis > 5000) {
        Serial.println("Connect failed, retry");
        transit<MQTTDisconnected>();
      }      
    }
  }
```

and

```
  void react(MQTTDisconnectedEvent const &) override { 
    transit<MQTTDisconnected>();
  }
```

So there are two places that try to react to a disconnection by starting the reconnect process?

Romkabouter, 2021-02-19:
No, it only reacts to events when in a state.

So what happens in the case you mention, the MQTTDisconnected state transits to itself when after 5 seconds no connection is succesfull.
There is no MQTTDisconnectedEvent fired.

That event is ONLY fired in the I2S task, whenever the code detects that the audioserver is disconnected.

flatsiedatsie, 2021-02-19:
Thanks. I'm trying to get it to work, but I'm a bit out of my depth. So far I've gotten it to stay in Idle without crashing. I'm seeing some things that I don't understand.

Still seeing this double `MQTTDisconnected` after the I2STask notices an MQTT disconnect, followed by a crash.
```
onnected to Wifi with IP: 192.168.2.137
WifiConnected entry complete, moving to MQTTDisconnected
Enter MQTTDisconnected
MQTT was not initialized yet.
past setClientId
192.168.2.167
1883
clientID = atomechoAudio
past asyncclient.connect
past audioServer connect, reached end of mqttdisconnected init
end of setupEnter MQTTConnected
Connected as atomecho
subscribed to topics
Enter Idle
Still Idle
start streaming
past stream
end of idle
-I2S believes audioserver is disconnected -> sending MQTTDisconnectedEvent()
idle react to MQTTDisconnectedEvent
Enter MQTTDisconnected
Enter MQTTDisconnected
- had to disconnect async client
past setClientId
past setClientId
192192.168.2.167
1883
.168.2.167
1883
clientID = atomechoAudio
clientID = atomechoAudio
past asyncclient.connect
past asyncclient.connect

Backtrace: 0x4008c728:0x3ffbe170 0x4008c959:0x3ffbe190 0x400e7803:0x3ffbe1b0 0x40084789:0x3ffbe1d0 0x400819cf:0x3ffddef0 0x400826bf:0x3ffddf10 0x4008126f:0x3ffddf30 0x400dcf4d:0x3ffddf50 0x400dcfc6:0x3ffddf90 0x400d1e46:0x3ffddfc0 0x400d1e87:0x3ffde050 0x400d1f78:0x3ffde070 0x400d27e9:0x3ffde090 0x40088ce5:0x3ffde510

```

If I do a test push of a message in the I2sTask, it arrives if I send a fake string. But the actual audio data never arrives.

```
            //audioServer.publish(audioFrameTopic.c_str(),(uint8_t *)payload, sizeof(payload)); // <- this does not arrive
            audioServer.publish(audioFrameTopic.c_str(),"x", 1); // <- this arrives 
```

Even when sending these fake packets, I2sTask still notices that the audioServer is disconnected. I'm not 100% sure it actually is.

Not receiving audio data might be what Snips Watch is complaining about here, but I'm not sure. I believe this is after pressing the button:
```
[13:18:31] [Dialogue] was asked to start a session on site atomecho
[13:18:31] [Asr] was asked to stop listening on site atomecho
[13:18:31] [Hotword] was asked to toggle itself 'off' on site atomecho
[13:18:31] [Dialogue] session with id '110943cd-8fea-40ac-9a15-2f990279144f' was started on site atomecho
[13:18:31] [Asr] was asked to listen on site atomecho
[13:18:47] [Dialogue] session with id '110943cd-8fea-40ac-9a15-2f990279144f' was ended on site atomecho. The session was ended because one of the component didn't respond in a timely manner
[13:18:47] [Asr] was asked to stop listening on site atomecho
[13:18:47] [Hotword] was asked to toggle itself 'on' on site atomecho
```

I suspect Rhasspy has added the `reason` attribute to the spec. I tried removing that check to make it Snips compatible again:

```
        if (root["siteId"] == SITEID){
          send_event(IdleEvent());
          if(root.containsKey("reason")
            && root["reason"] == "dialogueSession") {
              //send_event(IdleEvent());
          }
        }
```

flatsiedatsie, 2021-02-19:
I checked, and the microphone data does exist. It just doesn't see to be sent over.

```
82737070442008765866910210911632160001010128620001250020160100971169702007923681236762368023688236822368323682236882368623684236782369323690236103236992369223610423683236802369123684236882368223695236842361042361042369623693236952369723682236902369323683236102236972369223610323694236104236104236912368623696236102236962361032361092369723610523610923696236106236982361172361132361092361082361112361122361022361012361052361002361012369323610323610023612123611523612123612223612923612123614923613523613523614023613523614023611923612223612323611123612323612523612323612523612123613023611423611723611923611323612523612923612823612823612623614023613223613923614023613123612223613323613323612923610223611523687236812369423690236952369223684236852369523694236922369423672236812367023666236722368023682236762367423679236752366823691236792366823690236682367923658236712366523657236612366323638236482364223636236412364723626236352362323623236282362123630236242363023633236392363123634236412362823631236262363723662361723632366236255235023610236253235323625523511236423692361923652368236122361623613236523672365236523642362522357236623622362502352236122362532358236823632361023612367236222369236342362023642236332362923627236242362023617236172362462358236223672361823611236212363823620236192367236152361423610236232361823614236132361123616236523617236232361823616236122363123626236252362623621236232362532355236
8273707044200876586691021091163216000101012862000125002016010097116970200217236218236229236211236220236237236214236216236216236224236207236219236223236214236225236220236232236219236223236221236212236210236215236207236212236215236229236225236240236226236222236231236217236218236221236216236236236221236237236242236238236240236215236237236205236211236220236207236217236223236214236217236202236217236212236196236197236207236208236199236210236210236207236203236230236224236221236221236211236221236210236208236202236209236204236205236191236189236205236196236192236194236193236193236187236193236190236194236170236175236184236168236186236187236190236187236155236152236151236156236145236154236157236151236166236160236166236174236163236168236171236165236180236182236182236185236187236185236191236181236197236208236200236192236204236204236194236184236199236197236199236197236198236199236220236212236207236221236225236222236219236222236217236211236211236212236220236212236218236210236224236218236223236222236223236232236227236237236222236223236232236229236220236228236233236221236218236222236211236214236214236207236219236219236210236198236197236203236182236202236177236187236197236188236206236203236214236213236216236211236203236212236211236216236201236221236212236214236215236217236210236205236211236212236216236212236211236218236228236219236225236229236226236225236223236220236210236213236212236213236212236218236205236215236204236206236204236201236206236196236205236216236222236210236222236221236210236209236227236210236233236221236219236225236222236224236214236222236232236220236208236222236219236219236211236220236
8273707044200876586691021091163216000101012862000125002016010097116970200226236198236228236222236218236227236194236214236192236191236181236193236177236178236174236180236175236176236180236171236203236176236196236198236211236201236215236209236212236216236200236212236190236200236186236186236199236192236190236181236199236197236205236205236208236201236196236205236187236196236190236197236204236206236204236206236202236213236188236191236219236204236202236217236202236210236204236210236214236214236210236216236210236219236217236212236227236212236222236222236206236202236196236193236210236202236203236197236212236214236218236212236191236211236197236192236207236204236210236211236214236220236205236203236203236188236204236200236192236186236188236186236193236195236196236190236212236205236200236201236226236198236228236222236218236227236194236214236192236191236181236193236177236178236174236180236175236176236180236171236203236176236196236198236211236201236215236209236212236216236200236212236190236200236186236186236199236192236190236181236199236197236205236205236208236201236196236205236187236196236190236197236204236206236204236206236202236213236188236191236219236204236202236217236202236210236204236210236214236214236210236216236210236219236217236212236227236212236222236222236223236219236231236232236232236240236224236226236215236218236209236219236196236212236215236214236209236217236204236213236227236211236227236220236218236217236206236208236201236200236206236213236213236203236216236221236214236211236205236212236193236201236200236201236198236204236212236204236185236211236184236187236185236180236195236186236
8273707044200876586691021091163216000101012862000125002016010097116970200139236126236143236134236148236149236143236150236133236145236150236141236164236154236183236169236174236174236171236158236167236168236169236177236166236180236167236170236186236179236183236182236178236182236177236178236188236182236168236174236174236170236172236174236172236176236161236175236155236158236140236148236145236150236143236136236140236152236123236130236144236151236154236141236147236138236143236140236137236139236135236150236130236135236127236127236134236119236126236127236136236133236147236142236136236156236137236135236116236129236124236108236126236129236129236135236129236125236127236133236142236134236129236137236142236140236138236138236133236131236132236125236126236131236139236139236131236140236146236149 [... cut ...] 234227234215234208234210234206234203234212234203234200234215234198234189234192234202234191234182234184234189234191234186234178234188234190234178234172234184234194234181234176234198234173234178234168234171234161234155234177234156234172234170234164234182234168234164234172234180234173234169234157234164234160234165234164234167234173234155234190234171234194234188234194234197234188234192234174234188234179234177234198234196234190234201234183234193234167234180234150234155234166234162234166234183234

```

Romkabouter, 2021-02-19:
The double MQTTDisconnected is indeed a bit strange.
About the payloads not arriving, can you verify that in your load_settings.py the MQTT_MAX_PACKET_SIZE is set to 2000?

flatsiedatsie, 2021-02-19:
> DId you put in the "idle react to MQTTDisconnectedEvent" in the MQTTConnected react as well?

No I hadn't. I have now, and will keep an eye out.

> can you verify that in your load_settings.py the MQTT_MAX_PACKET_SIZE is set to 2000?

I upload via the Arduino IDE. I do have `#define MQTT_MAX_PACKET_SIZE 2000` in the [code here](https://github.com/createcandle/voco-mini-satellite/blob/340ac1258b56d766339ec937a325f43c05c5e37c/voco-mini-satellite.ino#L124).

flatsiedatsie, 2021-02-19:
Found this in the PubSubClient issue queue:

https://github.com/knolleary/pubsubclient/issues/810

flatsiedatsie, 2021-02-19:
I've gotten a bit further!

> can you verify that in your load_settings.py the MQTT_MAX_PACKET_SIZE is set to 2000

That was the issue. I added `audioServer.setBufferSize(MQTT_MAX_PACKET_SIZE);` in MQTTDisconnected entry, and now I receive actual audio data. Earlier on I only had `#define MQTT_MAX_PACKET_SIZE 2000`, but that wasn't enough in the Arduino IDE.

Romkabouter, 2021-02-19:
Still reboots or is that issue also fixed?

flatsiedatsie, 2021-02-19:
Yes. Altought I've made a number of changes (disabled color control and OTA), I think that's what fixed it. Also added:

```
    if(!I2StaskCreated){
      device->muteOutput(true);
      xEventGroupClearBits(audioGroup, STREAM);
      xEventGroupClearBits(audioGroup, PLAY);
      xTaskCreatePinnedToCore(I2Stask, "I2Stask", 30000, NULL, 3, NULL, 0);
      I2StaskCreated = true;
    }
    else{
      Serial.println("NOT CREATING I2S TASK AGAIN");
    }
```
because I noticed that task was created multiple times, and I wasn't sure if that was an issue.

Can't speak for @hpmueller1971 though.

At the risk of further hijacking this thread: my issue now is that I can't get Snips to react to "Hey Snips". I looked at the older code, but can't find which MQTT commands are needed to work with Snips.

I've previously created satellites based on the Pi Zero, and there pointing the satellite audio server to the main server 'automatically' made it work.

koenvervloesem, 2021-02-19:
@flatsiedatsie For your Snips compatibility issues: have a look at https://rhasspy-hermes.readthedocs.io/en/latest/api.html. The documentation is not complete yet, but this should become the documentation of Rhasspy's dialect of the Snips Hermes protocol. It already should indicate which parts of the current protocol are Rhasspy-specific.

flatsiedatsie, 2021-02-19:
The reboot issue re-appears if I disable sending audio data all the time. So if `HW_REMOTE` is not active, it crashes.

```
    if (config.hotword_detection == HW_REMOTE)
    {
      //xEventGroupSetBits(audioGroup, STREAM); // <- commenting this line causes crash
      Serial.println("idle started streaming");
    }
```

flatsiedatsie, 2021-02-19:
Any concrete hints on what changed when Snips compatibility was removed would be greatly appreciated. I went over the code, but couldn't find anything concrete. Since I don't use audio playing ability, I already can't get it to simply detect the hotword on the main server.

koenvervloesem, 2021-02-19:
Sorry, I haven't touched Snips for a long time, so I can't give some concrete hints. Maybe ask on the Rhasspy forum? A couple of users there are still using Snips or have been using it very recently.

flatsiedatsie, 2021-02-19:
I suspect it has something to do with `bind`. Snips has to be told `atomecho@mqtt` somehow.

Did the old code only work if you made changes to `snips.toml` on the main server?

Romkabouter, 2021-02-19:
> because I noticed that task was created multiple times, and I wasn't sure if that was an issue.

Pretty strange behaviour, are you using an ino file or my codebranch? 
The I2Stask is created in the WifiDisconnected state, which I can see is an issue now when the state is re-entered. Good catch.

I was already investigating the HW_REMOTE and am making some changes. 

> I appreciate the suggestion Koen, but any concrete hints on what changed when Snips compatibility was removed would be greatly appreciated. I went over the code, but couldn't find anything concrete. Looking at the API won't help me either.
> 
> I don't use audio playing ability. I already can't get it to simply detect the hotword on the main server.

I think you need to check the void onMqttMessage function.
There are checks in the topics toggleOn and toggleOff, but the hotword is only triggers when a reason is "dialogueSession"
This was not the case with snips

Try changing this (on two places):
```
        if (root["siteId"] == SITEID 
          && root.containsKey("reason")
          && root["reason"] == "dialogueSession") {
            send_event(HotwordDetectedEvent());
        }
```
to this:
```
        if (root["siteId"] == SITEID) {
            send_event(HotwordDetectedEvent());
        }
```

The old code was:
```
        if (topicstr.find("toggleOff") != std::string::npos) {
            std::string payloadstr(payload);
            // Check if this is for us
            if (payloadstr.find(SITEID) != std::string::npos) {
                hotword_detected = true;
                xEventGroupSetBits(everloopGroup, EVERLOOP);  // Set the bit so the everloop gets updated
            }
        } else if (topicstr.find("toggleOn") != std::string::npos) {
            // Check if this is for us
            std::string payloadstr(payload);
            if (payloadstr.find(SITEID) != std::string::npos) {
                hotword_detected = false;
                xEventGroupSetBits(everloopGroup, EVERLOOP);  // Set the bit so the everloop gets updated
            }
        } 
```

flatsiedatsie, 2021-02-19:
> Try changing this (on two places)

I already had :-) See the [bottom of this post](https://github.com/Romkabouter/ESP32-Rhasspy-Satellite/issues/39#issuecomment-782048185) a few posts up.

indeed, I couldn't find anything other than that. Still, Snips just isn't responding to "hey snips". In your experience, does the Atom Echo require saying it near or far from the device? You you have to speak loudly?

I was wondering if I had to use "atomecho@mqtt" somewhere, but couldn't find any mention in the code anywhere. As far as I can tell, the current code is similar to the older versions.

I also tried playing with `snips.toml`, but no luck.

Romkabouter, 2021-02-20:
Ah sorry, missed that.
You need to disable the feedback sound, otherwise snips will send it and will wait for a finished message which is send after the playing of the feedback sound.
That is why you get the timeouts, check the original:
` if (topicstr.find("playBytes") !=` 
part.

This is the finished message:
`finishedMsg = "{\"id\":\"" + topicparts[4] + "\",\"siteId\":\"" + SITEID + "\",\"sessionId\":null}";`
The id is that last part of the random topicid hermes/audioServer/siteID/playBytes/<randomid>
That message should be send to hermes/audioServer/siteID/playFinished.

Also, in later version you need top implement "playBytesStreaming".

I suggest you disable the feedback to try this. I will try if I can spinup snips somewhere to help you out :)

Romkabouter, 2021-02-20:
> Hi,
> 
> i successfully installed on an Atom Echo, it seems to work (ping, webui reachable), but it keeps rebooting every few seconds in a loop when in local hotword mode.
> 
> /hp
Hi @hpmueller1971, I can reproduce this issue so I will fix it asap :)

flatsiedatsie, 2021-02-20:
@Romkabouter thanks, that might be it. Don't worry about installing Snips, since Voco already drops a lot of Snips-isms too (satellites don't play audio streams). If you do really want you help out, you could try installing Voco on Webthings on a Raspberry Pi.

Romkabouter, 2021-02-20:
I will do, intresting project :)

flatsiedatsie, 2021-02-20:
@Romkabouter Alright, curious to hear your thoughts :-)

Romkabouter, 2021-02-20:
@hpmueller1971  please check version 7.2:

https://github.com/Romkabouter/ESP32-Rhasspy-Satellite/releases/tag/v7.2

Romkabouter, 2021-02-20:
@flatsiedatsie  can you check 7.3 if you still see double creations of the I2Stask?
Should be fixed

flatsiedatsie, 2021-02-20:
I don't use PlatformIO, so I can't test that for you, sorry.

I checked my code (which is essentially 7.1 in Arduino form), and it does already have the `playBytes` response code.

https://github.com/createcandle/voco-mini-satellite/blob/a2d4d5084bbed6f9412f4fb6c37a2b082287f5c6/StateMachine.hpp#L398

Romkabouter, 2021-02-20:
Ah ok, the fixes are small so you should be able to put it in your sketch as well :)

Romkabouter, 2021-02-20:
> I checked my code (which is essentially 7.1 in Arduino form), and it does already have the `playBytes` response code.

Yes, but depending on which version of snips is used, you also need playBytesStreaming

flatsiedatsie, 2021-02-22:
But isn't `playBytesStreaming` only required to play audio files? Voco doesn't (currently) stream audio to the satellites. At the moment, the satelites (Pi Zero's) generate the speech themselves. Altough if I could get ESP32 Rhasspy Satellite to work, then adding support for this would be worth it. A satellite the size of a sugar cube is amazing.

You mentioned that it would be necessary to play the audio files that indicate the wake word was heard. When I disabled the audio response, it still wouldn't work.

> @flatsiedatsie can you check 7.3 if you still see double creations of the I2Stask?

My code already has a similar "only start the task once" check, and since then it doesn't load multiple times.

did you happen to try an installation of Voco yourself? I was hoping you'd have a look. You could probably more easily spot and fix the issue than I could.

Romkabouter, 2021-02-22:
Yes, that is only required for playing audio.
I have tried voco indeed (see my comments in the Arduino version issue)
When voco completely implements the hermes protocol, I am convinced this code will work as a satellite for it :)
Voco basically uses snips, but snips' dialogue manager waits for a finished message as fas as I can recall.
That does not arrive (because it is not sent) and therefore times out. I have not investigated any further

flatsiedatsie, 2021-02-22:
I have noticed something: wifi often crashes at boot. Currently the I2STask is created in the WifiDisconnected class. Perhaps that should wait until the wifi is connected? Otherwise streaming will attempt to start while there is no wifi yet?

Romkabouter, 2021-02-22:
The WifiDisconnected is the initial state of the statemachine. That is why it is created there.
You could indeed create it in another state, but streaming of playing is not done when STREAM or PLAY is not set.
The task does nothing but loop in that case.

flatsiedatsie, 2021-02-22:
Hmm, then why did it crash the wifi?

I got a bit further: I've managed to get "Hey Snips" detected! But only straight after booting the ESP32.

```
14:28:15.397 -> clientID = atomechoAudio
14:28:15.397 -> past asyncclient.connect
14:28:15.433 -> Wifi ready
14:28:15.433 -> end of setuppast audioServer connect, reached end of mqttdisconnected init
14:28:15.579 -> Enter MQTTConnected
14:28:15.579 -> Connected as atomecho
14:28:15.617 -> subscribed to topics
14:28:15.617 -> Enter Idle
14:28:15.617 -> Still Idle
14:28:15.617 -> idle started streaming
14:28:15.617 -> idle init complete
14:28:18.947 -> mqtt message received: hermes/hotword/azrxidia/detected
14:28:18.947 -> {"siteId":"atomecho","modelId":"hey_snips","modelVersion":"workflow-hey_snips_subww_feedback_10seeds-2018_12_04T12_13_05_evaluated_model_0002","modelType":"universal","currentSensitivity":0.5,"detectionSignalMs":-10,"endSignalMs":-10}
14:28:20.261 -> mqtt message received: hermes/hotword/toggleOff
14:28:20.261 -> {"siteId":"atomecho","sessionId":"689cb8cf-ee24-46e8-9fcd-be72fb175aab"}⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮ﾭ⸮xV⸮⸮⸮⸮⸮⸮?T⸮?⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮ﾭ⸮xV⸮⸮⸮>⸮⸮⸮?T⸮?E⸮7⸮7⸮<⸮=⸮4⸮:⸮=⸮<⸮I⸮5⸮D⸮0⸮:⸮7⸮@⸮9⸮9⸮?⸮>⸮E⸮=⸮>⸮?⸮7⸮;⸮<⸮9⸮>⸮1⸮D⸮5⸮B⸮6⸮B⸮8⸮D⸮?⸮E⸮E⸮5⸮@⸮9⸮F⸮1⸮@⸮1⸮D⸮1⸮>⸮7⸮@⸮8⸮@⸮;⸮2⸮B⸮7⸮@⸮5⸮?⸮2⸮>⸮@⸮D⸮9⸮?⸮A⸮A⸮1⸮7⸮G⸮6⸮<⸮0⸮?⸮6⸮D⸮3⸮C⸮7⸮D⸮2⸮F⸮1⸮?⸮8⸮9⸮5⸮>⸮@⸮G⸮9⸮B⸮-⸮@⸮.⸮9⸮5⸮<⸮<⸮3⸮?⸮:⸮D⸮5⸮=⸮7⸮8⸮5⸮<⸮7⸮<⸮3⸮7⸮:⸮<⸮3⸮<⸮9⸮;⸮=⸮1⸮=⸮9⸮,⸮2⸮2⸮:⸮/⸮6⸮/⸮=⸮1⸮3⸮2⸮<⸮0⸮+⸮0⸮4⸮4⸮0⸮2⸮6⸮5⸮2⸮1⸮/⸮,⸮:⸮2⸮7⸮4⸮4⸮4⸮*⸮*⸮0⸮<⸮.⸮0⸮&⸮3⸮.⸮+⸮&⸮)⸮5⸮)⸮5⸮ﾭ⸮xV⸮⸮⸮⸮⸮⸮?T⸮?
14:28:20.373 -> got toggleOff
14:28:20.373 -> -its for us
14:28:20.373 -> idle react to HotwordDetectedEvent
14:28:20.373 -> Entry into HotwordDetected class
14:28:20.462 -> HotwordDetected class init complete
14:28:34.451 -> mqtt message received: hermes/hotword/toggleOn
14:28:34.451 -> {"siteId":"atomecho","sessionId":null}
14:28:34.451 -> got toggleOn
14:28:34.451 -> -toggleOn for me
14:28:34.451 -> Hotword switching to idle
14:28:34.451 -> Enter Idle
14:28:34.451 -> Still Idle
14:28:34.667 -> idle started streaming
14:28:34.667 -> idle init complete

```

Some strange things:
- `mqtt message received: hermes/hotword/azrxidia/detected`

That says `azrxidia` instead of `atomecho`. As if Snips is giving its own name to the site. On the next line the SideID is correct though.

- I wonder what all the gibberish in the output is about.

The `HotwordDetected` class disables and then re-enables streaming. Could it be that that is causing the voice activity being seen as down right when it's necessary ot hav it?

In Snips Watch:
```
[14:28:18] [Asr] was asked to listen on site atomecho
[14:28:20] [VoiceActivity] Down on site atomecho
[14:28:34] [Dialogue] session with id '689cb8cf-ee24-46e8-9fcd-be72fb175aab' was ended on site atomecho. The session was ended because one of the component didn't respond in a timely manner
```

The `hotwordDetected init`:
```
class HotwordDetected : public StateMachine
{
  void entry(void) override {
    Serial.println("Entry into HotwordDetected class");
    xEventGroupClearBits(audioGroup, PLAY);
    xEventGroupClearBits(audioGroup, STREAM);
    //device->updateBrightness(config.hotword_brightness);
    if (xSemaphoreTake(wbSemaphore, (TickType_t)10000) == pdTRUE) {
      //device->updateColors(COLORS_HOTWORD);
    }
    xSemaphoreGive(wbSemaphore);
    initHeader(device->readSize, device->width, device->rate);
    xEventGroupSetBits(audioGroup, STREAM);
    Serial.println("HotwordDetected class init complete");
  }
```

If I press the button to start a session, then things look a little different in Snips Watch:
```
[14:38:16] [Dialogue] session with id '5d60928f-dd37-4981-b124-56f2384626db' was started on site atomecho
[14:38:16] [Asr] was asked to listen on site atomecho
[14:38:18] [VoiceActivity] Up on site atomecho
[14:38:21] [VoiceActivity] Down on site atomecho
[14:38:23] [VoiceActivity] Up on site atomecho
[14:38:25] [VoiceActivity] Down on site atomecho
[14:38:27] [VoiceActivity] Up on site atomecho
[14:38:29] [VoiceActivity] Down on site atomecho
[14:38:32] [Dialogue] session with id '5d60928f-dd37-4981-b124-56f2384626db' was ended on site atomecho. The session was ended because one of the component didn't respond in a timely manner
[14:38:32] [Asr] was asked to stop listening on site atomecho
```
Here is seem that voice activity is up.. but it's perhaps not recognised as speech by the ASR?

Romkabouter, 2021-02-22:
Can you please create a new defect for this? The original defect is something else

closed 2021-02-22 by Romkabouter

flatsiedatsie, 2021-02-22:
It seems we were on the same track with the need for a delay to start the I2STask.

I'll open a new issue, good call.