Romkabouter/ESP32-Rhasspy-Satellite #40: Adding a video stream...

flatsiedatsie, 2021-02-17:
I'm exploring if it's possible to create a privacy friendly smart doorbell  / security camera based on an ESP32. Since I'm a fan of Rhasspy, I thought perhaps going the MQTT route isn't such a bad idea. In that case `ESP32-Rhasspy-Satellite` could perhaps form a basis for my experiment.

Do you think that once in a while streaming an additional JPG over MQTT back to the main server could work? Or would that just be too much for the ESP32? Is there any processing power or memory left?

Romkabouter, 2021-02-17:
Nice idea, device I own already run a bit high on temperature but I have not checked into CPU usage.
It might be possible, I guess it would be trail and error

That being said, when local hotword is hopefully available from WakeNet or Porcupine, this most certainly will drop CPU usage due to that fact that no audio is published in Idle state. In that case I think it will be plausable to have such a feature :)

Romkabouter, 2021-02-21:
Hi @flatsiedatsie, I think your idea should be possible. You can fork this repo and try it :)
I do not have an esp32 camera, so I can't help you with this now. Sorry about that.

closed 2021-02-21 by Romkabouter

flatsiedatsie, 2021-02-22:
Turn out there aren't really any ESP32 board that allow for..
- camera
- microphone
- speaker
..at the same time.

Romkabouter, 2021-02-22:
That is why I have the I2Stask, either playing or streaming. Or nothing.
I have not investigated, but quickly switch between camera/mic might be doable? Or is the same pin used?

flatsiedatsie, 2021-02-22:
The DAC pins on the ESP32 (pins 25 and 26) are also used by camera modules.

Which is probably why I couldn't find any ESP32 boards that had a camera and a speaker.

The TTGO with cam (white) has both a camera and a MEMS microphone. But it's only available remaining pins are I2C. So audio out would have to go via I2C. This is theoretically possible, but requires more hardware.

Romkabouter, 2021-02-22:
Maybe this can inspire you ;)

https://community.home-assistant.io/t/diy-video-doorbell-with-voice-response-esphome-esp32-camera-dfplayer-mini/208254/1

flatsiedatsie, 2021-02-22:
Wow! Yes that's very inspiring. Thank you!