Romkabouter/ESP32-Rhasspy-Satellite #42: Attempt to get it to work with Voco

flatsiedatsie, 2021-02-22:
This is a continuation of discussion [here](https://github.com/Romkabouter/ESP32-Rhasspy-Satellite/issues/39).

I've managed to get Snips to recognise the wake-word, but only right after booting the Atom Echo.

Snips does recognise that there is audio input.

While in idle mode, Snips Watch indicates that audio is being heard.
```
[14:55:18] [VoiceActivity] Down on site atomecho
[14:55:21] [VoiceActivity] Up on site atomecho
[14:55:22] [VoiceActivity] Down on site atomecho
[14:55:35] [VoiceActivity] Up on site atomecho
[14:55:53] [VoiceActivity] Down on site atomecho
[14:55:55] [VoiceActivity] Up on site atomecho
[14:55:56] [VoiceActivity] Down on site atomecho
[14:56:00] [VoiceActivity] Up on site atomecho
```

If I press the button to start a session, a session is created, and the dialogue manager listens to the stream from the Atom Echo. But the voice input is not recognised as a voice command:
```
[14:56:10] [Dialogue] was asked to start a session on site atomecho
[14:56:10] [Asr] was asked to stop listening on site atomecho
[14:56:10] [Hotword] was asked to toggle itself 'off' on site atomecho
[14:56:10] [Dialogue] session with id 'fe685174-5053-4651-8756-8cb3b066003e' was started on site atomecho
[14:56:10] [Asr] was asked to listen on site atomecho
[14:56:11] [VoiceActivity] Down on site atomecho
[14:56:12] [VoiceActivity] Up on site atomecho
[14:56:15] [VoiceActivity] Down on site atomecho
[14:56:16] [VoiceActivity] Up on site atomecho
[14:56:18] [VoiceActivity] Down on site atomecho
[14:56:19] [VoiceActivity] Up on site atomecho
[14:56:26] [Dialogue] session with id 'fe685174-5053-4651-8756-8cb3b066003e' was ended on site atomecho. The session was ended because one of the component didn't respond in a timely manner
[14:56:26] [Asr] was asked to stop listening on site atomecho
[14:56:26] [Hotword] was asked to toggle itself 'on' on site atomecho
```

If I don't speak into the Raspberry Pi version, then things look a bit different.
```
[15:07:45] [VoiceActivity] Up on site azrxidia
[15:07:46] [Hotword] detected on site azrxidia, for model hey_snips
[15:07:46] [Asr] was asked to stop listening on site azrxidia
[15:07:46] [Hotword] was asked to toggle itself 'off' on site azrxidia
[15:07:46] [Dialogue] session with id '16427483-191a-4c39-9f0b-199dd4cb0e7e' was started on site azrxidia
[15:07:46] [Asr] was asked to listen on site azrxidia
[15:07:46] [VoiceActivity] Up on site atomecho
[15:07:47] [VoiceActivity] Down on site azrxidia
[15:07:50] [Asr] captured text "" in 4.0s
[15:07:50] [Asr] was asked to stop listening on site azrxidia
[15:07:50] [Dialogue] session with id '16427483-191a-4c39-9f0b-199dd4cb0e7e' was ended on site azrxidia. The session was ended because the platform didn't understand the user
[15:07:50] [Asr] was asked to stop listening on site azrxidia
```
So what would support the idea that some MQTT message is missing.

As an aside, I also noticed that the wave header is slightly different:

ESP32
```
RIFF,WAVEfmt ?>}datay?u?y?t?z?v?|?~?q?v?u??s?z?u?y?t?v?v?w?y?q?o?u?|?x?y?n?t?t?u?r?r?v?s?x?q?w?s?x?m?v?s?o?t?s?y?s?w?u?o?~?w?s?t?s?u?y?~?s?~?t?u?{Հ?x?u?u?u?}?xՁ?y?x?x?zՀ?s?u?s?v?z?z?z?p?n?n?w?r?v?z?p?t?r?q?w?|?x?u?o?q?|?y?y?t?t?o?~?}?y?w?p?w?}?~?t?v?v?z?{?{?{?z?u?xՌՋ?~?}?w?yՄՅՂ?{?{?yՂՈ?|Մ?v?~ՂՃ?Հ?|?wՁ?Ղ?z?|?|Յ?ՂՀ?|?}Ղ???|?~Հ?~Մ?y?zՀՈՃՁՂ?}ՆՋ?Ձ?zՄՂՈՁ?Մ?Ո?ՇՂՃՂ?}?}??|ՅՃՄՄՃ?|Ձ?~ՊՏ?}ՂՀՅՉՅՇՂ?~?yՊՄՄՁ?x?~Ձ??~?{?~?zՂ?{?y?~?}?{?|?|?y?z??~?{Ղ?qՀ?uՂ?t?w?|?z?~?x?{?
```

USB microphone on Raspberry Pi:
```
RIFF4WAVEfmt ?>}tim??wdata????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????????#")%##0$*%%AAHNMVZPHNWALKGC))????????!
???????/!>@>$$2*8@1-?>GIKB:ECV=3/3<19C?=5H=80)& 
                                                ) <@5/,?23JH;
```

I also check the output of the various commands in Mosquitto to find out which exact message was missing.

```
[... 40 lines cut ...]
...

hermes/voiceActivity/atomecho/vadDown æ"siteId":"atomecho","signalMs":-386å
```

----Atom Echo hotword detected----
```
hermes/hotword/azrxidia/detected æ"siteId":"atomecho","modelId":"hey_snips","modelVersion":"workflow-hey_snips_subww_feedback_10seeds-2018_12_04T12_13_05_evaluated_model_0002","modelType":"universal","currentSensitivity":0.5,"detectionSignalMs":-70,"endSignalMs":-70å

hermes/voco/atomecho/play æ"sound_file": "start_of_input"å

hermes/audioServer/atomecho/audioFrame RIFF,WAVEfmt ?>ådata?????????ؽ??????????????????????????????????????????????ؽ????????غ????ؿؽ????غؿؾػؾؿ??ع??????ظ????????ؾ????ؾؾؾ????????ؽ??ؿ????ظ??????صطؽؼػغغ????????ؼ??ضؼؼشؾغ????ع??ؼ??ؼ??ظؼر??ػعػظ??ؾؼ??ذضظص??زؾشؼطؾظؽؾطؽخرخصؼذزشسزظخعػش??زطزػغطضضجزعج??ضرذ??غؼحذظ??ظرش??ؼذطصثزضص??سشش??رشص??عسصح??عظخؼجؽزػرخرشصشذظصحطظشغغصشسسسذسصغزضطشعغص??ؽظؿظرظطؼصؿ??عصغؼؽطػص???
hermes/asr/stopListening æ"siteId":"atomecho","sessionId":"a9e2c85d-cdfd-40ce-9ad6-ad30c9ed2868"å
hermes/audioServer/atomecho/audioFrame RIFF,WAVEfmt ?>ådata???غ??????ظؿؾ????ؼؾظغطغ??ظؿع??ط??ؾع??ططظعشظغ??ظؽعظؼطضسطضؼشرطسعسطظغؽؿ????صعصغ??غؿؼظط????????ػؾؽؼ????ؼ??صػظػظؾظػػععؼػ??ؾ??سؿشغػششرؽخغشػؽزؼضزسض??ضصض??سظػػرؽعؼػعظص??صغسؼؼػظؼظغؼؼظ????ظػ??ؿعغعظظ??ػع??ظحؽش??ضظعػطصذ??ظؿؽظطصؽصؽشغسسط??صؾؿغعغؽؽضعغ??ػػؽؽ??ؼص??ظدضعضصضغظخصػطؽضػتػشظرعصصدذشطزطخخظظذزظضصؾد??حرخ?
hermes/hotword/toggleOff æ"siteId":"atomecho","sessionId":"a9e2c85d-cdfd-40ce-9ad6-ad30c9ed2868"å

hermes/dialogueManager/sessionStarted æ"sessionId":"a9e2c85d-cdfd-40ce-9ad6-ad30c9ed2868","customData":null,"siteId":"atomecho","reactivatedFromSessionId":nullå

hermes/asr/startListening æ"siteId":"atomecho","sessionId":"a9e2c85d-cdfd-40ce-9ad6-ad30c9ed2868","startSignalMs":-70å

hermes/voco/atomecho/mute æ"mute": trueå

hermes/voiceActivity/atomecho/vadDown {"siteId":"atomecho","signalMs":-386}

```

Romkabouter, 2021-02-22:
What is the samplerate of voco?
This software output 16000 16bit mono, each MQTT messsage containt 512 bytes wavedata and 44 but header, a total of 556 bytes per message.

And this software does not support snips anymore, which version of snips does voco use?

flatsiedatsie, 2021-02-22:
That would rock! I was looking into the header to see if I had to change something there.

As far as I know, Voco uses the latest version of Snips.

Romkabouter, 2021-02-22:
Where can I find this "VoiceActivity" logs on such?

The audio produced is just a lot of small wave files. Rhasspy does not differ from snips with regards to that.
But the chunksize from Voco is different than what is send by the streamer

This is the wave format: http://soundfile.sapp.org/doc/WaveFormat/
From Voco: RIFF4WAVEfmt
From streamer: RIFF,WAVEfmt

The 4 and , are not a 4 and a , but represent the 4 bytes of the ChunkSize. 
So if the chunk size in not the same then depending on how Voco is implemented, it might deal with the audio unexpectedly

Romkabouter, 2021-02-22:
I will use this branch https://github.com/Romkabouter/ESP32-Rhasspy-Satellite/tree/voco

flatsiedatsie, 2021-02-23:
> The 4 and , are not a 4 and a , but represent the 4 bytes of the ChunkSize.

Ah, interesting. I figured it might matter. I also wonder if the `tim` part was some kind of time index.

To debug, I use these commands:

Snips Watch: `~/.webthings/addons/voco/snips/snips-watch -vvvv`

Looking at ALL MQTT traffic: `mosquitto_sub -v -h 192.168.2.167 -t 'hermes/#'`
(mosquitto sub isn't installed by default, so `sudo apt-get install mosquitto-clients` will fix that)

And if you need to quickly restart the gateway for some reason: `sudo systemctl restart webthings-gateway.service`

Romkabouter, 2021-02-23:
> Ah, interesting. I figured it might matter. I also wonder if the `tim` part was some kind of time index.

Yes, when I see this:
hermes/audioServer/azrxidia/replayResponse RIFF^WAVEfmt ?>}tim??wrpidazrxidia-1614003432719rprf
that does not really look like a good waveheader to me, it should somewhere contain the plain letter data as well
The link I posted was how a wave format header should look like, I do not know how the replayRespoonse is generated.

In the streamer I use a fixed header, because every wav audio send is the same length and format :)
I will ignore the replayResponse for now and will dive into voco a bit to find out what voco expects.

flatsiedatsie, 2021-02-23:
Yes I've only seen that `replayRequest` command once. It's not documented either, all I could find were two references to Snips code on Github. I'd just ignore it.

> that does not really look like a good waveheader to me

0_0 :-)

Romkabouter, 2021-03-01:
If have tried a bare snips with the demo assistant with a mic attached to the pi, I get timeouts:

[image]

I think Snips is broken and this is actually causing the timeouts, I will try this setup with voco as well.

flatsiedatsie, 2021-03-01:
AI, that doesn't sound good...

Romkabouter, 2021-03-01:
No, this is Snips installed on a Pi with an attached mic. Nothing to do with this code.
I wanted to see which message are going back and forth to snips to see what is needed for a correct Dialogue session.

That is why I installed the demo assistant.
I followed https://docs.snips.ai/articles/raspberrypi/manual-setup, which is old but still the latest version :)

flatsiedatsie, 2021-03-01:
Keeping my fingers crossed over here :-)

Romkabouter, 2021-03-01:
I followed this step and now snips is working :)

https://github.com/snipsco/snips-issues/issues/161

Romkabouter, 2021-03-01:
progress :)

[image]

flatsiedatsie, 2021-03-02:
Hey wow!! Great work!

What's still left to do? It looks to me like a 100% succes?

Romkabouter, 2021-03-02:
This was a test with a Pi with a build-in mike.
To verify that snips still works.

The next step for me is to check and adjust the code for the streamer to get Snips going (again). 
When that works, I can move on to Voco :)
First with a build-in mike and if that works the streamer. So there is a road ahead :)

flatsiedatsie, 2021-03-02:
Ah I see. Are you sure the effort is worth it? We could just wait for Voco to move to Rhasspy. That has to happen at some point anyway.

Romkabouter, 2021-03-02:
Well, it would be nice to find what the issue is, but that is just me :D

I see in the messages RIFF4, as you also already found. The bytes send per message is also 572 instead of the 556 that the streamer sends.
I believe this is the root cause of the bad performance, the wave audio does not match.

flatsiedatsie, 2021-03-02:
That's what I figured as well.

Romkabouter, 2021-03-02:
Got it :)

https://github.com/snipsco/hermes-protocol/blob/develop/hermes/src/ontology/audio_server.rs#L25..L87

Romkabouter, 2021-05-13:
Sorry about my lack of commitment, my sd card died so I had to start over which I did not have time for.
This is still on my to-do list however :)

flatsiedatsie, 2021-05-13:
No worries. Pretty busy overhere as well :-)

Romkabouter, 2021-05-23:
I have the code working with snips, however there are some header changes which I can't get right atm.
When hey snips is spoken, it works if you speak it a bit slower, so it is most likely a wave format thing.
I have recorder an audio file which triggers the hotword 100% with the code. I will attach that to the branch :)

flatsiedatsie, 2021-05-23:
That's great news! I'm going to check this out asap!

Romkabouter, 2021-05-23:
This is the result of the audio file
[image]

After the hotword it picks up the audio as expected
[image]

The timeout occurs because I have no intent handler.

flatsiedatsie, 2021-05-27:
Where can I find the code? I had a look at the voco branch, but that doesn't seem to be it?

Romkabouter, 2021-05-27:
Just pushed it. I have also included a record.py which records a stream for a couple of seconds.
I was experimenting with the header, but found the I did not have to make a change strange enough.

Just pushed branch works with met Atom Echo, but the recordlevel seems to be very low. This will also be the case with the master branch I assume

flatsiedatsie, 2021-05-27:
Cool, I will try it now!

flatsiedatsie, 2021-05-27:
It took a bit of work to be able to upload it via the Arduino IDE again. I had to strip out the LED parts.

Now that it uploads, I get this error. Nothing to worry about, I just have to look into it.

```
22:17:17.029 -> Creating I2Stask
22:17:17.029 -> Enter WifiDisconnected
22:17:17.029 -> Total heap: 293476
22:17:17.029 -> Free heap: 231364
22:17:19.401 -> Enter WifiConnected
22:17:19.401 -> Connected to Wifi with IP: 192.168.2.137, SSID: sterrenkijker_nomap, BSSID: B0:95:75:9F:FE:CF, RSSI: -49
22:17:19.438 -> Enter MQTTDisconnected
22:17:19.438 -> Connecting MQTT: 192.168.2.166, 1883
22:17:19.475 -> end of setupEnter MQTTConnected
22:17:19.586 -> Connected as atomecho
22:17:19.586 -> Enter Idle
22:17:19.586 -> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
22:17:19.586 -> Core 0 register dump:
22:17:19.586 -> PC      : 0x4015618e  PS      : 0x00060b30  A0      : 0x800d11ec  A1      : 0x3ffde570  
22:17:19.586 -> A2      : 0x00000000  A3      : 0x3ffde5e0  A4      : 0x00000200  A5      : 0x3ffde5ac  
22:17:19.623 -> A6      : 0x00000064  A7      : 0x00000000  A8      : 0x00000001  A9      : 0x0000000b  
22:17:19.623 -> A10     : 0x3ffb96ac  A11     : 0x3ffde58f  A12     : 0x00000000  A13     : 0x00000008  
22:17:19.623 -> A14     : 0x00060b23  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
22:17:19.623 -> EXCVADDR: 0x00000014  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
22:17:19.660 -> 
22:17:19.660 -> ELF file SHA256: 0000000000000000
22:17:19.660 -> 
22:17:19.660 -> Backtrace: 0x4015618e:0x3ffde570 0x400d11e9:0x3ffde5a0 0x400d2174:0x3ffde5d0 0x40089bce:0x3ffdea50
22:17:19.660 -> 
22:17:19.660 -> Rebooting...
```

flatsiedatsie, 2021-05-27:
Got a bit further.

Should I change some of the settings to get it to continuously stream audio? I'm assuming I should not use hotward detection.

The `Connected as null` is a bit strange. I tried adding `siteid` to the config file, but that didn't seem to solve it.

```
22:41:39.027 -> Rebooting...
22:41:39.027 -> ets Jun  8 2016 00:22:57
22:41:39.027 -> 
22:41:39.027 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
22:41:39.027 -> configsip: 188777542, SPIWP:0xee
22:41:39.027 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
22:41:39.027 -> mode:DIO, clock div:1
22:41:39.027 -> load:0x3fff0018,len:4
22:41:39.061 -> load:0x3fff001c,len:1044
22:41:39.061 -> load:0x40078000,len:10124
22:41:39.061 -> load:0x40080400,len:5856
22:41:39.061 -> entry 0x400806a8
22:41:39.386 -> Boo⸮M5Atom initializing...Loading configuration
22:41:39.574 -> {
22:41:39.574 ->   "mqtt_host": "192.168.2.166",
22:41:39.574 ->   "mqtt_port": 1883,
22:41:39.574 ->   "mqtt_user": "",
22:41:39.574 ->   "mqtt_pass": "",
22:41:39.574 ->   "sideid": "atomecho",
22:41:39.574 ->   "mqtt_valid": true,
22:41:39.574 ->   "mute_input": false,
22:41:39.574 ->   "mute_output": false,
22:41:39.574 ->   "amp_output": 0,
22:41:39.574 ->   "brightness": 30,
22:41:39.574 ->   "hotword_brightness": 100,
22:41:39.574 ->   "hotword_detection": 0,
22:41:39.574 ->   "volume": 100,
22:41:39.611 ->   "gain": 5
22:41:39.611 -> }
22:41:39.611 -> Creating I2Stask
22:41:39.611 -> Enter WifiDisconnected
22:41:39.611 -> Total heap: 293084
22:41:39.611 -> Free heap: 230916
22:41:43.910 -> Enter WifiConnected
22:41:43.910 -> Connected to Wifi with IP: 192.168.2.137, SSID: sterrenkijker_nomap, BSSID: B0:95:75:9F:FE:CF, RSSI: -50
22:41:43.945 -> Enter MQTTDisconnected
22:41:43.945 -> Connecting MQTT: 192.168.2.166, 1883
22:41:43.945 -> end of setupConnect failed, retry
22:41:48.932 -> Audio connected: 1, Async connected: 0
22:41:48.932 -> Enter MQTTDisconnected
22:41:48.932 -> Connecting MQTT: 192.168.2.166, 1883
22:42:20.105 -> Connect failed, retry
22:42:20.105 -> Audio connected: 0, Async connected: 0
22:42:20.105 -> Enter MQTTDisconnected
22:42:20.105 -> Connecting MQTT: 192.168.2.166, 1883
22:42:38.203 -> Connect failed, retry
22:42:38.203 -> Audio connected: 0, Async connected: 0
22:42:38.203 -> Enter MQTTDisconnected
22:42:38.240 -> Connecting MQTT: 192.168.2.166, 1883
22:42:56.724 -> Connect failed, retry
22:42:56.724 -> Audio connected: 0, Async connected: 0
22:42:56.724 -> Enter MQTTDisconnected
22:42:56.724 -> Connecting MQTT: 192.168.2.166, 1883
22:43:15.229 -> Connect failed, retry
22:43:15.229 -> Audio connected: 0, Async connected: 0
22:43:15.229 -> Enter MQTTDisconnected
22:43:15.229 -> Connecting MQTT: 192.168.2.166, 1883
22:43:33.736 -> Connect failed, retry
22:43:33.736 -> Audio connected: 0, Async connected: 0
22:43:33.772 -> Enter MQTTDisconnected
22:43:33.772 -> Connecting MQTT: 192.168.2.166, 1883
22:43:52.241 -> Connect failed, retry
22:43:52.279 -> Audio connected: 0, Async connected: 0
22:43:52.279 -> Enter MQTTDisconnected
22:43:52.279 -> Connecting MQTT: 192.168.2.166, 1883
22:44:02.380 -> Connect failed, retry
22:44:02.380 -> Audio connected: 1, Async connected: 0
22:44:02.380 -> Enter MQTTDisconnected
22:44:02.380 -> Connecting MQTT: 192.168.2.166, 1883
22:44:02.492 -> Enter MQTTConnected
22:44:02.492 -> Connected as null
22:44:02.492 -> going from mqtt connected to idle
22:44:02.492 -> Enter Idle
22:44:02.492 -> still in idle
22:44:02.492 -> end of idle
22:44:02.492 -> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
22:44:02.492 -> Core 0 register dump:
22:44:02.492 -> PC      : 0x40157776  PS      : 0x00060d30  A0      : 0x800d161c  A1      : 0x3ffde570  
22:44:02.492 -> A2      : 0x00000000  A3      : 0x3ffde5e0  A4      : 0x00000200  A5      : 0x3ffde5ac  
22:44:02.525 -> A6      : 0x00000064  A7      : 0x00000000  A8      : 0x0000000b  A9      : 0x00000068  
22:44:02.525 -> A10     : 0x3ffb20d0  A11     : 0x3ffde58f  A12     : 0x00000000  A13     : 0x00000008  
22:44:02.525 -> A14     : 0x00060d23  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
22:44:02.525 -> EXCVADDR: 0x00000014  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
22:44:02.525 -> 
22:44:02.560 -> ELF file SHA256: 0000000000000000
22:44:02.560 -> 

```

flatsiedatsie, 2021-05-27:
In case you're curious, I've uploaded my current code here. I will look into it more this weekend.
https://github.com/flatsiedatsie/voco_mini_sat

Romkabouter, 2021-05-27:
> Should I change some of the settings to get it to continuously stream audio?

Yes, the connected as null seems to imply that the config.siteid is empty.
You should be able to browse to 192.168.2.137 and check the siteid.

Romkabouter, 2021-05-27:
You have "sideid": "atomecho", in your config, that is why it is not filled most probably.
Should be siteid

flatsiedatsie, 2021-05-27:
> You have "sideid": "atomecho", in your config, that is why it is not filled most probably.

I added that after I noticed that the value was null.

I didn't realise I should visit the device's webpage.

Romkabouter, 2021-05-28:
It says Audio connected: 1, Async connected: 0

This indicates that the streamer audio is connected, but the asynch client has a problem. I cannot tell why

flatsiedatsie, 2021-05-28:
What does the async client do? Is that the webserver?

Could this setting be related?

`#define CONFIG_ASYNC_TCP_RUNNING_CORE 1`

flatsiedatsie, 2021-05-28:
Maybe I have to transplant some more settings from [this file](https://github.com/Romkabouter/ESP32-Rhasspy-Satellite/blob/c276acc99110da029842e4609a5ab0637f4f2fff/PlatformIO/load_settings.py) into the Arduino file.

Romkabouter, 2021-05-30:
> What does the async client do? Is that the webserver?
> 
> Could this setting be related?
> 
> `#define CONFIG_ASYNC_TCP_RUNNING_CORE 1`

Absolutely, by  default that ask runs on core 0. Where wifi and other tasks run as well.

flatsiedatsie, 2021-05-31:
So `I2Stask` should run on core 1?

In the code I see it running on core 0. I changed it to 1 here:

```
    if (i2sHandle == NULL) {
      Serial.println("Creating I2Stask");
      xTaskCreatePinnedToCore(I2Stask, "I2Stask", 30000, NULL, 3, &i2sHandle, 1); // core was 0, changed it to 1
    } else {
      Serial.println("We already have a I2Stask");
    }
```

Romkabouter, 2021-05-31:
No, the I2Stask should run on core 1.
The MQTT task should run on core 1, but that is used internally be that client. Therefore you should define it.

flatsiedatsie, 2021-05-31:
Removing `#define CONFIG_ASYNC_TCP_RUNNING_CORE 1` made it much more stable.

However, it seemed to run even better with `I2Stask` running on core 1 :-D

For my understanding:
- I2Stask is the audio streamer? It core 0 reserved for it? And all the other things run on core 1?
- Is the voco code currently set to permanently stream audio to the main controller after booting up? In theory the main controller could then detect a hotword in the stream. It currently can't get it to detect the hotword, so perhaps it's not streaming?
- What happens/should happen if I press the Atom Echo's button? Currently it triggers a "hotword detected". I've tried to then speak a command, but it hasn't recognised this yet.

flatsiedatsie, 2021-05-31:
Voco is receiving something if I press the button. If memory serves, this shows that is was ready to receive audio, but it didn't catch any intent, and then switches back to normal mode.

```
2021-05-31 18:38:00.197 INFO   : voco: MQTT message to topic hermes/hotword/toggleOn received on: azrxidia a.k.a. hostname thuis
2021-05-31 18:38:00.197 INFO   : voco: +
2021-05-31 18:38:00.198 INFO   : voco: {"siteId":"ATOMECHO","sessionId":null}
2021-05-31 18:38:00.198 INFO   : voco: +
2021-05-31 18:38:00.199 INFO   : voco: In unmute. current_control_name: Headphone
2021-05-31 18:38:00.199 INFO   : voco: No intent received
2021-05-31 18:38:00.200 INFO   : voco: siteId was in /toggleOn payload: ATOMECHO
```

I can also get the Echo to receive the response from Voco to play a sound (which it doesn't do yet).

```
18:37:17.020 -> Connected as ATOMECHO
18:37:17.020 -> Connected to asynch MQTT!
18:37:17.020 -> going from mqtt connected to idle
18:37:17.020 -> Enter Idle
18:37:17.020 -> still in idle
18:37:17.020 -> end of idle
18:37:25.252 -> Incoming MQTT message. Topic: hermes/hotword/toggleOff
18:37:25.252 -> toggleOff message was for us
18:37:25.252 -> SessionId in toggleOff
18:37:25.252 -> Hotword detected event
18:37:25.252 -> Enter HotwordDetected
18:37:50.999 -> Incoming MQTT message. Topic: hermes/hotword/toggleOn
18:37:50.999 -> toggleOn message was for us. Going to idle mode.
18:37:50.999 -> Enter Idle
18:37:50.999 -> still in idle
18:37:50.999 -> end of idleEnter MQTTDisconnected
18:37:50.999 -> Incoming MQTT message. Topic: 
18:37:51.032 -> hermes/voco/ATOMECHO/play
18:37:51.032 -> Connecting MQTT: 192.168.2.166, 1883
18:38:01.011 -> Connect failed, retry
```

Romkabouter, 2021-05-31:
> * I2Stask is the audio streamer? It core 0 reserved for it? And all the other things run on core 1?
Yes

> * Is the voco code currently set to permanently stream audio to the main controller after booting up? In theory the main controller could then detect a hotword in the stream. It currently can't get it to detect the hotword, so perhaps it's not streaming?
Yes, it is a permanent stream. You can check it by connecting to the broker and subscribe to hermes/audioServer/#

> * What happens/should happen if I press the Atom Echo's button? Currently it triggers a "hotword detected". I've tried to then speak a command, but it hasn't recognised this yet.
The button triggers the hotword, so it changes state. If there is no stream, no command will be caught.

Based on 2 and 3, I think your code is not streaming.

Romkabouter, 2021-05-31:
I see you are using 18:37:51.032 -> hermes/voco/ATOMECHO/play.
Did you change the topics in General.hpp as well?

flatsiedatsie, 2021-05-31:
No I left all that as it was. I only added that one extra subscription to see if it was getting anything in return. And it was. So MQTT seemed to be working.

Romkabouter, 2021-06-03:
Do you see any messages on the audioFrame topic? hermes/audioServer/atomecho/audioFrame.

I don't know it topics are case sensitive, you might want to try all lowercase just to be sure.

flatsiedatsie, 2021-06-03:
Coïncidentally I was just checking this. I don't seen any data arrive on `mosquitto_sub -t 'hermes/audioServer/ATOMECHO/#'`

flatsiedatsie, 2021-06-03:
I managed to receive audio! I modified the PubSubClient to have a larger buffer.

flatsiedatsie, 2021-06-03:
```
hermes/audioServer/azrxidia/audioFrame RIFF4WAVEfmt ?>}timc???ydata?????????j?F?\?Q?`???????????????????????!?-?3?_???y?q?????????
                                                                                                                                  #JkpgQSL4????????????????X?5????????x??{?????????????%?=?U?q???????????????	 ????????|?[?F?=?7????
                                                              ???????????????????????(?7?7?Y?n???????????????1J┌????????????????┬±D=5???????????????????????┴???????????????????#L┼─??????????
??????⎺─│≥┬V[??????????
                       ? <␉┌°⎻␍S@0=G▒⎺␉Y␤.????␉A??????????
␤␊⎼└␊⎽/▒┤␍␋⎺S␊⎼┴␊⎼/▒≥⎼│␋␍␋▒/▒┤␍␋⎺F⎼▒└␊ RIFF4WAVE°└├ ?>£├␋└└???≤␍▒├▒??????│?▒?┴?????????????????????????)4,O␍Y??????┤NM<7??????$%#EXSH=>+????????????
????????????≥?▒?▒?±?E?N?T?_?┴?⎼?┌?[?.?"??????????????????????7?I?@?\????????@▒?????'4D[±␌V▒??⎺±QQC>"????±.??????????????????????≥?????≠???????????????????????1G7$
                                                                                                                                                                  ????????????
 17
   -T␊┴?????????????
                    !<<@8?H␌XX─┌┼├\?????┴┼IE@  ??
␤␊⎼└␊⎽/??????????vbWytlP4ELZXiv|????????????????????lD@0/4????≤␍0-2'?????????????????????⎼????????????K°⎺??????????
????%=\{???????????(7L8@KB????cBDN(	??????????????u?b?P?V?K?L?5?+?%?#? ?.?*?<?S??o?Q?e?h?T?"??
?!?5?H?i?e?Q?`?d?w?f?l?????????w?e?^?W?]?U?g?h?[?H?D?,?*??!?:?8?>?,?O?S?d?t???????????u?????????????????????????????j?f?K?k?J?0?
                                                                                                                                ?
                                                                                                                                 ?????????????j?`?a?2?/?*?
```

And the Atom Echo's output
```
hermes/audioServer/ATOMECHO/audioFrame RIFF,WAVEfmt ?>}data?!?!?&?&?"?)?+????	???&?!?3??4?$?/?*?@?<?9?G?J?N???M?D?M?G???E???<?=?3?B?2?<?/?0?&?&?????? ?%??"?#?)?(?#??&?"?%?)?.?0?7?+?<?2?<?7?B???A?7?6?9?1?<?2?6?/?9?/?9?-?2?%? ?#? ?!? ?%?'?(?%?+?'?.?(?*?*?1?2?2?4?<?/?B?C?H?D?L?O?X?R?V?Q?S?T?Q?Q?O?[?[?V?U?O?N?P?G?I?G???D?8?;?-?4?0?(?+?&?+??,??,??!?#?!??*??'??%????#??-??0??0?'?0?4?4?3?-?5?0?1?/?6?0?0?4?/?,?0?-?&?&?$?&?&?%?!?-?%?+?'?,?.?-?,?1?-?8?&?6?4?9?-?<?A?8?D?N?L?I?I?Q?Q?E?F?H?>?=?>?:?@?9?>?,?9?/?:?&?)?'?&?!??$?????	????????
hermes/audioServer/azrxidia/audioFrame RIFF4WAVEfmt ?>}tim????ydata??
                                                                     ?*?3?Y?d?J?m?{?l?p?~???????????????%?F?3?[?p?p?????????????@FI+????????????????????????????????????????????????????????????????????????~?k?g?\?g?T?T?b?Q?[?l?g?R?T?V?G?N?X?l?n????~?|?~???????????????3Gdv?????????????????????kV^K/D?RMB+??????????????????????????????????????????????≥?┼?T?I?A?5?7?'?? ??
??&?$?6?'?6?3?(?,?(?????????????????????????????????????????????????(?/?G?F?<?3?(??1???
                                                                                       ????????????????⎻?D?>?H?Z?┐?π?┤?⎻?
␤␊⎼└␊⎽/▒┤␍␋⎺S␊⎼┴␊⎼/ATOMECHO/▒┤␍␋⎺F⎼▒└␊ RIFF,WAVE°└├ ?>£␍▒├▒(?!?%?,?1?4?9?9?@?8???<?D???N?P?U?W?◆?Q?[?R?]?T?P?[?Y?]?W?V?P?O?O?S?G?L?L?I?D?K?C?>???B?D?E?C?@?@?A???=?E?=???;?????@?9?@?>?C?B?E?D?J?F?F?@?D???=?>?@?@?6???9?D?9?5?:?8?5?%?/?$?-?.?)?/?-?/?'?)?/?5?3?1?8?2?6?>?E?;?E?C?G?M?J?I?O?G?W?G?P?I?O?U?M?N?R?L?J?E?C???5?C?1?B?@?4?5?3?=?8?-?:?/?4?)?+?)?,?(?5?:?*?5?,?5?*?)?6?8?4?:?7?@?5?F?C?E?J?D?L?H?V?K?P?L?V?V?J?N?N?U?O?S?M?O?G?E?H?G?H?I?D?L?=?E?<?G?I?C?J?>?G?F?K?T?I?U?O?S?T?W?Q?V?Y?P?]?\?␉?^?▒?◆?Y?]?X?Y?X?X?[?^?_?P?]?T?T?J?Q?Y?R?R?P?P?P?Q?O?M?N?R?Q?M?V?R?[?S?]?\?T?W?T?
␤␊⎼└␊⎽/▒┤␍␋⎺S␊⎼┴␊⎼/▒≥⎼│␋␍␋▒/▒┤␍␋⎺F⎼▒└␊ RIFF4WAVE°└├ ?>£├␋└????≤␍▒├▒⎺?┤???????????????????!?4?K?␋?????????????????????????????????????#????????????????????????????????????????
 ??????????????????????'0??????+<4Aab2
                                      &)IPe??????????
                                                     ./62V[]bn??????????????vI.?	%19

                                                                                           ?????wx^ZY]Yot???????????????yp?z?????????????????????????????????????????????????????bK???????????
                 )4AIB
hermes/audioServer/ATOMECHO/audioFrame RIFF,WAVEfmt ?>}data'?,?/?.?4?'?+?&?3?"?-?/?;?.?0?1?.?1?,?0?0?,?.?'?/?3?:?4?2?6?@?;?9?9?9?<?:???I?7?=?B?C?8?6???;?9?;?:?;?>?>?B???3?:?9???8?;?A?;?6?D?5?;?7???=?9?>?>?=?;?C?9?@?G?I?L?D?=?E?A?H? ?)??!?!?&?&?"?)?+?+?(?.?'?*?+?(?)?7?6?4?7?:???8?;?A?<?=?=?@?C?6?;?5?<?7?>?.?6?4?/?0?*??!??*? ??? ???????????
????%?$?,?"?)?1?.?-?1?1?=?.?9?5?=?;?@?A?B?>?J?G???B?D???:?=?4?;?5?=?:?>?7?9?2?9?8?/?=?:?=?/?8?<?6?=?=?=?6?5?9?7?9?:?<?3?8?8?=?@?C?;?;?C?=?=???>?8???8?9?5?9?6?7?1???9?:?;?2?8?-?2?(?.?,?(?/?.?0?5?A?:?8?9?D?:?1?:?7?
hermes/audioServer/azrxidia/audioFrame RIFF4WAVEfmt ?>}tim????ydata6$FENQN3BHSijs??????????????svaUJOFA+9>GO5$??

????????????u?h?i?C?C?M?N?T?7?R?f?t?n?h?r?a?k?o?s?n?o?????????v?n?h?|?d???????????????????????????!!'')&$;?DB????????????????????????
?????????????????????????????????????????????????????????????????????????? DCA_o????????????????????????rvZTLKZhov?caaZZ[IREVUT^dbh`hxg??????????????????????
hermes/audioServer/azrxidia/audioFrame RIFF4WAVEfmt ?>}tim??ydata??????v?mb?zv????????????X^\Xo??vlo?s>H<??????????????????????????????????????
???&JHCW[aO@HM\XTD/BH=????????????????????????????????????????????????????????????????????????
[... cut ...]
                                                                                                 ???????????+@N[\±┐VG41'?071UCYA-)32-????????3G8,.)!7
?
 !66K/	???? 1*%=I?F.%-2&!1
????%&*
        ????????????????????????????????z?k?p?o?h?y?t?m?^?a?h?n?{???????????
```

`hermes/audioServer/azrxidia/audioFrame RIFF4WAVEfmt ?>}timc???ydata`
`hermes/audioServer/ATOMECHO/audioFrame RIFF,WAVEfmt ?>}data`

I see a difference between the two strings, where the Snips one has `}timc???ydata` where the Atom has `}data` only?

Romkabouter, 2021-06-03:
> I managed to receive audio! I modified the PubSubClient to have a larger buffer.

Ah ok, this is the ("MQTT_MAX_PACKET_SIZE", 2000) setting in this file ;)

https://github.com/Romkabouter/ESP32-Rhasspy-Satellite/blob/master/PlatformIO/load_settings.py

Snips has an additional header, found here: 
https://github.com/snipsco/hermes-protocol/blob/develop/hermes/src/ontology/audio_server.rs#L25..L87

I have no such extra bytes, which is why the header is different,
I have tested this the header with Snips and that worked for me, so I think it has some additional functionality for snips but is not causing issues if those bytes are not there.

flatsiedatsie, 2021-06-03:
Yes I already had a `MQTT_MAX_PACKET_SIZE` define in the code, but it seemed to not take. The makers of pubsubclient seemed to recommend using a function to change this vlue, so I added that.

`audioServer.setBufferSize(MQTT_MAX_PACKET_SIZE);`

We're getting closer :-)

flatsiedatsie, 2021-06-10:
I tried recording the audio using the tool you created. It worked! There is a slight metalic sound to it, but it's definitely understandable. The volume is very low though. I'll try playing with the gain option.

Romkabouter, 2021-06-10:
Gain is actualy only used in the Matrix Voice I think. Expect unexpected results!

flatsiedatsie, 2021-06-10:
Good news: I managed to get it to detect a hotword by shouting very loudly.

I'm looking closer at how the back-and-forth with Snips is going. After it detects the hotword, the ASR doesn't receive audio (timeout).

```
{"sessionId":"42b02e1c-331e-4aaf-abb1-5a548abedeec","customData":null,"siteId":"ATOMECHO","reactivatedFromSessionId":null}
{"sessionId":"42b02e1c-331e-4aaf-abb1-5a548abedeec","customData":null,"termination":{"reason":"timeout","component":"asr"},"siteId":"ATOMECHO"}
```

flatsiedatsie, 2021-06-10:
There is a doubling going on again it seems.

```
14:10:31.055 -> end of idle. Stream was set to true.
14:10:31.055 -> Total heap: 293320
14:10:31.055 -> Free heap: 147664
14:10:31.055 -> Incoming MQTT message. Topic: hermes/voco/ATOMECHO/play
14:11:10.843 -> Incoming MQTT message. Topic: hermes/hotword/azrxidia/detected
14:11:11.058 -> Incoming MQTT message. Topic: hermes/voco/ATOMECHO/play
14:11:11.058 -> Incoming MQTT message. Topic: hermes/hotword/toggleOff
14:11:11.093 -> toggleOff message was for us
14:11:11.093 -> SessionId in toggleOff:59a6374e-89c4-49b7-a654-d77f77a7384c
14:11:11.093 -> Hotword detected event
14:11:11.093 -> Enter HotwordDetected
14:11:11.093 -> -Semaphone something
14:11:11.093 -> -Re-stream
14:11:25.930 -> Incoming MQTT message. Topic: hermes/hotword/toggleOn
14:11:25.930 -> toggleOn message was for us. Going to idle mode.
14:11:25.930 -> hw-detected-go-back-to-idle
14:11:25.930 -> Enter Idle
14:11:25.968 -> still in idle
14:11:25.968 -> end of idle. Stream was set to true.
14:11:25.968 -> Total heap: 293412
14:11:25.968 -> Free heap: 150684
14:11:26.472 -> Incoming MQTT message. Topic: hermes/voco/ATOMECHO/play
14:12:32.626 -> One of them failed: Enter MQTTDisconnected
14:12:32.626 -> Audio connected: 0, Async connected: 0
14:12:32.626 -> Enter MQTTDisconnected
14:12:32.626 -> Connecting MQTT: 192.168.2.165, 1883
14:12:32.626 -> Connecting MQTT: 192.168.2.165, 1883
14:12:32.626 -> asyncclient connect was called
14:12:32.626 -> asyncclient connect was called
14:12:32.626 -> also reconnecting to audio
14:12:32.626 -> also reconnecting to audio
14:12:47.011 -> 
14:12:47.011 -> ELF file SHA256: 0000000000000000
14:12:47.046 -> 
14:12:47.046 -> Backtrace: 0x40088938:0x3ffbf9d0 0x40088bb5:0x3ffbf9f0 0x40140d30:0x3ffbfa10 0x400870c9:0x3ffbfa30 0x4000cff5:0x3ffde0d0 0x400db815:0x3ffde0f0 0x400db88e:0x3ffde130 0x400d156d:0x3ffde160 0x400d15a7:0x3ffde1f0 0x400d16c7:0x3ffde210 0x400d1fa2:0x3ffde230 0x40089c06:0x3ffde6b0
14:12:47.046 -> 
14:12:47.046 -> Rebooting...

```

flatsiedatsie, 2021-06-10:
These are some messages going to the ASR:

```
hermes/asr/startListening {"siteId":"azrxidia","sessionId":"569ac21f-cde0-4004-be21-f6112640cfdf","startSignalMs":1623327749156}
hermes/asr/stopListening {"siteId":"azrxidia","sessionId":null}
hermes/asr/stopListening {"siteId":"ATOMECHO","sessionId":"bf788e48-fe11-4206-9469-5ac4ec3fd8bd"}
hermes/asr/startListening {"siteId":"ATOMECHO","sessionId":"bf788e48-fe11-4206-9469-5ac4ec3fd8bd","startSignalMs":-20}
```

The StartSignalMS seems to be a strange value: -20. Maybe that's because the time data isn't in the audio stream?

flatsiedatsie, 2021-06-15:
The ASR does work for other satellites in the house, which are based on Voco/Snips. Perhaps they are sending an extra message.

The latest Arduino code can be found here: https://github.com/flatsiedatsie/voco_mini_sat

Romkabouter, 2021-06-15:
Ok, checking your code.

1) why do you have the publish to asr/stopListening" on line 44?
2) Is that the exact code? Is this also p[rinted twice? Serial.println("Creating I2Stask"); If so, maybe the check f (i2sHandle == NULL)  does not work as expected

flatsiedatsie, 2021-06-16:
1.
Sharp eyes -) I was trying to stop and then restart the ASR, hoping that would fix the issue. But then I tried skipping the HotwordDetected state alltogether. So currently that code is never called. All the HotwordDetected state did, was to stop the stream and restart it, which I suspected wasn't needed if there wasn't on-board hotword detection being done.

2.
I've only removed the wifi password :-)

Just in case you'd like to try uploading via the arduino IDE yourself:

1. Tou'll need to add ESP32 support. In the menu go to settings, and add these two lines under `additional board manager urls`:

```
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
https://dl.espressif.com/dl/package_esp32_index.json
```

(maybe restart the IDE)

2.
Then under `tools` -> boards -> boards manager select `M5Atom-stack` as the device.

3. Make sure the USB port is selected as the serial port (under `tools` as well)

4. Make sure the serial monitor is closed in case it's already open. Then click on `ESP32 Sketch data upload` under the tools menu. This will upload the settings file to the SPIFF storage.

5. Upload the code (arrow button in the top-left)

6. Open the serial monitor (under `tools`), and you can see the serial output.

I believe I've managed to remove the double call of MMQTTDisconnected state. The `run` method was apparently calling it before it has switched to the new state, causing the state to be initiated twice. It now no longer crashes after the first recognition of "hey snips".

The strange thing is that the ASR stops responding for the entire system if I use the AtomEcho. The ASR also stops responding to the main microphone, although it still does hotword detection fine. A session is also created just fine.

Sometimes the ASR stops working alltogether, and sometimes it will work 50%, intermittently: after a succesfull run it will not respond the next time, and then start responding again after that, and so forth. This seems to only happens if the AtomEcho is on the network.

The AtomEcho also seems to go into reboot loops. I'm not sure how that's even possible. It's as if it remembers that the previous time it booted up, it failed, and will continue to do so until I unplug it, and then plug it in again.

flatsiedatsie, 2021-06-16:
Just saw another strange situation where I disconnected the AtomEcho, and then the ASR started only listening for 1 second on the main microphone.

```
[11:15:07] [Hotword] detected on site azrxidia, for model hey_snips
[11:15:07] [Asr] was asked to stop listening on site azrxidia
[11:15:07] [Hotword] was asked to toggle itself 'off' on site azrxidia
[11:15:07] [Dialogue] session with id '1819212f-7fe3-40c9-83f7-26021d46f671' was started on site azrxidia
[11:15:07] [Asr] was asked to listen on site azrxidia
[11:15:09] [Asr] captured text "unknownword" in 1.0s
[11:15:09] [Asr] was asked to stop listening on site azrxidia
[11:15:09] [Nlu] was asked to parse input "unknownword"
[11:15:09] [Nlu] intent not recognized for "*"
[11:15:09] [Dialogue] session with id '1819212f-7fe3-40c9-83f7-26021d46f671' was ended on site azrxidia. The session was ended because the platform didn't understand the user
[11:15:09] [Asr] was asked to stop listening on site azrxidia
[11:15:09] [Hotword] was asked to toggle itself 'on' on site azrxidia
[11:15:13] [Hotword] detected on site azrxidia, for model hey_snips
[11:15:13] [Asr] was asked to stop listening on site azrxidia
[11:15:13] [Hotword] was asked to toggle itself 'off' on site azrxidia
[11:15:13] [Dialogue] session with id '5efe57ea-f984-4ad1-8342-ba4a9d6a1e47' was started on site azrxidia
[11:15:13] [Asr] was asked to listen on site azrxidia
[11:15:28] [Dialogue] session with id '5efe57ea-f984-4ad1-8342-ba4a9d6a1e47' was ended on site azrxidia. The session was ended because one of the component didn't respond in a timely manner
[11:15:28] [Asr] was asked to stop listening on site azrxidia
[11:15:28] [Hotword] was asked to toggle itself 'on' on site azrxidia
[11:15:38] [Hotword] detected on site azrxidia, for model hey_snips
[11:15:38] [Asr] was asked to stop listening on site azrxidia
[11:15:38] [Hotword] was asked to toggle itself 'off' on site azrxidia
[11:15:38] [Dialogue] session with id '13bfee2f-4b4a-4ae0-8896-916fb8e6d27b' was started on site azrxidia
[11:15:38] [Asr] was asked to listen on site azrxidia
[11:15:40] [Asr] captured text "unknownword" in 1.0s
[11:15:40] [Asr] was asked to stop listening on site azrxidia
[11:15:40] [Nlu] was asked to parse input "unknownword"
[11:15:40] [Nlu] intent not recognized for "*"
[11:15:40] [Dialogue] session with id '13bfee2f-4b4a-4ae0-8896-916fb8e6d27b' was ended on site azrxidia. The session was ended because the platform didn't understand the user
[11:15:40] [Asr] was asked to stop listening on site azrxidia
[11:15:40] [Hotword] was asked to toggle itself 'on' on site azrxidia
[11:16:37] [Hotword] detected on site azrxidia, for model hey_snips
[11:16:37] [Asr] was asked to stop listening on site azrxidia

```

After that it reverted to the intermittent "ASR listens, ASR is deaf" situation.

flatsiedatsie, 2021-06-16:
I've tried to manually run the ASR and check it's output. Here's what happens with a "normal" call from Voco:

```
pi@thuis:~/.webthings/addons/voco/snips $ LD_LIBRARY_PATH=. /home/pi/.webthings/addons/voco/snips/snips-asr -u /home/pi/.webthings/data/work -a /home/pi/.webthings/addons/voco/snips/assistant -c /home/pi/.webthings/addons/voco/snips/snips.toml
[11:27:30.198765] INFO :snips_asr_hermes::handler: Using model from "/home/pi/.webthings/data/work/injections/20210209T163004178929730/inj_20210616T092026773150365/asr"
[11:27:30.332529] INFO :snips_kaldi::decode::model: Loading model v2
[11:27:31.958167] INFO :snips_asr_hermes::handler : Preparing decoder
[11:27:31.958415] INFO :snips_asr_hermes::handler : Preparing decoder
[11:28:11.557659] INFO :snips_asr_hermes::handler : Listening at site id azrxidia
[11:28:11.557826] INFO :snips_asr_hermes::handler : Listening
[11:28:11.704154] INFO :snips_asr_lib::asr        : T0       entered AsrRunner::run
[11:28:11.704224] INFO :snips_asr_lib::asr        : T0+0.000 capture started
[11:28:13.883099] INFO :snips_asr_lib::asr        : T0+2.179 endpoint detected (rule:4) frame:155 samples:39680 signal_time:2.48 rtf:0.327
[11:28:13.883973] INFO :snips_asr_lib::asr        : Source thread stop on push: "SendError(..)"
[11:28:13.884145] INFO :snips_asr_lib::asr        : T0+2.180 capture ended
[11:28:13.885827] INFO :snips_asr_lib::asr        : T0+2.182 decoder finalized
[11:28:13.894667] INFO :snips_asr_lib::asr        : T0+2.191 lookup and post-processing done
[11:28:13.894747] INFO :snips_asr_lib::asr        : decoded: [Recognition { decoded_string: "what time is it", likelihood: 1.0, tokens: Some([Token { value: "what", confidence: 1.0, time: (0.0, 1.38), range: 0..4 }, Token { value: "time", confidence: 1.0, time: (1.38, 1.4399999), range: 5..9 }, Token { value: "is", confidence: 1.0, time: (1.4399999, 1.62), range: 10..12 }, Token { value: "it", confidence: 1.0, time: (1.62, 2.31), range: 13..15 }]) }]
[11:28:13.895411] INFO :snips_asr_hermes::handler : Publishing the recognition
```

And this is all that happens with the AtomEcho:
```
[11:28:25.235052] INFO :snips_asr_hermes::handler : Preparing decoder
[11:29:24.793911] INFO :snips_asr_hermes::handler : Listening at site id ATOMECHO
[11:29:24.793989] INFO :snips_asr_hermes::handler : Listening
```

[18 more replies]

closed 2021-11-29 by Romkabouter