Romkabouter/ESP32-Rhasspy-Satellite #45: Crash (possible memory corruption) after changing configuration in the web interface

ayavilevich, 2021-03-10:
After changing config in the web UI, the code will sometimes crash and reboot. It doesn't seem to be related to a specific parameter.

I noticed that while working in a fork but also confirmed this is the case with the "master" branch as well. I wasn't able to find the cause.

This bug is not critical as the config will be applied after the restart anyway and since configuration is rarely changed. But it might be related to something else or might have effect on a more "core" functionality in the project so maybe worth looking at.

Here are the crash dumps. I was able to witness two types:

```

Parameter mqtt_host, value 192.168.x
Mqtt host changed
Parameter mqtt_port, value 1883
Parameter mqtt_user, value x
Parameter mqtt_pass, value x
Parameter amp_output, value 1
Parameter volume, value 100
Parameter brightness, value 15
Parameter hw_brightness, value 15
Parameter hotword_detection, value 1
Parameter gain, value 5
Settings changed, saving configuration
Saving configuration
{
  "mqtt_host": "192.168.x",
  "mqtt_port": 1883,
  "mqtt_user": "x",
  "mqtt_pass": "x",
  "mqtt_valid": true,
  "mute_input": false,
  "mute_output": false,
  "amp_output": 1,
  "brightness": 15,
  "hotword_brightness": 15,
  "hotword_detection": 1,
  "volume": 100,
  "gain": 5
}
Connect failed, retry
Enter MQTTDisconnected
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d95cb  PS      : 0x00060130  A0      : 0x800d979f  A1      : 0x3ffbc020
A2      : 0x00000000  A3      : 0x3fff127c  A4      : 0x00000000  A5      : 0x3fff0eee  
A6      : 0x00000014  A7      : 0x3ffbb5c0  A8      : 0x800853a8  A9      : 0x3ffbc000
A10     : 0x3fff1474  A11     : 0x0000166f  A12     : 0x5730a6b5  A13     : 0x00000b38  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace: 0x400d95cb:0x3ffbc020 0x400d979c:0x3ffbc040 0x400d97de:0x3ffbc060 0x4013b866:0x3ffbc090 0x40141eaf:0x3ffbc0b0 0x40147142:0x3ffbc0d0 0x4013327b:0x3ffbc0f0 0x40089015:0x3ffbc120
  #0  0x400d95cb:0x3ffbc020 in AsyncClient::_lwip_fin(tcp_pcb*, signed char) at .pio\libdeps\esp32dev\AsyncTCP@src-7fb2940bccb78b8d2de6915ae328b7fc\src/AsyncTCP.cpp:1039
  #1  0x400d979c:0x3ffbc040 in AsyncClient::_s_lwip_fin(void*, tcp_pcb*, signed char) at .pio\libdeps\esp32dev\AsyncTCP@src-7fb2940bccb78b8d2de6915ae328b7fc\src/AsyncTCP.cpp:1039
  #2  0x400d97de:0x3ffbc060 in _tcp_recv(void*, tcp_pcb*, pbuf*, signed char) at .pio\libdeps\esp32dev\AsyncTCP@src-7fb2940bccb78b8d2de6915ae328b7fc\src/AsyncTCP.cpp:1039
  #3  0x4013b866:0x3ffbc090 in tcp_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_in.c:475 (discriminator 1)
  #4  0x40141eaf:0x3ffbc0b0 in ip4_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c:746
  #5  0x40147142:0x3ffbc0d0 in ethernet_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/netif/ethernet.c:184
  #6  0x4013327b:0x3ffbc0f0 in tcpip_thread at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c:483
  #7  0x40089015:0x3ffbc120 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...

Parameter mqtt_host, value 192.168.x
Parameter mqtt_port, value 1883
Parameter mqtt_user, value x
Mqtt user changed
Parameter mqtt_pass, value x
Parameter amp_output, value 1
Parameter volume, value 100
Parameter brightness, value 15
Parameter hw_brightness, value 15
Parameter hotword_detection, value 1
Parameter gain, value 5
Settings changed, saving configuration
Saving configuration
{
  "mqtt_host": "192.168.x",
  "mqtt_port": 1883,
  "mqtt_user": "x",
  "mqtt_pass": "x
  "mqtt_valid": true,
  "mute_input": false,
  "mute_output": false,
  "amp_output": 1,
  "brightness": 15,
  "hotword_brightness": 15,
  "hotword_detection": 1,
  "volume": 100,
  "gain": 5
}
Enter MQTTDisconnected

Backtrace: 0x4008ca58:0x3ffbe170 0x4008cc89:0x3ffbe190 0x400ec4eb:0x3ffbe1b0 0x40084ab9:0x3ffbe1d0 0x4000bfed:0x3ffeb140 0x4008a181:0x3ffeb150 0x40088e20:0x3ffeb170 0x40147242:0x3ffeb1b0 0x40135ecd:0x3ffeb1d0 0x400dabd9:0x3ffeb1f0 0x400daf2d:0x3ffeb220 0x400da668:0x3ffeb250 0x400da6e6:0x3ffeb290 0x400d23b5:0x3ffeb2c0 0x400d23eb:0x3ffeb360 0x400d24eb:0x3ffeb380 0x400d2f53:0x3ffeb3a0 0x40089015:0x3ffeb820
  #0  0x4008ca58:0x3ffbe170 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #1  0x4008cc89:0x3ffbe190 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #2  0x400ec4eb:0x3ffbe1b0 in task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c:252
  #3  0x40084ab9:0x3ffbe1d0 in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
  #4  0x4000bfed:0x3ffeb140 in ?? ??:0
  #5  0x4008a181:0x3ffeb150 in vTaskExitCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
  #6  0x40088e20:0x3ffeb170 in xQueueGenericReceive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1996
  #7  0x40147242:0x3ffeb1b0 in sys_mutex_lock at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c:203
  #8  0x40135ecd:0x3ffeb1d0 in lwip_ioctl_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/sockets.c:3532
  #9  0x400dabd9:0x3ffeb1f0 in WiFiClientRxBuffer::r_available() at C:\Users\arik\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src/WiFiClient.cpp:564
  #10 0x400daf2d:0x3ffeb220 in WiFiClientRxBuffer::available() at C:\Users\arik\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src/WiFiClient.cpp:564
      (inlined by) WiFiClient::available() at C:\Users\arik\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src/WiFiClient.cpp:458
  #11 0x400da668:0x3ffeb250 in PubSubClient::connect(char const*, char const*, char const*, char const*, unsigned char, bool, char const*, bool) at .pio\libdeps\esp32dev\PubSubClient\src/PubSubClient.cpp:706
  #12 0x400da6e6:0x3ffeb290 in PubSubClient::connect(char const*, char const*, char const*) at .pio\libdeps\esp32dev\PubSubClient\src/PubSubClient.cpp:706
  #13 0x400d23b5:0x3ffeb2c0 in MQTTDisconnected::entry() at src/StateMachine.hpp:19
  #14 0x400d23eb:0x3ffeb360 in void tinyfsm::Fsm<StateMachine>::transit<MQTTDisconnected>() at src/StateMachine.hpp:19
  #15 0x400d24eb:0x3ffeb380 in Idle::react(MQTTDisconnectedEvent const&) at src/StateMachine.hpp:19
  #16 0x400d2f53:0x3ffeb3a0 in void tinyfsm::Fsm<StateMachine>::dispatch<MQTTDisconnectedEvent>(MQTTDisconnectedEvent const&) at src/StateMachine.hpp:19
      (inlined by) void send_event<MQTTDisconnectedEvent>(MQTTDisconnectedEvent const&) at src/StateMachine.hpp:270
      (inlined by) I2Stask(void*) at src/StateMachine.hpp:567
  #17 0x40089015:0x3ffeb820 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```

Romkabouter, 2021-03-10:
Thanks, I think it has something to do with the disconnecting of the MQTT causing a crash.
In previous I use a specific commit on the asynch mqtt code which fixes this, I will see if that is still the issue or if it is something else.

Romkabouter, 2021-04-21:
I could not find any history on this anymore.
Therefore, I decided to just reboot the device when a configuration is saved.

Please try v7.5.1

closed 2021-04-21 by Romkabouter