Romkabouter/ESP32-Rhasspy-Satellite #52: New simple device INMP441

mttstt, 2021-03-20:
Hi, I tried modify existing devices for INMP441 but without success.
Below the code with my esp32cam. I see raw data on Mqtt Explorer, but remore wakeword doesn't work. Any ideas?

```
#include <driver/i2s.h>

// I2S pins
#define IIS_SCLK 14
#define IIS_WSLC 2
#define IIS_DSIN 15
#define IIS_DSOU -1 //not used, only for speaker

#define CONFIG_I2S_BCK_PIN IIS_SCLK
#define CONFIG_I2S_LRCK_PIN IIS_WSLC
#define CONFIG_I2S_DATA_PIN IIS_DSIN
#define CONFIG_I2S_DATA_IN_PIN IIS_DSOU

#define I2S_PORT I2S_NUM_0

class Inmp441 : public Device
{
public:
  Inmp441();
  void init();
  bool readAudio(uint8_t *data, size_t size);
};

Inmp441::Inmp441() {};

void Inmp441::init() {
  Serial.printf("Connect to Inmp441... ");
  esp_err_t err = ESP_OK;
  i2s_config_t i2s_config = {
    .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX),
    .sample_rate = 16000,
    .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
    .channel_format = I2S_CHANNEL_FMT_ONLY_LEFT,
    .communication_format = i2s_comm_format_t(I2S_COMM_FORMAT_I2S),
    .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,
    .dma_buf_count = 4,
    .dma_buf_len = 64,
    .use_apll = false,
    .tx_desc_auto_clear = false,
    .fixed_mclk = 0
  };
  
  err += i2s_driver_install(I2S_PORT, &i2s_config, 0, NULL);
  i2s_pin_config_t tx_pin_config;
  
  tx_pin_config.bck_io_num = CONFIG_I2S_BCK_PIN;
  tx_pin_config.ws_io_num = CONFIG_I2S_LRCK_PIN;
  tx_pin_config.data_out_num = CONFIG_I2S_DATA_PIN;
  tx_pin_config.data_in_num = CONFIG_I2S_DATA_IN_PIN;
  
  err += i2s_set_pin(I2S_PORT, &tx_pin_config);
  err += i2s_set_clk(I2S_PORT, 16000, I2S_BITS_PER_SAMPLE_16BIT, I2S_CHANNEL_MONO);
  return;
}

bool Inmp441::readAudio(uint8_t *data, size_t size) {
  size_t byte_read;
  i2s_read(I2S_PORT, data, size, &byte_read, (100 / portTICK_RATE_MS));
  return true;
}
```

Romkabouter, 2021-03-20:
Well, i can see a difference in communication_format and the tx_pin_config when comparing to this project:
https://github.com/0015/IdeasNProjects/blob/master/ESP32_MICROPHONE/ESP32_INMP441_RECORDING/ESP32_INMP441_RECORDING.ino
Also, the pins are different in above project.
I2S_READ_LEN is set to #define I2S_READ_LEN      (16 * 1024), and you use the default settings I think?

Maybe this can help you a bit

mttstt, 2021-03-30:
Here a new version and it seems OK. I hear the voice correctly collected by mqtt script, but wake word (alexa) is not recognised by Rhasspy. Any idea?

```
#include <driver/i2s.h>

// I2S pins on ESp32cam
//GPIO2 <--> WS
//GPIO14 <--> SCK
//GPIO15 <--> SD

#define I2S_SCK 14
#define I2S_WS 2
#define I2S_SD 15

#define I2S_PORT I2S_NUM_0
#define I2S_SAMPLE_RATE   (16000)
#define I2S_SAMPLE_BITS   (16)
#define I2S_READ_LEN      512

class Inmp441 : public Device
{
  public:
    Inmp441();
    void init();
    bool readAudio(uint8_t *data, size_t size);
  private:
    char* i2s_read_buff = (char*) calloc(I2S_READ_LEN, sizeof(char));
};

Inmp441::Inmp441() {};

void Inmp441::init() {
  esp_err_t err = ESP_OK;
  Serial.printf("Connect to Inmp441... \n");

  i2s_config_t i2s_config = {
    .mode = (i2s_mode_t)(I2S_MODE_MASTER|I2S_MODE_RX),
    .sample_rate = I2S_SAMPLE_RATE,
    .bits_per_sample = i2s_bits_per_sample_t(I2S_SAMPLE_BITS),
    .channel_format = I2S_CHANNEL_FMT_ONLY_LEFT,
    .communication_format = i2s_comm_format_t(I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB),
    .intr_alloc_flags = 0,
    .dma_buf_count = 6,
    .dma_buf_len = 512,
    .use_apll = 1
  };

  i2s_pin_config_t pin_config = {
    .bck_io_num = I2S_SCK,
    .ws_io_num = I2S_WS,
    .data_out_num = -1,
    .data_in_num = I2S_SD
  };

  err += i2s_driver_install(I2S_PORT, &i2s_config, 0, NULL);
  if (err != ESP_OK) {
     Serial.printf("Failed installing driver: %d\n", err);
     while (true);
  }

  err += i2s_set_pin(I2S_PORT, &pin_config);
  if (err != ESP_OK) {
    Serial.printf("Failed setting pin: %d\n", err);
    while (true);
  }
  Serial.println("I2S driver installed.\n");
  return;
}

bool Inmp441::readAudio(uint8_t *data, size_t size) {
      size_t bytes_read;
      i2s_read(I2S_PORT, (void*) i2s_read_buff, size, &bytes_read, portMAX_DELAY);
      uint32_t j = 0;
      uint32_t dac_value = 0;
      for (int i = 0; i < size; i += 2) {
          dac_value = ((((uint16_t) (i2s_read_buff[i + 1] & 0xf) << 8) | ((i2s_read_buff[i + 0]))));
          data[j++] = 0;
          data[j++] = dac_value * 256 / 2048;
      }
      return true;
}

```

Romkabouter, 2021-03-30:
Great that it is working now :)

The wakeword issue depend on your Rhasspy settings probably, I suggest posting issue with Rhasspy here:
https://community.rhasspy.org/

closed 2021-03-30 by Romkabouter