Romkabouter/ESP32-Rhasspy-Satellite #57: Documentation and configuration

ondragu, 2021-04-16:
Hello,
i´m new to rhasspy and to your esp32-satellite and i´ve setup rhasspy on a raspberry pi 4 and your software on a esp32-audio-kit 2.2.
I have managed to see the satellite on my mqtt-explorer producing audio and connected my server-rhasspy to it too.
But I failed in geting output or wakeword detection on the server.
Here is my config file : 

`{
    "dialogue": {
        "satellite_site_ids": "satellite-01",
        "system": "hermes",
        "volume": "0.2"
    },
    "handle": {
        "system": "command"
    },
    "intent": {
        "satellite_site_ids": "satellite-01",
        "system": "fsticuffs"
    },
    "microphone": {
        "pyaudio": {
            "device": "1"
        },
        "system": "hermes"
    },
    "mqtt": {
        "enabled": "true",
        "host": "192.168.188.23",
        "password": "test",
        "site_id": "server",
        "username": "test"
    },
    "sounds": {
        "aplay": {
            "device": "sysdefault:CARD=Headphones",
            "volume": "0.6"
        },
        "remote": {
            "url": "http://192.168.188.26:12101/api/play-wav"
        },
        "system": "hermes"
    },
    "speech_to_text": {
        "satellite_site_ids": "satellite-01",
        "system": "kaldi"
    },
    "text_to_speech": {
        "satellite_site_ids": "satellite-01",
        "system": "nanotts"
    },
    "wake": {
        "porcupine": {
            "keyword_path": "computer_raspberry-pi.ppn"
        },
        "raven": {
            "keywords": {
                "Computer": {
                    "enabled": true
                }
            }
        },
        "satellite_site_ids": "satellite-01",
        "system": "porcupine"
    }
}`
[image]
[image]

How has the server to be configured to work with the satellite ?
Is there a documentation or a wiki ?
I can´t program code, but I am willing to contribute in other ways if I can !?
Can the wakeword-detection in the satellite be set to "Local"
Thank you for your work
Greetings from Aachen, Germany

Romkabouter, 2021-04-16:
What is the siteid you use for the AudioKit?

DialogueManager should be Rhasspy

ondragu, 2021-04-16:
My siteID is satellite-01. Seems to be o.k ? Ports are all open.
By the way, what does the DeployHostname do ? I´ve set it to the same adress as the hostname. Both to my rhasspy server.

Romkabouter, 2021-04-16:
DialogueManager should be Rhasspy, is it?

ondragu, 2021-04-16:
Sorry, changed it, but no success.

Romkabouter, 2021-04-16:
No, I have that running. Can't tell what is the problem, maybe you can try here https://community.rhasspy.org/

Open a topic and post the log files from Rhasspy :)

riesemann, 2021-04-21:
Hi!
I've got AudioKit working with the latest version of ESP32-Rhasspy-Satellite. It works very good! Thx @Romkabouter :-)

My Configuration:

- The Base is answering by hotword 
- The AudioKit is triggered via Key4 on the board or alternatively by setting IO23 to GND (i used a pushbutton)

## Rhasspy-Base (AUTOMAT)

Here's my base configuration:

````
{
    "dialogue": {
        "satellite_site_ids": "WANZE2,AUTOMAT",
        "system": "rhasspy"
    },
    "intent": {
        "satellite_site_ids": "WANZE2,AUTOMAT",
        "system": "fuzzywuzzy"
    },
    "microphone": {
        "arecord": {
            "device": "plughw:CARD=USB,DEV=0"
        },
        "system": "arecord"
    },
    "mqtt": {
        "enabled": "true",
        "host": "automat",
        "password": "xxxxxxxxx",
        "site_id": "AUTOMAT",
        "username": "xxxxxxxxx"
    },
    "sounds": {
        "aplay": {
            "device": "sysdefault:CARD=USB"
        },
        "error": "${RHASSPY_PROFILE_DIR}/sounds/computerbeep_42.wav",
        "recorded": "${RHASSPY_PROFILE_DIR}/sounds/computerbeep_41.wav",
        "system": "aplay",
        "wake": "${RHASSPY_PROFILE_DIR}/sounds/computerbeep_26.wav"
    },
    "speech_to_text": {
        "satellite_site_ids": "WANZE2,AUTOMAT",
        "system": "pocketsphinx"
    },
    "text_to_speech": {
        "satellite_site_ids": "WANZE2,AUTOMAT",
        "system": "picotts"
    },
    "wake": {
        "raven": {
            "keywords": {
                "default": {
                    "enabled": true
                }
            }
        },
        "satellite_site_ids": "WANZE2",
        "system": "raven"
    }
}
````

## AudioKit (WANZE2)

Here's my AudioKit config:

````
[General]
hostname=WANZE2
deployhost=automat
siteId=WANZE2
;supported: M5ATOMECHO=0, MATRIXVOICE=1, AUDIOKIT=2, INMP441=3, INMP441MAX98357A=4
device_type=2

...
````

Romkabouter, 2021-04-21:
> My siteID is satellite-01. Seems to be o.k ? Ports are all open.
> By the way, what does the DeployHostname do ? I´ve set it to the same adress as the hostname. Both to my rhasspy server.

DeployToHostname does nothing when you device is not a MatrixVoice. If it is a MatrixVoice, it is used for flashing the first time.
The hostname should not be your Rhasspy server, but the IP address you want your device to get. It might be better to use a static IP address

Romkabouter, 2021-04-21:
> Is there a documentation or a wiki ?

For Rhasspy yes: https://rhasspy.readthedocs.io/en/latest/
This software is aiming to be a satellite, all settings you have already found in the webUI.

If you have any questions about Rhasspy, it is better to post it here: https://community.rhasspy.org/
Since the audio out seems to work, I am going to close this issue.

If you feel this is incorrect, I can reopen it :)

closed 2021-04-21 by Romkabouter