Romkabouter/ESP32-Rhasspy-Satellite #61: How hard would be to integrate with ESPHome?

ashald, 2021-05-30:
Thanks a lot for an amazing work!
I'm trying to install it on my MATRIX devices and thought it would be amazing if it could be used together with
[ESPHome](https://esphome.io), which provides a super neat way to build firmware for ESP based on desired modules.

Romkabouter, 2021-05-30:
What is your vision that this software should do within ESPHome?

ashald, 2021-05-31:
My apologies for not providing enough details.

ESPHome provides a simple way to configure ESP32-based devices. As the use, I can write a simple YAML config that enables desired modules. From there on, ESPHome combines source code for the selected modules, configures them according to what I entered in the YAML config, compiles the firmware and flashes the device.

The extra benefit is that ESPHome integrates with systems like HomeAssistant which makes it breeze to wire things together.
While flashing instructions for [Matrix Voice](https://github.com/Romkabouter/ESP32-Rhasspy-Satellite/blob/master/matrixvoice.md) do not sound too complicated, it would be amazing if one could get it flashed even simpler by just providing few lines of config in ESPHome YAML. I'd imagine it'd work equally well for other supported ESP32 devices.

TL;DR: AFAIU, ESPHome would provide a uniform way for flashing `ESP32-Rhasspy-Satellite` onto compatible devices.

Romkabouter, 2021-05-31:
Yes, I know a bit of ESPHome.
But the matrix voice cannot by flashed by USB, so that makes it a no-go I guess.

ashald, 2021-06-02:
Ugh, my bad - IDK how I missed it.
I wish that limitation goes away one day - it would be nuts if we would've been able to flash Matrix Voice with ESPHome, and get `ESP32-Rhasspy-Satellite` on it that way. 😅
Thanks!

closed 2021-06-02 by ashald

Romkabouter, 2021-06-12:
I saw you can compile and download the firmware. That makes it possible to at least flash ESPHome on the Matrix Voice.
Maybe I'll try something :)

Romkabouter, 2021-06-12:
Ok, that worked :)

[image]

Now what to do with it....

ashald, 2021-06-12:
Wow, that's amazing! As far as I understand that means that in theory, it's possible to leverage ESPHome for subsequent OTA updates, checking logs, etc. Assuming `ESP32-Rhasspy-Satellite` can be converted into ESPHome-compatible add-on.

P.S.: I wonder if there any chances ESPHome can flash MarixVoice-like devices directly, if they're connected to an RPi via HAT (wishful thinking, ha-ha).

Romkabouter, 2021-06-12:
> P.S.: I wonder if there any chances ESPHome can flash MarixVoice-like devices directly, if they're connected to an RPi via HAT

That is actually exactly what I did. From Esphome addon, configured a new OTA device:

```
esphome:
  name: matrixvoice
  platform: ESP32
  board: esp32dev

wifi:
  ssid: "SSID"
  password: "password"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Matrixvoice Fallback Hotspot"
    password: "generatedpassword"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "apipassword"

ota:
  password: "otapassword"
  
switch:
  - platform: gpio
    name: "Test"
    pin: 5

```
Then, on the right side you have three vertical dots, one f them being "Compile""
After is compiles, you can download the binary.
I renamed it firmware.bin and copied it to my streamer project.
The line cp .pio/build/esp32dev/firmware.bin . had to be commented, but then I executed "sh deploy.sh xxx.xxx.xxx.xxx" which flashed the matrix voice

I have added ESPhome as integration and I have 1 device:
[image]
and 1 enitity:

[image]

ashald, 2021-06-12:
I just realized that `ESP32-Rhasspy-Satellite` allows control of the LEDs via MQTT - now, if that can be exported to HA as a light accessory - that would be mind-blowing.
Like, it streams audio to MQTT, Rhasspy handles voice and intent recognition, and as a result we run an HA automation, that signals feedback via LED by using its native APIs. 😳

sehraf, 2021-06-12:
> now, if that can be exported to HA as a light accessory - that would be mind-blowing.

mindblowing? :D thats an absolute common case [this](https://www.home-assistant.io/integrations/light.mqtt/#default-schema---configuration) adds the led as a light and then you just need some automatic that controls it.

When using ESPHome is similar [easy](https://esphome.io/components/light/binary.html). You just need the gpio + a light component.

Romkabouter, 2021-06-12:
Well, the led uses a separate library, also for microphone and such. 
So simply add "the led as a light" will not work, because the led ring is not attached to a GPIO

ashald, 2021-11-19:
@Romkabouter pardon me if I did not catch your reply in our conversation above, but I'm curious what's your verdict on integrating `ESP32 Rhasspy Satellite` with `ESPHome`? What I mean by this is that:
1. User creates a new device config in ESPHome, and enables desired components - `ESP32 Rhasspy Satellite` being one of them. If there are any settings to tweak - user does that in the ESPHome config within `ESP32 Rhasspy Satellite` section
2. For initial flashing - as you described above, it happens via RPi
3. For subsequent re-flashing - user can leverage `ESPHome`'s OTA, so that there is a single place to manage OTA updates for all ESP32-based devices. As well as stream logs.

Given that ESPHome is now maintained by HomeAssitant team, I think its adoption going to grow, and if `ESP32 Rhasspy Satellite` will be available as one of the components, it will significantly lower the barrier for people to start using it. On the other hand, `ESP32 Rhasspy Satellite` can leverages ESPHome's OTA and log streaming systems so that you don't need to maintain that code on your own. As well as MQTT communications, and option for encrypting ESP's traffic.

Romkabouter, 2021-11-19:
Well, I do not now much of ESPHome, so it might be possible,
Without a good knowlegde of ESPHome, it is a bit hard to say how much effort it would take. 
I do not have a verdict, I just do not know :)
I will add it on my to-do list, but do expect any change soon.
I am willing to accept pull-request as always, if you are a developer and want to contribute.

LucaDev, 2022-06-21:
ESPHome just introduced support for playing media (through i2s for example). The media_player interface is currently being used for open portable speakers and media receiver replacements.

It would be quite interesting to extend it to also allow for smart speaker capabilities through Rhasspy. It would just need wake word detection and the mqtt audio streaming capabilities. Could be a interesting way to go. 

I'll have a look at it in my spare time (when I finally have some again)

formatBCE, 2022-11-16:
I recently converted my own BLE scan firmware to ESPHome custom module.
If you need help understanding, how to do it - I may help.

It's very convenient, as users can have additional functionality added to same device easily.
E.g. my BLE scanners (different devices) additionally serve as Bluetooth lock gateway, presence mmWave sensors, pm2.5 sensors and so on. And actually, adding custom component to ESPHome is very easy for user (not for developer though :))

Romkabouter, 2022-11-16:
Thanks for the offer, but I am tied up into other projects at the moment. I always welcome PR's, so if you are willing to put effort in it, that would be much appreciated. But I also understand that would be a bit to much :)