Romkabouter/ESP32-Rhasspy-Satellite #62: Is it possible to integrate local Wake word detection

juliend20, 2021-06-01:
I saw a sdk from espressif https://github.com/espressif/esp-skainet/tree/master/examples/wake_word_detection with an example of Wake word detection feature.

Can you look that and tell me if you can implemented in your great project?
Thanks

Romkabouter, 2021-06-01:
I had already implemented that in an early version, but it does not compile with a newer version of the SDK.
That is why it is not supported at the moment and I am waiting for porcupine to release a lib or esp32

closed 2021-06-01 by Romkabouter

sehraf, 2021-06-01:
looking forward to this, too!

Romkabouter, 2021-06-01:
You can check this release: https://github.com/Romkabouter/ESP32-Rhasspy-Satellite/releases/tag/v6.0.0

But that is only for the Matrix Voice

juliend20, 2021-06-01:
Ok, thank for your answer.