Romkabouter/ESP32-Rhasspy-Satellite #66: M5 Atom Echo not capturing nor playing audio

mpeterson, 2021-07-04:
My `settings.ini` General section is as follows

```
[General]
hostname=satellite
deployhost=192.168.43.24
siteId=satellite
;supported: M5ATOMECHO=0, MATRIXVOICE=1, AUDIOKIT=2, INMP441=3, INMP441MAX98357A=4
device_type=0
;network_type: 0: WiFi, 1: Ethernet
network_type=0
```

When I subscribe to `hermes/#` with `mosquito_sub` to see what the mic is sending I see a lot of:

```
hermes/audioServer/satellite/audioFrame RIFF,WAVEfmt >}data))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))))
```

When I play something from rhasspy I see the proper logging being displayed and in the serial interface, yet not sound is played:

```
Samplerate: 16000, Channels: 1, Format: 1, Bits per Sample: 16
```

I wonder if I'm missing something?

Romkabouter, 2021-07-04:
That is strange, because the logging is actually coming from an audio message being received.

The large amount of audioFrames is also normal, that is the audiostream.
Can you try and adjust this file and see what comes in over the audioFrames?

https://github.com/Romkabouter/ESP32-Rhasspy-Satellite/blob/voco/record.py

mpeterson, 2021-07-05:
oh this is embarrassing... what happened is I had another ESP32 connected to the computer that I forgot about... that ESP was flashed instead :)

Now that I flashed the correct device everything is working as expected! 👍 

Amazing project, thanks for your work!!

closed 2021-07-05 by mpeterson