Romkabouter/ESP32-Rhasspy-Satellite #69: Changing LED effects on profile update

WhiteSockedDancer, 2021-08-12:
Hey there,

i used your streamer for quite some time with a version before you renamed it with a matrix voice module.
It worked quite well, but i decided to update so i can use some of your newer features (more satellites, OTA updates..).

I heavily used the possibility to change the LED colors on the fly (when my assistant is saying something, when i go to bed and want the led to turn off, and so on). 
This seem to not work anymore and there is just an update to the leds if i trigger a voice command.

Is there any way to change this behaviour?

Also the "hotword" color was activated when rhasspy was "not listening".. so when it was speaking i also got another color then on idle. i want to get this procedere back to change colors, when rhasspy is speaking. 

i could trigger all these color changes over MQTT, if there would be an option to change colors whenever a profile update is incoming.

I would love to help you test any code to improve this software! :)

Thanks for this awesome project and keep up the very good work!

Romkabouter, 2021-08-12:
> Is there any way to change this behaviour?

Well, this should still work.

> i could trigger all these color changes over MQTT, if there would be an option to change colors whenever a profile update is incoming.
I am not sure about what you mean by this. What is your definition of profile update?

WhiteSockedDancer, 2021-08-18:
Dear Romkabouter,

thx for replying and excuse my late answer. My computer just exploded twice and i could not get this project more time in the last week.

So.

At the moment. If i send following command over MQTT:
Topic: matrixvoice/led
Payload:  {"brightness":255,"idle":[66,38,0,0]}

The LEDs stay at the same color as before until i use my hotword  and the system goes back to idle mode after that. Just then the new color i sent with my payload is actually activated.

In versions before your renaming and rebuilding this project i could send this command to change the color immediately.

Is there a way to force an update of the color?
I use this for example to turn the LEDs down when going to sleep or activate them again, when waking up. Also when i switch to an DND mode i state that by changing the color to something else.

You also changed the way, the "hotword"-state triggers. Before this state always activated, when the wakeword was "not active" and i guess you just watched these topics?: hermes/hotword/toggleOn and hermes/hotword/toggleOff
I like this change to the wakeword-state, but i also would love to change the colors, when rhasspy is saying something. i think i could use node red for this, if the idle state would update immediately, when sending a new command ({"brightness":255,"idle":[66,38,0,0]})

i know, github is usually not for this kind of post, but are there any forums, where i can present my issue?

Thx very much in advance! 
Jay

Romkabouter, 2021-08-19:
> In versions before your renaming and rebuilding this project i could send this command to change the color immediately.

I wil test this, because the colors should update when a message arrives.
The wakeword flow is not changed, but glad you like it ;)

The functionality needed with "when Rhasspy is saying something" needs to be added. I will flag this with the feature request flag.
The code must listen to another topic for that. This kind of post here is no problem by the way.

WhiteSockedDancer, 2021-08-19:
Thx!

If you want me to test something, I'm always up to help you. Just send me a pm or a post here. 
(I will watch this issue closely, since i get reminded every night when my matrix voice doesnt turn off! ;-) )

Greetings, Jay

Romkabouter, 2021-08-20:
> In versions before your renaming and rebuilding this project i could send this command to change the color immediately.

I have made a commint in the development branch which should it. I have not tested it yet myself, I was not at home.
You can try it if you want.

If it works, you can also create the feature about Rhasppy speaking with node red (just post a message whenever you want), but I think I add an extra color array for it as well.

Romkabouter, 2021-08-20:
I have been checking out the messages and I think I can add some nice features
- subscribe to hermes/nlu/query -> on message do an animation if the device supports it.
- subscribe to hermes/nlu/intentParsed and hermes/nlu/intentNotRecognized -> on message stop animation (go to idle)
- subscribe to hermes/tts/say -> on message do another animation
- subscript to hermes/tts/sayFinished -> on message return to idle.

The animation part is maybe for a later stage, first add two color modes. One for the nluQuery and one for the ttsSay.

Any thoughts?

WhiteSockedDancer, 2021-08-21:
> If it works, you can also create the feature about Rhasppy speaking with node red (just post a message whenever you want), but I think I add an extra color array for it as well.

This works like a charm! i already thank you for this update! ^^
i also changed my messages that they don't use brightness 255 and use 100 instead.

What do you mean by a feature of Rhasspy speaking with node red?

In node red i just also react to different mqtt topics/values and do whatever logic i desire and send some mqtt messages back for example.
So node red could replace all logic in your module and the satellite could just react to one led topic with no states at all and just update the color on a separate incoming mqtt topic.
i guess that would just be an "raw led mode" where you could just send mqtt messages to one topic and reaction to different rhasspy topics would be disabled.

i don't know if we should try to implement this, since we could already use node red to send a complex message to just overwrite all your different states with one color.

> I have been checking out the messages and I think I can add some nice features
>
>     * subscribe to hermes/nlu/query -> on message do an animation if the device supports it.
>
>     * subscribe to [...]

very good ideas!
some thoughts:
usually we have different states (and transitions between):

1. Idle
2. Wakeup Sound/Phrase (in my case its a wav thats saying: "how can i help" (playBytes))
3. Listening (at the moment what the "wakeword state is tracking"
4. Error (if nothing useful is captured, in my case there is also a .wav saying "what?" (playBytes))
5. Answer (say or playBytes)

I guess we could subscribe following topics to get a state, when there is sound-output (2, 4, 5) We could even implement different states for playBytes and say (i'm not yet sure how say and playbytes interact, probably there is always a playBytes when there is a say message and we don't need say at all?)

Start:
hermes/tts/say
hermes/audioServer/<satelliteName>/playBytes/#

Stop:
hermes/tts/sayFinished
hermes/audioServer/<satelliteName>/playFinished

for the Listening state (2) you probably already use 
hermes/asr/startListening
hermes/asr/stopListening

Another way would be to check for toggleOn and toggleOff. 
These messages also contain information on which satellite and the reason. I think you used this method in the early stages of the project?:

hermes/asr/toggleOn => {"siteId": "matrixvoice", "reason": "playAudio"}
hermes/asr/toggleOff => {"siteId": "matrixvoice", "reason": "playAudio"}

Romkabouter, 2021-08-21:
Great that is works.

> What do you mean by a feature of Rhasspy speaking with node red?

You mentioned this before: "but i also would love to change the colors, when rhasspy is saying something."
Without any further code change, you can subscribe to any topic from rhasspy in node-red and send updates to the \<siteid>/led topic to achieve this

> So node red could replace all logic in your module and the satellite could just react to one led topic with no states at all and just update the color on a separate incoming mqtt topic.

That is what is does now

> i'm not yet sure how say and playbytes interact, probably there is always a playBytes when there is a say message and we don't need say at all?

Correct. But if a Speaking state is wanted, the code should react on the say topic.

> for the Listening state (2) you probably already use
No, that is the hotword detected state. I do not use the topics from asr, those are not needed. The state change to HotwordDetected is on the hermes/hotword/toggleOn and hermes/hotword/toggleOff.

All that stuff is already in place, I am only thinking about adding an extra state Speaking, but it might actually bite the playAudio state./

Romkabouter, 2021-08-28:
Hi,

Check my latest merge of the development branch.
I am working on the Matrix Voice audio playback, but I also created an animation so I had to add that code as well.

The code now goes in or out a "Speaking" state when a ttsSay reason is a toggleOn or toggleOff message.
When in Speaking state, a call to device->animate is done in the while loop for the audio playing.
This is needed, because otherwise the animating stops while playing the audio.

Please flash and let me know what you think, for now I have a simple OTA color led rotation thingy.
I will add a specific color enum later, but I wanted to share it with you to get your opinion on the current behaviour.
Personally, I like it. Ik does not animate with a normal message, only in the speaking state

WhiteSockedDancer, 2021-08-30:
Hey,

first: 
thx for working on the audio playback. since i switched from direct audio-output of the RPI to matrix voice my assistant sounds like he has a cold.. probably an increase of the buffer would help but i'm not sure if the hardware is just not good enough for clear sound.. (Same USB speakers as on the RPI in use with the aux output)

With he update the sound output changed, but there is now another kind of "cracking" noise in it.

---

Now to the topic of this issue:

I don't get any animation or color change when rhasspy is speaking, can you tell me how to configure it right?

But my toggleOn  and toggleOff messages also don't contain "reason": "ttssay". They contain:

On Wakeup-Phrase:
toggleOff -> {"siteId": "matrixvoice", "reason": "playAudio"}
toggleOn -> {"siteId": "matrixvoice", "reason": "playAudio"}

On Listening:
toggleOff -> {"siteId": "matrixvoice", "reason": "dialogueSession"}

On Error-Phrase:
On Wakeup-Phrase:
toggleOff -> {"siteId": "matrixvoice", "reason": "playAudio"}
toggleOn -> {"siteId": "matrixvoice", "reason": "playAudio"}

When changing this in your code it works for wakeup phrase.
(StateMachine.hpp: line 540 and 558) 

My Answer when asked for the time (hermes/tts/say: {"text": "09:14", "siteId": "matrixvoice", "lang": null, "id": "d0c4c448-c065-49cf-8c4a-eca3f0655367", "sessionId": "", "volume": 1.0})
does not evoke the hermes/hotword/toggleOff and then there is no animation.

Probably we have to check for 
toggleOff(playAudio) -> toggleOn(playAudio)
and for
hermes/tts/say -> hermes/tts/sayFinished
?

----

I also love that first Animation! Maybe a little bit to bright for my setup, but a very good first start.

Romkabouter, 2021-08-30:
> thx for working on the audio playback. since i switched from direct audio-output of the RPI to matrix voice my assistant sounds like he has a cold.. pr
> With he update the sound output changed, but there is now another kind of "cracking" noise in it.

I am working on that. If you output 16000 hz (mono or stereo) it should sound fine :)
For the rest I am working on the resampling, so WIP. I found that 16000 played good quality, so I am aiming on resampling because I have tried various methods for setting the sampleRate, but none seem to work on the Matrix Voice. The other devices are much easier :)

The animation and state switching assumes normal Rhasspy usage.
Animation occurs when in state Speaking AND there are bytes to play.
So your messages do not trigger it, the code is not subscribed to hermes/tts/say

The ttsSay is as reason for the hotword/toggleOn and hotword/toggleOff messages, with this reason the statemachine enters the Speaking state. Which topics do you refer to with toggleOn and toggelOff?
When th actual bytes are recieved from Rhasspy on playBytes topic, a call is done to the animate() function while playing those bytes.

I could add extra topics to subscribe to for determining when to enter Speaking state, I was indeed thinking about hermes/tts/say and hermes/tts/sayFinished. I will add those.
But still, the animation will only take place when actual bytes are being played. I can start it ealier if you want, but important to understand is the animation call is also done in the audio play. The reason for this is simple. There is a while loop in the playBytes. When animate() is not called in that while loop, the animation stops when audio is played.
The while loop is a must however, so that is why the call is done.

Romkabouter, 2021-08-30:
> * Idle
> * Wakeup Sound/Phrase (in my case its a wav thats saying: "how can i help" (playBytes))
> * Listening (at the moment what the "wakeword state is tracking"
> * Error (if nothing useful is captured, in my case there is also a .wav saying "what?" (playBytes))
> * Answer (say or playBytes)

So with the above points, where do you see an animation happening in your ideal situation?

> We could even implement different states for playBytes and say (i'm not yet sure how say and playbytes interact, probably there is always a playBytes when there is a say message and we don't need say at all?)

playBytes is not a state as such, it plays bytes in whatever state (like in HotwordDetected or Idle)
I have added a reacting on the hermes/tts/say and sayFinished.
The reaction on sayFinished is to go to Idle
https://github.com/Romkabouter/ESP32-Rhasspy-Satellite/pull/72

Romkabouter, 2021-08-30:
> Probably we have to check for
> toggleOff(playAudio) -> toggleOn(playAudio)

That is not such a good solution I guess. Because with a Rhasspy usage, that would trigger an animation for all audio not just for speech.

Question, why do your messages not contain ttsSay as reason? You are using some custom logic?
I am willing to implement so sort of support for non-native Rhasspy usage, but this must not break the native usage

WhiteSockedDancer, 2021-08-31:
> I am working on that. If you output 16000 hz (mono or stereo) it should sound fine :)
> For the rest I am working on the resampling, so WIP. I found that 16000 played good quality, so I am aiming on [...]

I switched to 16k and its a very good quality compared! Thanks for the hint.

> The animation and state switching assumes normal Rhasspy usage.
> Animation occurs when in state Speaking AND there are bytes to play.
> So your messages do not trigger it, the code is not subscribed to hermes/tts/say

1. Wakeup Sound, Recorded Sound, Error Sound:
First of i use the normal Rhasspy Sounds for Wake WAV, Recorded WAV and Error WAV. 
I use .wav i generated with the say command to get my assistant to ask me, how he could help.
With your code there is no animation when these 3 sounds are played.
to get a "speaking" animation there i think we need to react on [hermes/asr/toggleOff] with payload of{"siteId": "matrixvoice", "reason": "playAudio"}.
I understand, that many people probably use some kind of sound there instead of speaking, but i would also prefer any kind of visual feedback there.

2. Answer via ttssay:
Second i use Node-Red to send ttssay actions to rhasspy via http-api: /api/text-to-speech. This now works if i send a string from there in a test scenario. If i ask any question it seems like matrixvoice is still in listening or recorded state and therefor doesn't play the animation.

3. summary for the ideal solution
> So with the above points, where do you see an animation happening in your ideal situation?

In my ideal situation i would love to see animations:
* Idle   => No Animation
* Wakeup Sound/Phrase => Animation that Rhasspy is reacting to my wakeword
* Listening => Different Color or Animation that signals, that Rhasspy is listening
* Error => Probably just a red "error" animation
* Answer (say or playBytes) => some animation that could be similar to the wakeup sound since rhasspy is speaking with me.

> Question, why do your messages not contain ttsSay as reason? You are using some custom logic?
> I am willing to implement so sort of support for non-native Rhasspy usage, but this must not break the native usage.
> If your On Wakeup-Phrase is changes with a ttsSay as reason, it would probably work right?
> I'm trying to find how you are using Rhasspy

I totally agree. I hope i could explain a little bit more how i use rhasspy in point 1 and 2.

Romkabouter, 2021-08-31:
> 1. to get a "speaking" animation there i think we need to react on [hermes/asr/toggleOff] with payload of{"siteId": "matrixvoice", "reason": "playAudio"}.

That will trigger an animation on every sound in every state. That is possible, but then I need to have a switch for users to animate yes or no in the various states. 

> * but i would also prefer any kind of visual feedback there.

You are aware that there is a different set for COLORS_HOTWORD?

> 2\. If i ask any question it seems like matrixvoice is still in listening or recorded state and therefor doesn't play the animation.

Correct, after a say, Rhasspy sends a sayFinished which triggers the IdleEvent.

> * Idle   => No Animation
> * Wakeup Sound/Phrase => Animation that Rhasspy is reacting to my wakeword
> * Listening => Different Color or Animation that signals, that Rhasspy is listening
> * Error => [...]

Ok, sounds like you want to be able to use an animation in every state.
But in general audioPlay is not a state switch, audio is played during a state.
The Wakeup Sound/Phrase is played during the HotwordDetected state, which you call the Listening state.
So we need an animation during the playBytes in HotwordDetected and then return to no animation when the audio is done.
We need the possiblity to have an animation during playBytes in every state and then a set of colors per state.
This is already somewhat into place for the Speaking state.

Currently it is:
* Idle -> show COLORS_IDLE (configurable via the siteid/led topic)
* HotwordDetected (aka Listening) -> show COLORS_HOTWORD (configurable via the siteid/led topic)
* Tts audio -> COLORS_OTA animated (WIP, there should be a separate COLORS_SPEAK or something)
* Back to Idle (COLORS_IDLE)

I need to implement functionality to add animation possibilities on every state. Obviously configurable via MQTT.
That is possible and also what I have already given some thoughts.
Besides that, we seem to have a need for an ERROR state, which is now missing.
Also, we do not need hermes/asr/toggleX, but only hermes/hotword/toggleX. The hotword messages are the same as the asr messages, only sooner.

WhiteSockedDancer, 2021-08-31:
> That will trigger an animation on every sound in every state. That is possible, but then I need to have a switch for users to animate yes or no in the various states.

Probably the most flexible solution.

 
> > * but i would also prefer any kind of visual feedback there.
> 
> You are aware that there is a different set for COLORS_HOTWORD?

Sure, i'm already using this set heavily since the beginning :)

> > 2. If i ask any question it seems like matrixvoice is still in listening or recorded state and therefor doesn't play the animation.
> 
> Correct, after a say, Rhasspy sends a sayFinished which triggers the IdleEvent.

Hm.. but that is a problem. Following Example:

COLORS_IDLE
Me: "Jarvis?"

Jarvis: "How can i help?" (no animation yet, but should be possible with hermes/asr/toggleOff] with payload of{"siteId": "matrixvoice", "reason": "playAudio"} --> COLORS_PLAYSOUND?)

Jarvis is listening (COLORS_HOTWORD)

Me: "What's the time?"

Jarvis: "<Sound for recorded> e.g. 'Understood' (or nothing in my case)" (Probably COLORS_PLAYSOUND?)

Jarvis: "It's 9 o'clock" (COLORS_SPEAK, but that is not triggered yet, since its overlapping with the recorded or listening state?
Why do i not get this animation? If i trigger this phrase from elsewhere i get the animation.

> Ok, sounds like you want to be able to use an animation in every state.
> But in general audioPlay is not a state switch, audio is played during a state.
> The Wakeup Sound/Phrase is played during the [...]

As i described before, i'm not sure, why the animation is not playing when i'm going into the speaking state when an intent is recognized (i thought by then the "hotwordDetected" state should be finished already)

> Currently it is:
>
>     * Idle -> show COLORS_IDLE (configurable via the siteid/led topic)
>
>     * HotwordDetected (aka Listening) -> show COLORS_HOTWORD (configurable via the siteid/led topic)
>
>     * [...]

nice! exactly my thoughts.

> Besides that, we seem to have a need for an ERROR state, which is now missing.

would be nice to have

> Also, we do not need hermes/asr/toggleX, but only hermes/hotword/toggleX. The hotword messages are the same as the asr messages, only sooner.

i agree.

Romkabouter, 2021-09-01:
> Jarvis: "It's 9 o'clock" (COLORS_SPEAK, but that is not triggered yet, since its overlapping with the recorded or listening state?
> Why do i not get this animation? If i trigger this phrase from elsewhere i get the animation.

That is strange, can you post the messages you get in the correct order from hermes/hotword/toggleOn, hermes/hotword/toggleOff, hermes/tts/# and hermes/dialogueManager/# ?

> (i thought by then the "hotwordDetected" state should be finished already)

That is why I'd like to have a log of your messages
HotwordDetected can flow to Speaking or Idle
* message on toggleOff with reason "ttsSay" -> flow to Speaking. playBytes should then animate.
* message on toggleOn with reason "ttsSay" or "dialogueSession" -> flow to Idle

This is my flow:
* hermes/hotword/toggleOff -> reason playAudio -> ignored
* hermes/hotword/toggleOn -> reason playAudio -> ignored
* hermes/dialogueManager/sessionStarted -> ignored
* hermes/hotword/toggleOff -> reason dialogueSession -> flow from Idle to HotwordDetected
* ---- listening ------
* hermes/hotword/toggleOn -> reason dialogueSession -> flow from HotwordDetected to Idle
* hermes/dialogueManager/endSession -> text repsonse from Intent
* hermes/hotword/toggleOff -> reason ttsSay-> flow from Idle to Speaking
* hermes/tts/say -> same text message as endSession -> tiggers SpeakEvent but already in Speaking so ignored
* hermes/tts/sayFnished -> flow from Speaking to Idle
* hermes/hotword/toggleOn -> reason ttsSay-> tiggers IdleEvent but already in Idle so ignored
* hermes/dialogueManager/sessionEnded -> ignored
* hermes/hotword/toggleOn -> reason dialogueSession  -> tiggers IdleEvent but already in Idle so ignored

You probably have some messages extra for the playing of the recorded sound, I will check into that as well.
The time between the first two playAudio messages are the length of the wakeup sound.
So I can create an extra state for that if that helps. 
* hermes/hotword/toggleOff -> reason playAudio -> Idle to PlayAudio
* hermes/hotword/toggleOn -> reason playAudio -> PlayAudio to Idle

Romkabouter, 2021-09-01:
I have added a recorded sound, two extra messages are received as expected.
After ---- listening --- and before the flow from HotwordDetected to Idle:
* hermes/hotword/toggleOff -> reason playAudio
* hermes/hotword/toggleOn -> reason playAudio
This is the: Jarvis: " e.g. 'Understood' (or nothing in my case)" (Probably COLORS_PLAYSOUND?)

WhiteSockedDancer, 2021-09-03:
> This is needed for several state transitions, I have a good idea on how to change it and started working on it. I will push my commits to the matrixvoice_output branch so it might be best that you checkout that branch as well

i switched to the matrixvoice_output branch, but now i don't have any animations at all. 
Just idle and wakeword colors are showing.

> That is why I'd like to have a log of your messages

hermes/hotword/jarvis_linux/detected
hermes/hotword/toggleOff -> "reason": "playAudio"
hermes/asr/toggleOff -> "reason": "playAudio"
hermes/audioServer/matrixvoice/playBytes/3899d8b2-487f-4c55-bedf-d89e9bd4e2d6 -> ("How can i help?")
hermes/audioServer/matrixvoice/playFinished
hermes/hotword/toggleOn -> "reason": "playAudio"
hermes/asr/toggleOn -> "reason": "playAudio"
hermes/dialogueManager/sessionStarted
hermes/hotword/toggleOff -> "reason": "dialogueSession"
hermes/asr/startListening 

Now i'm asking what time it is.

rhasspy/asr/recordingFinished
hermes/asr/textCaptured
rhasspy/asr/matrixvoice/matrixvoice/audioCaptured
hermes/asr/stopListening
hermes/hotword/toggleOn -> "reason": "dialogueSession"

hermes/nlu/query
hermes/nlu/intentParsed
hermes/intent/GetTime
hermes/dialogueManager/endSession (I think the difference between us lays here.. i send the end session dialog from node-red if i recognize any intent. i implemented that, because my sessions always lasted very long until ended by a timeout and i could not say something in these 30? seconds. If i disable this message the animation for ttssay is correctly shown! Would you send this message after the say-command?)

hermes/tts/say
hermes/asr/stopListening
hermes/dialogueManager/sessionEnded
hermes/hotword/toggleOn
hermes/audioServer/matrixvoice/playBytes/ca255c90-22d5-4afe-8122-7f65175bb135 -> "It's x o'clock.."
hermes/audioServer/matrixvoice/playFinished
hermes/tts/sayFinished

---

When i'm just sending a "say time" from node-red i get the animation.

Flow for this:
hermes/tts/say
hermes/audioServer/matrixvoice/playBytes/bffa5c5a-27e8-4562-9372-e41c495dd5a5
hermes/audioServer/matrixvoice/playFinished
hermes/tts/sayFinished

Romkabouter, 2021-09-03:
> i switched to the matrixvoice_output branch, but now i don't have any animations at all.

ok, that is strange because I actually coded in this branch and pushed to dev.

> (I think the difference between us lays here.. i send the end session dialog from node-red if i recognize any intent.

Ah... wait a minute.
So you send a message to endSession, and that also a message to ttsSay for the response?
That is not needed at all.

The reason you got the timeout is indeed because the dialogueManager did not get a timely reponse from an intenthandler.
So, you IntentHandler should send a message so hermes/dialogueManager/endSession.
But you can add "text" to the payload, which is then processed by Rhasspy to the tts

If you use Node-Red as intent handler, try adding "text":"It's x o'clock.." to your endSession payload.
Then you will not need to send a message to ttsSay

WhiteSockedDancer, 2021-09-03:
> ok, that is strange because I actually coded in this branch and pushed to dev.

When delaying the endSession-Message it works and i get an animation with the ttssay

> Ah... wait a minute. ....

I would love to do it this way, but i got many intents that do alot of stuff before sending an actual text to rhasppy (even just sending a text if my voiceDND-variable is not true).

I guess i can reprogram this logic but at the moment i just use rhasspy as one of many Input-Modules besides dashboards, buttons, smartphone, smartwatch.. 
So my intents not always come with an rhasspy-session to begin with.
I have to think about how to build this maybe better.

> 

but i dont understand, why the animation is not played if i ended that session and then send a 'independent' say through my logic.
if i send an independent say without any dialog before it works like a charm.

---

>     * hermes/hotword/toggleOff -> reason playAudio -> ignored

So now i get the animations when rhasspy is using ttssay. :) 
But how do we proceed with the playAudio for  wake recorded and error .wav?
At the moment there is no animation or colorchange?

Romkabouter, 2021-09-03:
> but i dont understand, why the animation is not played if i ended that session and then send a 'independent' say through my logic.
> if i send an independent say without any dialog before it works like a charm.

I think because of the ordering, currently the animation only plays in Speaking State
* hermes/tts/say -> Enter **Speaking** 
* hermes/asr/stopListening -> ignored
* hermes/dialogueManager/sessionEnded -> ignored 
* hermes/hotword/toggleOn -> reason is probably "dialogueSession"? -> to **Idle**
* hermes/audioServer/matrixvoice/playBytes/ca255c90-22d5-4afe-8122-7f65175bb135 -> "It's x o'clock.." -> **in Idle, no animation**
* hermes/audioServer/matrixvoice/playFinished -> outgoing from code
* hermes/tts/sayFinished -> to **Idle**, but already there

So you need to find a way to publish the hermes/tts/say after the hermes/dialogueManager/sessionEnded.
That way, the code re-enters Speaking when tts/say is received

> But how do we proceed with the playAudio for wake recorded and error .wav?
> At the moment there is no animation or colorchange?

Yes, I am giving that some thoughts 👍

Romkabouter, 2021-09-03:
> I have to think about how to build this maybe better.

I think so, maybe build some logic to be able to determine if the call is made by a Rhasspy session. 
Obviously I do not know how complex it is, but sending a message to endSession with a text always works I think, even if no session was started. I tried that and I remember it working.
So you could find some way to post all your speech feedback to endSession, elimintating the tts/say part?

Romkabouter, 2021-09-04:
I have been studying the messages both with and without audiofeedback and I have found a good way for this.
Some extra states will be added, as well as some extra topics to subscribe to. I will refactor HotwordDetected to Listening, because that is a more appropriate name.
Extra states
* IDLEPLAY
* LISTENINGPLAY
* TTS
* TTSPLAY
* ERROR
* ERRORPLAY
In all states (including existing) there will be some logic to support both static colors and animation.
The states with PLAY are there to support animation or static colors during the audioPlay.
The statemachine will know what PLAY state because when in IDLE as playAudio on toggleOff will trigger a PlayAudioEvent. The reaction in IDLE is to flow to IDLEPAY. 
playAudio on toggleOn will trigger a another event. The reaction in IDLEPLAY is to flow to IDLE. Likewise in all other states.

Analyis:

```
IDLE - All sounds
hermes/hotword/toggleOff + hermes/asr/toggleOff {"siteId": "matrixvoice", "reason": "playAudio"}        IDLEPLAY
hermes/audioServer/matrixvoice/playBytes        WAKE sound
hermes/hotword/toggleOn + hermes/asr/toggleOn   {"siteId": "matrixvoice", "reason": "playAudio"}        IDLE
hermes/dialogueManager/sessionStarted           {"sessionId": "matrixvoice-snowboy_linux-6046cba4-8fbf-4cd0-95e1-43c5eafcc6fe", "siteId": "matrixvoice", "customData": "snowboy_linux", "lang": null}
hermes/hotword/toggleOff                        {"siteId": "matrixvoice", "reason": "dialogueSession"}  LISTENING
hermes/asr/startListening                       {"siteId": "matrixvoice", "sessionId": "matrixvoice-snowboy_linux-6046cba4-8fbf-4cd0-95e1-43c5eafcc6fe", "lang": null, "stopOnSilence": true, "sendAudioCaptured": true, "wakewordId": "snowboy_linux", "intentFilter": null}
hermes/asr/textCaptured                         {"text": "wat voor weer wordt het", "likelihood": 1.0, "seconds": 1.2748125350335613, "siteId": "matrixvoice", "sessionId": "matrixvoice-snowboy_linux-6046cba4-8fbf-4cd0-95e1-43c5eafcc6fe", "wakewordId": null, "asrTokens": [[{"value": "wat", "confidence": 1.0, "rangeStart": 0, "rangeEnd": 4, "time": {"start": 0.0, "end": 0.63}}, {"value": "voor", "confidence": 1.0, "rangeStart": 4, "rangeEnd": 9, "time": {"start": 0.63, "end": 0.63}}, {"value": "weer", "confidence": 1.0, "rangeStart": 9, "rangeEnd": 14, "time": {"start": 0.63, "end": 1.05}}, {"value": "wordt", "confidence": 1.0, "rangeStart": 14, "rangeEnd": 20, "time": {"start": 1.05, "end": 1.11}}, {"value": "het", "confidence": 1.0, "rangeStart": 20, "rangeEnd": 24, "time": {"start": 1.11, "end": 2.07}}]], "lang": null}
hermes/hotword/toggleOff + hermes/asr/toggleOff {"siteId": "matrixvoice", "reason": "playAudio"}        LISTENINGPLAY
hermes/audioServer/matrixvoice/playBytes        WAKE END
hermes/hotword/toggleOn + hermes/asr/toggleOn   {"siteId": "matrixvoice", "reason": "playAudio"}        LISTENING
hermes/asr/stopListening                        {"siteId": "matrixvoice", "sessionId": "matrixvoice-snowboy_linux-6046cba4-8fbf-4cd0-95e1-43c5eafcc6fe"}
hermes/hotword/toggleOn                         {"siteId": "matrixvoice", "reason": "dialogueSession"}  IDLE
hermes/nlu/query
hermes/nlu/intentParsed
hermes/dialogueManager/endSession               {"sessionId": "matrixvoice-snowboy_linux-6046cba4-8fbf-4cd0-95e1-43c5eafcc6fe", "text": "Morgen wordt het lekker"}
hermes/hotword/toggleOff + hermes/asr/toggleOff {"siteId": "matrixvoice", "reason": "ttsSay"}           TTS
hermes/tts/say                                  {"text": "Morgen wordt het lekker"}                     TTSPLAY
hermes/audioServer/matrixvoice/playBytes        SPEAKING TTS
hermes/tts/sayFinished                          {"siteId": "matrixvoice""}                              
hermes/hotword/toggleOn + hermes/asr/toggleOn   {"siteId": "matrixvoice", "reason": "ttsSay"}           TTS
hermes/asr/stopListening                        {"siteId": "matrixvoice", "sessionId": "matrixvoice-snowboy_linux-6046cba4-8fbf-4cd0-95e1-43c5eafcc6fe"}
hermes/dialogueManager/sessionEnded             {"termination": {"reason": "nominal"}"}                 
hermes/hotword/toggleOn                         {"siteId": "matrixvoice", "reason": "dialogueSession"}  IDLE

IDLE - No sounds
hermes/dialogueManager/sessionStarted           {"sessionId": "matrixvoice-snowboy_linux-6046cba4-8fbf-4cd0-95e1-43c5eafcc6fe", "siteId": "matrixvoice", "customData": "snowboy_linux", "lang": null}
hermes/hotword/toggleOff                        {"siteId": "matrixvoice", "reason": "dialogueSession"}  LISTENING
hermes/asr/startListening                       {"siteId": "matrixvoice", "sessionId": "matrixvoice-snowboy_linux-6046cba4-8fbf-4cd0-95e1-43c5eafcc6fe", "lang": null, "stopOnSilence": true, "sendAudioCaptured": true, "wakewordId": "snowboy_linux", "intentFilter": null}
hermes/asr/textCaptured                         {"text": "wat voor weer wordt het", "likelihood": 1.0, "seconds": 1.2748125350335613, "siteId": "matrixvoice", "sessionId": "matrixvoice-snowboy_linux-6046cba4-8fbf-4cd0-95e1-43c5eafcc6fe", "wakewordId": null, "asrTokens": [[{"value": "wat", "confidence": 1.0, "rangeStart": 0, "rangeEnd": 4, "time": {"start": 0.0, "end": 0.63}}, {"value": "voor", "confidence": 1.0, "rangeStart": 4, "rangeEnd": 9, "time": {"start": 0.63, "end": 0.63}}, {"value": "weer", "confidence": 1.0, "rangeStart": 9, "rangeEnd": 14, "time": {"start": 0.63, "end": 1.05}}, {"value": "wordt", "confidence": 1.0, "rangeStart": 14, "rangeEnd": 20, "time": {"start": 1.05, "end": 1.11}}, {"value":  [... cut ...]
hermes/audioServer/matrixvoice/playBytes        ERROR SOUND
hermes/hotword/toggleOn + hermes/asr/toggleOn   {"siteId": "matrixvoice", "reason": "playAudio"}        ERROR
hermes/asr/stopListening                        {"siteId": "matrixvoice", "sessionId": "matrixvoice-snowboy_linux-6046cba4-8fbf-4cd0-95e1-43c5eafcc6fe"}
hermes/dialogueManager/sessionEnded             {"termination": {"reason": "intentNotRecognized"}"} 
hermes/hotword/toggleOn                         {"siteId": "matrixvoice", "reason": "dialogueSession"}  IDLE
```

WhiteSockedDancer, 2021-09-04:
Nice. That sounds really good. I'm looking forward to test these new states!!

WhiteSockedDancer, 2021-09-04:
If i'm thinking ahead, how do you realize the animations or do you consider adding different animations or options to customize them? (I guess you also have already alot of different ideas, but i can just share some thoughts for redundancy ;D)
Btw i'm totally fine with just one animation or even just solid changing colors. Of course these features are awesome and I really appreciate your work!

So are there Animation Start, Loop, Animation Stop - Sequences?

---
Simple Implementation:
There is just one Animation and you just can choose the color and brightness. Would be enough for most people :)

---

Advanced Animations:

For the customization part. What and how stuff can be changed:
Animation Type, Colors, Speed

Maybe you can choose an animation type for a STATE, 1 to n different colors (depending on animation type) and speed.

If you choose "Static/Solid/none" as animation Type it just takes one color.
If you choose "blinking" it takes n colors and switches between them in the frequency of speed.

Current animation could be "running light" it takes 2 colors. One for the backlight and one for the brighter running dot that is been fading out to the first color. usually you could just take the same color with different brightness-levels to get the current effect.

if you dont provide enough colors it will just take white as placeholder. If you provide to many colors, overflow is going to be ignored. if you dont choose a valid AnimationType, it just uses "Solid"

Romkabouter, 2021-09-04:
> do you consider adding different animations or options to customize them?

Yes, I am thinking about a set of animations to choose from. And every state can have its own type.
Colors per state should also be configurable, so that implies a fair amount of work.
Your ideas are a bit what I have been thinking about. 
For now I thought of four types: 
* solid (no animation effectively, but there should not be flickering)
* running light (good name)
* pulsing (1 color, fading in and out)
* blinking (like pulsing, but without fading effect)

> So are there Animation Start, Loop, Animation Stop - Sequences?

Not yet, but I am aiming to start simple. I am starting with the Simple Animation for now. If I can get the whole state switching with correct animations we can proceed with the more advanced stuff. It should also be configurable via webUI and MQTT, so there lies a challenge as well.

Romkabouter, 2021-09-07:
Ok cool, I have got something going here.
* simplified all states with default behaviour in some events
* added Error on hermes/nlu/intentNotRecognized
* renamed HotwordDetected to Listening
* added a \<state>Play for every state

The expected behaviour is now:
* switch to the \<state>Play when a playBytes message is received (we actually do not need a playAudio reason for that)
* switch back to \<state> when the playBytes is done.

For now I have not added extra colors, but I have added extra logging to the \<siteId>/debug topic.
That is currently default true.

If you subscribe to that topic, you can see all the different state transitions. I think I got it right, but would like to hear your comments.
If you have a MQTT tool like MQTTfx, you can subscribe to these topics and you will see the messages from Rhasspy and the debug statements with the transitions
* hermes/nlu/intentNotRecognized
* hermes/dialogueManager/#
* hermes/hotword/#
* hermes/tts/#
* hermes/intent/# (not really needed)
* hermes/audioServer/matrixvoice/playBytes/#

You should see something like this, notice all the debug messages in between.
[image]

WhiteSockedDancer, 2021-09-08:
Hey there. Nice work!

I got it running with no errors. Only interesting message i didnt quite understand was this:
[image]

Why is there an "IdleEvent in Listening" after the parsed intent?

Also later there is an "IdleEvent in Tts" after the sayFinished of my answer ("its x o'clock") just before entering Idle.

I'm on vacation for lil' bit over one week now, but i would love to support and test more when i'm back!

Cant wait to get some custom colors going for when Jarvis talks back to me! :)
Keep up your good work

Romkabouter, 2021-09-08:
> Why is there an "IdleEvent in Listening" after the parsed intent?

Hmm, I do not seem to have to that event there.

> Also later there is an "IdleEvent in Tts" after the sayFinished of my answer ("its x o'clock") just before entering Idle.

That looks as expected to me, because you are in the Tts state at that moment

I will soon push some colors per state, and then animate them in the Play state with that simple running light

Romkabouter, 2021-09-09:
Correction, I indeed get the IdleEvent in Listening and I am checking it out

Romkabouter, 2021-09-10:
I am thinking on implementing state transitions on playAudio reason as well.
Because now, Rhasspy plays the wake sound before sending the message with dialogueSession reason.
So when the wake sound is played, the state is still Idle and not Listening. This is strange behaviour in my opinion.

WhiteSockedDancer, 2021-09-20:
So i'm back.
trying the latest version in the matrixvoice_output branch got me some strange behaviours:

1. led is not changing dynamically anymore when sending new idle colors.
2. my wake .wav is not being played (after restarting rhasspy it worked for one try and i could hear the wake .wav once)

Romkabouter, 2021-09-22:
Yes, I also saw those issues and thought it was my network. I will trace some of the steps back :\

WhiteSockedDancer, 2021-09-24:
I went back some commits.

20d0d235c10049447a6aacb8bb824aa4ba61d4ea
from Aug 28, 2021 seems okay. But afterwards dynamic change of idle-color doesnt work :) The error seems to be somewhere between this and the next commit (090e7cc9553835b002170182daee636a100e81af)

Romkabouter, 2021-09-24:
Good find, thanks :)