Romkabouter/ESP32-Rhasspy-Satellite #70: Supported Esp models

Floplosion05, 2021-08-24:
Just a quick question, are there any particular reasons, why I couldn't use an esp8266 as a rhasspy satellite?

Romkabouter, 2021-08-25:
> Just a quick question, are there any particular reasons, why I couldn't use an esp8266 as a rhasspy satellite?

esp8266 is a single core board, so this code will  not work on that.

> So i wanted to know does the device constantly stream audio to a Rhasspy base station and the wakeword and intent detection is done there or do I have to trigger the stream by pressing or holding a button?

There is a constant stream to Rhasspy in any case, but you can force a wakeword trigger with a button.

closed 2021-08-25 by Romkabouter

Floplosion05, 2021-08-25:
What setup would u recommend, to get started with ur library. I've done some research and the M5Atom Echo seems to be my preference, because it comes fully assembled and i only have to configure and flash it, right? Do u have any experience with the M5 in particular, how is the microhpone quality and the audio quality of the speakers?

Romkabouter, 2021-08-25:
I have a M5 as well, this was the reason for my rewrite because I wanted to support multiple devices.
The M5 Atom Echo is a device, although the recording level seems to be low. (Not the quality, but the sound level)
Speakers are ok for such a small device :)

Follow the getting started and you should be ok:
https://github.com/Romkabouter/ESP32-Rhasspy-Satellite/blob/master/m5atomecho.md

Floplosion05, 2021-08-25:
So will it be able to understand me from a few meters away (with understand i mean capture a loud enough audio input for Rhasspy to understand)?

Romkabouter, 2021-08-25:
Highly depends on the environment, but a few meters works for me.