Romkabouter/ESP32-Rhasspy-Satellite #86: Make remote hotword detection optional

hogthrob, 2021-12-28:
Hi, 
as I understand from following up the discussion from #30 , at the moment a local hotword detection is not implemented so remote hotword detection is now the only option. However, there is already local activation via a key press supported. 

I propose to enable this as a choice, i.e. offer two settings for activation of Rhasspy voice recognition: `Key Press Activation Only` and `Remote Detection + Key Press`. 
Changes to existing code are very limited, it uses the existing HW_REMOTE/HW_LOCAL concept. I have the code working locally. 

Shall I submit a pull request or is this out of scope?

Romkabouter, 2021-12-28:
Hi there, that is a good idea :)

Thanks for the effort

Romkabouter, 2021-12-29:
I actually have this somewhat working for local hotword https://www.edgeimpulse.com/blog/tutorial-keyword-spotting
But I have to work things out a bit more

closed 2021-12-29 by Romkabouter