Romkabouter/ESP32-Rhasspy-Satellite #92: Matrix Voice esp32 version volume lowering on (almost) each activation when on speakers

Dimitar-Boychev, 2022-01-04:
Hello,
I have downloaded and compiled the latest version.
Cleared the flash and flashed the new version.
Using rhasspy 2.5.11 with internal mqtt.
If I run the matrix voice with "headphones" output there are no problems.
However if I switch to speakers, interesting things start to happen.
Volume is on 100% never touched since flashing.
At the first and sometimes the second activation the volume is loud.
However on every subsequent audio played (I notice this on the wake wav, recored wav and error wav) the volume gradually lowers up to around 20% maybe...
The web interface still shows 100%, if I switch to headphones at this time the volume is low.
If I switch to headphones and restart... no issues for as many activations as I want ... If I switch to speakers same issue starts....

Romkabouter, 2022-01-07:
Interesting, I will see if I can reproduce this!

Romkabouter, 2022-04-30:
Hmm, I cannot reproduce this yet. I will try it with a clean flash

Romkabouter, 2022-05-01:
@Dimitar-Boychev can you check it this is still an issue?
I could play a news broadcast of about 2 minutes without any volume lowering. Also repeatedly waking and processing intents worked as expected.

mansia, 2022-05-07:
Hello, same problem here.

Rhasspy 2.5.11 in MQTT Audio with Matrix Voice ESP32.
Each time I unplug/plug (meaning powering off/on) the satellite, it's ok. Then sound is systematically lowering to become inaudible whatever the volume set. Only with speakers, headphones have no problem. Speakers are 2x 8ohm 1W.

mansia, 2022-05-07:
Rhasspy Conf here below
Used by Jeedom as domotic solution for handling events.
{
    "command": {
        "webrtcvad": {
            "vad_mode": "2"
        }
    },
    "dialogue": {
        "system": "rhasspy",
        "volume": "0.4"
    },
    "handle": {
        "remote": {
            "url": "http://192.168.1.12/core/api/jeeApi.php?apikey=xxxx&plugin=jeerhasspy&type=event"
        },
        "system": "remote"
    },
    "intent": {
        "system": "fsticuffs"
    },
    "microphone": {
        "system": "hermes"
    },
    "mqtt": {
        "enabled": true,
        "host": "192.168.1.12",
        "password": null,
        "site_id": "mvsatellite",
        "username": null
    },
    "sounds": {
        "system": "hermes"
    },
    "speech_to_text": {
        "deepspeech": {
            "open_transcription": true
        },
        "kaldi": {
            "cancel_word": "annule",
            "mix_weight": "0.05"
        },
        "satellite_site_ids": "mvsatellite",
        "system": "kaldi",
        "vosk": {
            "open_transcription": true
        }
    },
    "text_to_speech": {
        "nanotts": {
            "volume": "0.5"
        },
        "system": "nanotts"
    },
    "wake": {
        "raven": {
            "keywords": {
                "hermes": {
                    "enabled": true
                }
            }
        },
        "satellite_site_ids": "mvsatellite"
    },
    "webhooks": {
        "awake": [
            "http://192.168.1.12/core/api/jeeApi.php?apikey=xxxx&plugin=jeerhasspy&type=event"
        ]
    }
}

mansia, 2022-05-07:
It seems to correspond to this conversation https://community.rhasspy.org/t/matrix-voice-esp32-speaker-output/571/12 you already took part to.

Romkabouter, 2022-05-07:
@mansia I see you have Rhasspy volume set to 0.4, I'll try that as well. Since I do have this issue and volume on my Rhasspy is 1. Can't really relate to the issue, but clutching at straws here.

You are on the latest release right?

mansia, 2022-05-08:
@Romkabouter I upgraded to latest (but I think it was already) through OTA. Still same issue.
One noticeable thing : it happens only after the satellite has been waked up at least once from Rhasspy. If I send some TTS before waking up the matrix, the sound is ok. As soon as I woke it up and recognized some intent, the problem appears. 

I have a second matrix voice still in its box. I'm going to upload the exact same firmware and test to see if it's not a board problem. I'll let you know.

Romkabouter, 2022-05-08:
@mansia tried with volume set to 0.4. No issues here :\
I will upload a video to youtube to show a demo

Romkabouter, 2022-05-10:
@mansia I have reset the chip and reïnstalled from scratch. No issues...... As long as I cannot reproduce this, it is hard to fix ;)

mansia, 2022-05-10:
Hello,

I have tried the same installation from scratch on a new hardware, same issue !
It's even weirder than I thought after some other tests :

I connect the satellite to the wifi network and waited a few hours before first activation : no sound problem.
Then, after the first installation, volume decreases until being almost mute, but increases again during tts if sentence is long enough !

I'll record a video and post it.
I'll do some other tests (like switching tts  engine and come back with results).

Thanks

Sent from MailDroid

https://github.com/Romkabouter/ESP32-Rhasspy-Satellite/issues/92#issuecomment-1122158303

Romkabouter, 2022-12-13:
Hi @mansia I still cannot reproduce this, I am sorry.
Do you still have issue or are you using something else now?