Romkabouter/ESP32-Rhasspy-Satellite #97: TTS not Working on Satellite

sascha005, 2022-04-09:
Hi, i have Setup ESP32 Satellite on a ESP32 Audio Kit. Wake Word detection and Word detection works without Problems. But I can't get TTS on the Satellite working. The Strange thing is, if i set the SiteID on the Base to the same ID as the Satellite has, then the TTS goes to the Satellite. Can someone tell me how to configure the Rhasspy Base TTS Service correctly ? 

Regards,
Sascha 
[image]

sehraf, 2022-04-09:
How are you testing this? If you use the "speak" button in the screenshot above, you are **not** sending this to `satellite`

You setting is correct, i'm using this and home assistant can answer to voice commands.
[image]
And when i use the "speak" button, the command is send to `default` (in my case)
[image]

sascha005, 2022-04-09:
Hi, 

yes i try the speak button. But if i say the Wake Word and for example "What Time is it", the Intent is Recocnized on the Base, but i sont hear the Answer on the Satellite. I use Homeassistant too and the Rhasspy HA Addon for the base. Do i have to configure the TTS in Homeassistant and use HA for that ? 
[image]

sehraf, 2022-04-09:
Not sure about the HA addon, i've followed this guid https://github.com/rhasspy/rhasspy/wiki/Intent-Handling-with-HA-and-Events, especially https://github.com/rhasspy/rhasspy/wiki/Intent-Handling-with-HA-and-Events#change-tts-response

Romkabouter, 2022-04-09:
> Hi,
> 
> yes i try the speak button. But if i say the Wake Word and for example "What Time is it", the Intent is Recocnized on the Base, but i sont hear the Answer on the Satellite. 

When using the speak button, TTS will be send to the siteId of the base, that it why it works when you name your satellite the same.

> I use Homeassistant too and the Rhasspy HA Addon for the base. Do i have to configure the TTS in Homeassistant and use HA for that ? !

No, you need to configure events or intents or use Node Red.
Check out this wiki: https://github.com/rhasspy/rhasspy/wiki

That should help you out.
You can always ask about the configuration here: https://community.rhasspy.org/

closed 2022-04-09 by Romkabouter

sascha005, 2022-04-09:
Ah okay, thank you both for pointing me into the right direction. I just trying to use it via events in Homeassistant. If i add the automation like in the first howto link, then i get a error Message like : "Message malformed: extra keys not allowed @ data['data']" in the mqtt publish part.

sascha005, 2022-04-09:
Okay, now i can see the Events comming in in Homeassistant. So  it works now. Thank you again.

AndreasMettlen, 2022-05-17:
@sascha005 : Could you please share your configuration - how did you manage to get the WakeWord detection to work.
it seems my AudioKit is not recognizing any sound at all (allthough I get everchanging messages in mqtt on my hermes/audioServer/<satelite-id> )