Romkabouter/ESP32-Rhasspy-Satellite #99: Problem with AudioKit (A1S) - No supported controller

AndreasMettlen, 2022-05-15:
Hi,
I'm having problems getting my AudioKit (v2.2 A149) to run (or even to properly boot).
From running squeezelite-esp32 on this board I conclude that it uses ES8388 variant 2 (Squeezelite-esp32 runs fine), 
but when I build and try to run the rhasspy satellite I get "No supported controller" during boot and then I'm hung in the neverending loop.

Maybe it has to do with the 5 DIP-switches - is there a needed combination for them to work ?

Code in AudioKit.hpp shows same config as used in squeezelite (except extra GPIO16 for i2c - compare here https://screenzone.eu/esp32-squeezelite-on-esp32-audio-kit/  the part about "dac_config:"

I'm at a loss here - I bought this A1S to surprise my son with a "home-made" Alexa ;-).

kind regards

Romkabouter, 2022-05-16:
Hi @AndreasMettlen, I really cannot help you with this since I do not own such a device.
Might be the dip-switches, because that is what is mentioned here
https://screenzone.eu/esp32-squeezelite-on-esp32-audio-kit/

The code tries the find the correct variant at setup, so it should be able to detect ES8388_V2 but maybe there is an issue with that.
If you are sure you have that variant, open up https://github.com/Romkabouter/ESP32-Rhasspy-Satellite/blob/master/PlatformIO/src/devices/AudioKit.hpp and change the code. Replace lines 184..186 with 
`variant = ES8388_V2;`
and see what happens

hogthrob, 2022-05-16:
Ok, the i2c=16 in dac_config is the I2C address of the ES8388 controller (not a pin). Audiokit uses of course the same address:

https://github.com/Romkabouter/ESP32-Rhasspy-Satellite/blob/9be94b266c32d78ee1e4113681321259d2f272ce/PlatformIO/src/devices/ES8388Control.cpp#L5

So  we are using exactly the same configuration as squezzelite-esp32 in our detection code.

Not sure why detection could fail. And it is not a good idea to just comment out the lines 184..186
The subsequent code assume that the variant is correctly set. And the main problem is, if we cannot start the initial I2C communication not much is going to work as the ES8388 is in this case not really configured for work.

AndreasMettlen, 2022-05-16:
Hi all, I did a clean build and install (pio run -t erase & pio run -t upload) and the monitor shows the following at boot:
`Identifying Codec... Checking for ES8388 Pinout Variant 1...
Checking for ES8388 Pinout Variant 2...
Checking for AC101...
[   122][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
[   125][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
[   137][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
[   149][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
[   162][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
[   275][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
[   287][E][Wire.cpp:313] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
No supported controller found, stopping operation!
`
I haven't changed anything from the config used for squeezelite.
I'm not deep into the code or even C programming - maybe this give anybody a hint...

hogthrob, 2022-05-17:
Thanks for the log. I have never seen a log containing the above statements regarding beginTransmission BEFORE I updated the PlatformIO environment to the newest version of the libraries etc. With the ESP32 tools 3.5.0 it was working, with 4.2.0 it is not working anymore and my device shows the same error messages. So for what it is worth:  The update broke perfectly working application code. Nothing wrong with your hardware or the settings.

I am not be able to look into this right now. Will maybe have time to check its later this week.

hogthrob, 2022-05-17:
Ok, had a quick look and found the issue. See pull request #100.

AndreasMettlen, 2022-05-17:
Big shoutout !!
I works - I got the board up 'n' running - now I just need to fiddle with the different settings on the Server/"Base station"..

MANY THANKS !!!!

@Romkabouter : Test of this fix was successful ;-)

Romkabouter, 2022-05-18:
Fixed with #100

closed 2022-05-18 by Romkabouter