Romkabouter/ESP32-Rhasspy-Satellite #102: Yet another setup question

erte1, 2022-06-08:
Hi all,

I am running a new instance of Rhasspy 2.5.11 in a docker container of a headless Rasp 2b.   Since it is headless, I also used this repo to setup a M5 Atom Echo.   Both are configured to an external MQTT server, but my configuration settings in Rhasspy are not picking up the MQTT feed from the M5 satellite.  

My satellite config is as follows:
[General]
hostname=satellite1
deployhost=192.168.41.104
siteId=satellite1
;supported: M5ATOMECHO=0, MATRIXVOICE=1, AUDIOKIT=2, INMP441=3, INMP441MAX98357A=4, ESP32_POE_ISO=5, TAUDIO=6
device_type=0
;network_type: 0: WiFi, 1: Ethernet
network_type=0

How exactly does the base Rhasspy need to be configured?  I've read through the open/closed issues and have seen some examples but I'm still not able to get it to work.   When should I be using Hermes MQTT as the selection vs. what Rhasspy has recommended?

Also, does the site of Rhasspy need to match the site used in my satellites?   How does the M5 know what topic to subscribe to in order to act as a speaker as well as a mic if these values are different?

thanks in advance for the tips!

Romkabouter, 2022-06-09:
Rhasspy siteId needs to be different from satellite1, maybe call it "base" or "server" or whatever you want.
If no mike is on your server, you can set Audio Recording to "disabled"

Then, in all input boxes named "Satellite siteIds" fill in satellite1 and sent Audio Play to "Hermes MQTT"

erte1, 2022-06-10:
Awesome, thank you! That seemed to do the trick.

For others that may need help, my Rhasspy settings were as follows:

{
    "dialogue": {
        "satellite_site_ids": "satellite1",
        "system": "rhasspy"
    },
    "handle": {
        "satellite_site_ids": "satellite1",
        "system": "hass"
    },
    "home_assistant": {
        "access_token": "XXXXXX",
        "handle_type": "event",
        "url": "http://192.168.XXX.XXX:8123"
    },
    "intent": {
        "satellite_site_ids": "satellite1",
        "system": "fsticuffs"
    },
    "mqtt": {
        "enabled": "true",
        "host": "192.168.XXX.XXX",
        "password": "XXX",
        "site_id": "base",
        "username": "XXX"
    },
    "sounds": {
        "system": "hermes"
    },
    "speech_to_text": {
        "satellite_site_ids": "satellite1",
        "system": "kaldi"
    },
    "text_to_speech": {
        "satellite_site_ids": "satellite1",
        "system": "nanotts"
    },
    "wake": {
        "porcupine": {
            "keyword_path": "porcupine_raspberry-pi.ppn"
        },
        "satellite_site_ids": "satellite1",
        "system": "porcupine"
    }
}

closed 2022-06-10 by Romkabouter