Romkabouter/ESP32-Rhasspy-Satellite #104: Compilation error

fredinka77, 2022-08-04:
Hi,

after a very long time I wanted to setup my Matrix Voice from scratch. I did all the setup steps and can flash it with the standard sample. However, once I try to compile the ESP32-Rhasspy-Satellite within Visual Studio Code and PlatformIO, I always get the errors shown below. 

I already tried that within Windows as well as in Linux, git and python are installed, Any further ideas? 

Thank you!

Fred

Compiling .pio\build\esp32dev\lib0a5\MATRIXVoiceESP32HAL\wishbone_bus.cpp.o
In file included from src/Satellite.cpp:166:
src/General.hpp: In function 'void loadConfiguration(const char*, Config&)':
src/General.hpp:499:25: error: 'class ArduinoJson6194_F1::StaticJsonDocument<512>' has no member named 'getMember'
     config.siteid = doc.getMember("siteid").as<std::string>();
                         ^~~~~~~~~
src/General.hpp:499:59: error: expected primary-expression before '>' token
     config.siteid = doc.getMember("siteid").as<std::string>();
                                                           ^
src/General.hpp:499:61: error: expected primary-expression before ')' token
     config.siteid = doc.getMember("siteid").as<std::string>();
                                                             ^
src/General.hpp:500:28: error: 'class ArduinoJson6194_F1::StaticJsonDocument<512>' has no member named 'getMember'
     config.mqtt_host = doc.getMember("mqtt_host").as<std::string>();
                            ^~~~~~~~~
src/General.hpp:500:65: error: expected primary-expression before '>' token
     config.mqtt_host = doc.getMember("mqtt_host").as<std::string>();
                                                                 ^
src/General.hpp:500:67: error: expected primary-expression before ')' token
     config.mqtt_host = doc.getMember("mqtt_host").as<std::string>();
                                                                   ^
src/General.hpp:501:28: error: 'class ArduinoJson6194_F1::StaticJsonDocument<512>' has no member named 'getMember'
     config.mqtt_port = doc.getMember("mqtt_port").as<int>();
                            ^~~~~~~~~
src/General.hpp:501:54: error: expected primary-expression before 'int'
     config.mqtt_port = doc.getMember("mqtt_port").as<int>();
                                                      ^~~
Compiling .pio\build\esp32dev\lib71b\AsyncTCP@src-7fb2940bccb78b8d2de6915ae328b7fc\AsyncTCP.cpp.o
src/General.hpp:502:28: error: 'class ArduinoJson6194_F1::StaticJsonDocument<512>' has no member named 'getMember'  
     config.mqtt_user = doc.getMember("mqtt_user").as<std::string>();
                            ^~~~~~~Archiving .pio\build\esp32dev\lib0a5\libMATRIXVoiceESP32HAL.a
~~
src/General.hpp:502:65: error: expected primary-expression before '>' token
     config.mqtt_user = doc.getMember("mqtt_user").as<std::string>();
                                                                 ^
src/General.hpp:502:67: error: expected primary-expression before ')' token
     config.mqtt_user = doc.getMember("mqtt_user").as<std::string>();
                                                                   ^
src/General.hpp:503:28: error: 'class ArduinoJson6194_F1::StaticJsonDocument<512>' has no member named 'getMember'
     config.mqtt_pass = doc.getMember("mqtt_pass").as<std::string>();
                            ^~~~~~~~~
src/General.hpp:503:65: error: expected primary-expression before '>' token
     config.mqtt_pass = doc.getMember("mqtt_pass").as<std::string>();
                                                                 ^
src/General.hpp:503:67: error: expected primary-expression before ')' token
     config.mqtt_pass = doc.getMember("mqtt_pass").as<std::string>();
                                                                   ^
src/General.hpp:504:29: error: 'class ArduinoJson6194_F1::StaticJsonDocument<512>' has no member named 'getMember'
     config.mute_input = doc.getMember("mute_input").as<int>();
                             ^~~~~~~~~
src/General.hpp:504:56: error: expected primary-expression before 'int'
     config.mute_input = doc.getMember("mute_input").as<int>();
                                                        ^~~
src/General.hpp:505:30: error: 'class ArduinoJson6194_F1::StaticJsonDocument<512>' has no member named 'getMember'  
     config.mute_output = doc.getMember("mute_output").as<int>();
                              ^~~~~~~~~
Compiling .pio\build\esp32dev\libd9a\AsyncMqttClient\AsyncMqttClient.cpp.o
src/General.hpp:505:58: error: expected primary-expression before 'int'
     config.mute_output = doc.getMember("mute_output").as<int>();
                                                          ^~~
src/General.hpp:506:29: error: 'class ArduinoJson6194_F1::StaticJsonDocument<512>' has no member named 'getMember'
     config.amp_output = doc.getMember("amp_output").as<int>();
                             ^~~~~~~~~
src/General.hpp:506:56: error: expected primary-expression before 'int'
     config.amp_output = doc.getMember("amp_output").as<int>();
                                                        ^~~
src/General.hpp:507:29: error: 'class ArduinoJson6194_F1::StaticJsonDocument<512>' has no member named 'getMember'  
     config.brightness = doc.getMember("brightness").as<int>();
                             ^~~~~~~~~
src/General.hpp:507:56: error: expected primary-expression before 'int'
     config.brightness = doc.getMember("brightness").as<int>();
                                                        ^~~
src/General.hpp:508:37: error: 'class ArduinoJson6194_F1::StaticJsonDocument<512>' has no member named 'getMember'
     config.hotword_brightness = doc.getMember("hotword_brightness").as<int>();
                                     ^~~~~~~~~
src/General.hpp:508:72: error: expected primary-expression before 'int'
     config.hotword_brightness = doc.getMember("hotword_brightness").as<int>();
                                                                        ^~~
src/General.hpp:509:36: error: 'class ArduinoJson6194_F1::StaticJsonDocument<512>' has no member named 'getMember'
     config.hotword_detection = doc.getMember("hotword_detection").as<int>();
                                    ^~~~~~~~~
src/General.hpp:509:70: error: expected primary-expression before 'int'
     config.hotword_detection = doc.getMember("hotword_detection").as<int>();
                                                                      ^~~
src/General.hpp:510:25: error: 'class ArduinoJson6194_F1::StaticJsonDocument<512>' has no member named 'getMember'  
     config.volume = doc.getMember("volume").as<int>();
                         ^~~~~~~~~
src/General.hpp:510:48: error: expected primary-expression before 'int'
     config.volume = doc.getMember("volume").as<int>();
                                                ^~~
src/General.hpp:511:23: error: 'class ArduinoJson6194_F1::StaticJsonDocument<512>' has no member named 'getMember'
     config.gain = doc.getMember("gain").as<int>();
                       ^~~~~~~~~
src/General.hpp:511:44: error: expected primary-expression before 'int'
     config.gain = doc.getMember("gain").as<int>();
                                            ^~~
src/General.hpp:512:28: error: 'class ArduinoJson6194_F1::StaticJsonDocument<512>' has no member named 'getMember'
     config.animation = doc.getMember("animation").as<int>();
                            ^~~~~~~~~
src/General.hpp:512:54: error: expected primary-expression before 'int'
     config.animation = doc.getMember("animation").as<int>();
                                                      ^~~
*** [.pio\build\esp32dev\src\Satellite.cpp.o] Error 1
=========================================== [FAILED] Took 26.02 seconds ===========================================

Environment    Status    Duration
-------------  --------  ------------
esp32dev       FAILED    00:00:26.024

Romkabouter, 2022-08-04:
Hi there @fredinka77 , it looks to me you need to upgrade your arduinojson version. What version are you using?

fredinka77, 2022-08-04:
It is the latest version which was downloaded: v6.19.4

fredinka77, 2022-08-04:
Ok, I switched manually to v6.19.1 and that seems to work: 

https://github.com/bblanchon/ArduinoJson.git#v6.19.1

Thank you!

closed 2022-08-04 by fredinka77