Romkabouter/ESP32-Rhasspy-Satellite #105: Issue with build

tstill88, 2022-08-07:
Its my first time using VSCode / Platform io to work with ESP so maybe im missing some basic knowledge.

Im getting the following error when i try to build to the code:

`Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (5.1.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20004.0 (2.0.4)
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 6.19.3
|-- ArduinoOTA @ 2.0.0
|   |-- Update @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- ESPmDNS @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|-- Ethernet @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- WiFi @ 2.0.0
|-- Wire @ 2.0.0
|-- indicatorlight
|-- NeoPixelBus
|   |-- I2S @ 1.0
|   |-- SPI @ 2.0.0
|-- SPIFFS @ 2.0.0
|   |-- FS @ 2.0.0
Building in release mode
Compiling .pio\build\esp32dev\src\Satellite.cpp.o
Archiving .pio\build\esp32dev\lib6b0\libUpdate.a
Compiling .pio\build\esp32dev\libf82\WiFi\WiFiAP.cpp.o
Compiling .pio\build\esp32dev\libf82\WiFi\WiFiClient.cpp.o
Compiling .pio\build\esp32dev\libf82\WiFi\WiFiGeneric.cpp.o
In file included from src/Satellite.cpp:147:
src/devices/M5AtomEcho.hpp:5:10: fatal error: M5Atom.h: No such file or directory

****************************************************************
* Looking for M5Atom.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:M5Atom.h"
* Web  > https://registry.platformio.org/search?q=header:M5Atom.h
*
****************************************************************

 #include "M5Atom.h"
          ^~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32dev\src\Satellite.cpp.o] Error 1
========================================================================================= [FAILED] Took 13.11 seconds =========================================================================================

Environment    Status    Duration
-------------  --------  ------------
esp32dev       FAILED    00:00:13.113
==================================================================================== 1 failed, 0 succeeded in 00:00:13.113 ==================================================================================== 

 *  The terminal process "C:\Users\tstil\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. `

tstill88, 2022-08-07:
Think i may have moved past that error but now i have this:

`Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (5.1.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20004.0 (2.0.4)
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 45 compatible libraries
Scanning dependencies...
Dependency Graph
|-- MATRIXVoiceESP32HAL @ 0.77.0+sha.2189dcb
|-- AsyncMqttClient @ 0.9.0+sha.89bf464
|   |-- AsyncTCP @ 1.1.1+sha.ca8ac5f
|-- AsyncTCP @ 1.1.1+sha.ca8ac5f
|-- PubSubClient @ 2.8.0+sha.2d228f2
|-- ArduinoJson @ 6.19.4+sha.1d21027
|-- ESP Async WebServer @ 1.2.3
|   |-- AsyncTCP @ 1.1.1+sha.ca8ac5f
|   |-- FS @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- M5Atom @ 0.1.0
|   |-- FastLED @ 3.5.0
|   |   |-- SPI @ 2.0.0
|   |-- Wire @ 2.0.0
|-- FastLED @ 3.5.0
|   |-- SPI @ 2.0.0
|-- AC101 @ 0.0.1
|   |-- Wire @ 2.0.0
|-- wm8978-esp32 @ 1.0.4
|   |-- Wire @ 2.0.0
|-- NeoPixelBus @ 2.7.0
|   |-- SPI @ 2.0.0
|   |-- I2S @ 1.0
|-- ArduinoOTA @ 2.0.0
|   |-- Update @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- ESPmDNS @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|-- Ethernet @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- WiFi @ 2.0.0
|-- Wire @ 2.0.0
|-- indicatorlight
|-- SPIFFS @ 2.0.0
|   |-- FS @ 2.0.0
Building in release mode
Compiling .pio\build\esp32dev\libc0e\FastLED\FastLED.cpp.o
Compiling .pio\build\esp32dev\libc0e\FastLED\bitswap.cpp.o
Compiling .pio\build\esp32dev\libc0e\FastLED\colorpalettes.cpp.o
Compiling .pio\build\esp32dev\libc0e\FastLED\colorutils.cpp.o
Compiling .pio\build\esp32dev\libc0e\FastLED\hsv2rgb.cpp.o
Compiling .pio\build\esp32dev\libc0e\FastLED\lib8tion.cpp.o
Compiling .pio\build\esp32dev\libc0e\FastLED\noise.cpp.o
Compiling .pio\build\esp32dev\libc0e\FastLED\platforms.cpp.o
Compiling .pio\build\esp32dev\libc0e\FastLED\platforms\esp\32\clockless_rmt_esp32.cpp.o
Compiling .pio\build\esp32dev\libc0e\FastLED\power_mgt.cpp.o
Compiling .pio\build\esp32dev\libc0e\FastLED\wiring.cpp.o
Compiling .pio\build\esp32dev\src\Satellite.cpp.o
Compiling .pio\build\esp32dev\src\devices\ES8388Control.cpp.o
Compiling .pio\build\esp32dev\lib98a\MATRIXVoiceESP32HAL\everloop.cpp.o
Compiling .pio\build\esp32dev\lib98a\MATRIXVoiceESP32HAL\matrix_driver.cpp.o
Compiling .pio\build\esp32dev\lib98a\MATRIXVoiceESP32HAL\microphone_array.cpp.o
Compiling .pio\build\esp32dev\lib98a\MATRIXVoiceESP32HAL\microphone_core.cpp.o
[... 96 lines cut ...]
     config.volume = doc.getMember("volume").as<int>();
                                                ^~~
src/General.hpp:511:23: error: 'class ArduinoJson6194_F1::StaticJsonDocument<512>' has no member named 'getMember'
     config.gain = doc.getMember("gain").as<int>();
                       ^~~~~~~~~
src/General.hpp:511:44: error: expected primary-expression before 'int'
     config.gain = doc.getMember("gain").as<int>();
                                            ^~~
src/General.hpp:512:28: error: 'class ArduinoJson6194_F1::StaticJsonDocument<512>' has no member named 'getMember'
     config.animation = doc.getMember("animation").as<int>();
                            ^~~~~~~~~
src/General.hpp:512:54: error: expected primary-expression before 'int'
     config.animation = doc.getMember("animation").as<int>();
                                                      ^~~
Compiling .pio\build\esp32dev\lib87a\AsyncTCP@src-7fb2940bccb78b8d2de6915ae328b7fc\AsyncTCP.cpp.o
*** [.pio\build\esp32dev\src\Satellite.cpp.o] Error 1
========================================================================================= [FAILED] Took 40.12 seconds =========================================================================================

Environment    Status    Duration
-------------  --------  ------------
esp32dev       FAILED    00:00:40.121
==================================================================================== 1 failed, 0 succeeded in 00:00:40.121 ==================================================================================== 

 *  The terminal process "C:\Users\tstil\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.`

Romkabouter, 2022-08-07:
Check this issue:

https://github.com/Romkabouter/ESP32-Rhasspy-Satellite/issues/104

tstill88, 2022-08-07:
I did see that, but i noticed it was back to 6.19.4 again. Ive changed back to 6.19.3 now and unfortunately get another error.

Hopefully getting close to having this working now. Appreciate the help!

`PS C:\Users\tstil\OneDrive\Documents\ESP32-Rhasspy-Satellite-7.8.3\PlatformIO> pio run --target upload
Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (5.1.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20004.0 (2.0.4) 
 - tool-esptoolpy @ 1.30300.0 (3.3.0) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 6.19.3
|-- ArduinoOTA @ 2.0.0
|   |-- Update @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- ESPmDNS @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|-- Ethernet @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- WiFi @ 2.0.0
|-- Wire @ 2.0.0
|-- indicatorlight
|-- SPIFFS @ 2.0.0
|   |-- FS @ 2.0.0
Building in release mode
Compiling .pio\build\esp32dev\src\Satellite.cpp.o
Archiving .pio\build\esp32dev\lib6b0\libUpdate.a
Compiling .pio\build\esp32dev\libf82\WiFi\WiFiAP.cpp.o
Compiling .pio\build\esp32dev\libf82\WiFi\WiFiClient.cpp.o
Compiling .pio\build\esp32dev\libf82\WiFi\WiFiGeneric.cpp.o
In file included from src/Satellite.cpp:146:
src/devices/M5AtomEcho.hpp:5:10: fatal error: M5Atom.h: No such file or directory

****************************************************************
* Looking for M5Atom.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:M5Atom.h"
* Web  > https://registry.platformio.org/search?q=header:M5Atom.h
*
****************************************************************

 #include "M5Atom.h"
          ^~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32dev\src\Satellite.cpp.o] Error 1
========================================================================================= [FAILED] Took 22.22 seconds =========================================================================================

Environment    Status    Duration
-------------  --------  ------------
esp32dev       FAILED    00:00:22.224
==================================================================================== 1 failed, 0 succeeded in 00:00:22.224 ==================================================================================== 
PS C:\Users\tstil\OneDrive\Documents\ESP32-Rhasspy-Satellite-7.8.3\PlatformIO> `

Romkabouter, 2022-08-07:
This is the same you had before right? It apparently cannot find the M5Atom lib. It should be downloaded, but since you are on windows I not sure if that works.
You can always install it manually

tstill88, 2022-08-08:
Yeah thats right. Where does that library need to be installed? Does it need to be in the PlatformIO source folder with the rest of the files?

I can see in the platformio.ini file under lib_deps i have this:

lib_deps = 
	https://github.com/matrix-io/matrixio_hal_esp32.git
	https://github.com/marvinroger/async-mqtt-client.git
	https://github.com/me-no-dev/AsyncTCP.git
	https://github.com/knolleary/pubsubclient.git
	https://github.com/bblanchon/ArduinoJson.git
	ESP Async WebServer
	m5stack/M5Atom
	fastled/FastLED
	yveaux/AC101 @ ^0.0.1
	celliesprojects/wm8978-esp32
	makuna/NeoPixelBus

Sorry for what are just user related problems here!

Romkabouter, 2022-08-08:
This is the location on my Mac:

[image]

tstill88, 2022-08-08:
Yeah that must be it. When i go into PIO and Libraries to add ArduinoJSON 19.6.3 it asks me to pick which project to install it in, and at that point it wipes out all of the other libraries in the esp32dev folder leaving only ArduinoJSON.

If you have any thoughts im all ears otherwise you can probably close this and il try and figure it out. Im all good with ArduinoIDE but never used VSCode / PIO before :) Thanks for your help so far!

tstill88, 2022-08-08:
Ok, i thought i had it fixed. Ive manually installed all of the libraries this time

`Compiling .pio\build\esp32dev\lib3e6\AsyncMqttClient\AsyncMqttClient\Packets\Out\Disconn.cpp.o
Compiling .pio\build\esp32dev\lib3e6\AsyncMqttClient\AsyncMqttClient\Packets\Out\OutPacket.cpp.o
Compiling .pio\build\esp32dev\lib3e6\AsyncMqttClient\AsyncMqttClient\Packets\Out\PingReq.cpp.o
Compiling .pio\build\esp32dev\lib3e6\AsyncMqttClient\AsyncMqttClient\Packets\Out\PubAck.cpp.o
Compiling .pio\build\esp32dev\lib3e6\AsyncMqttClient\AsyncMqttClient\Packets\Out\Publish.cpp.o
Compiling .pio\build\esp32dev\lib3e6\AsyncMqttClient\AsyncMqttClient\Packets\Out\Subscribe.cpp.o
Compiling .pio\build\esp32dev\lib3e6\AsyncMqttClient\AsyncMqttClient\Packets\Out\Unsubscribe.cpp.o
Compiling .pio\build\esp32dev\lib3e6\AsyncMqttClient\AsyncMqttClient\Packets\PingRespPacket.cpp.o
Compiling .pio\build\esp32dev\lib3e6\AsyncMqttClient\AsyncMqttClient\Packets\PubAckPacket.cpp.o
Compiling .pio\build\esp32dev\lib3e6\AsyncMqttClient\AsyncMqttClient\Packets\PubCompPacket.cpp.o
Compiling .pio\build\esp32dev\lib3e6\AsyncMqttClient\AsyncMqttClient\Packets\PubRecPacket.cpp.o
Compiling .pio\build\esp32dev\lib3e6\AsyncMqttClient\AsyncMqttClient\Packets\PubRelPacket.cpp.o
Compiling .pio\build\esp32dev\lib3e6\AsyncMqttClient\AsyncMqttClient\Packets\PublishPacket.cpp.o
Compiling .pio\build\esp32dev\lib3e6\AsyncMqttClient\AsyncMqttClient\Packets\SubAckPacket.cpp.o
Compiling .pio\build\esp32dev\lib3e6\AsyncMqttClient\AsyncMqttClient\Packets\UnsubAckPacket.cpp.o
Compiling .pio\build\esp32dev\lib002\PubSubClient\PubSubClient.cpp.o
Compiling .pio\build\esp32dev\libc5e\AC101\AC101.cpp.o
Compiling .pio\build\esp32dev\lib84c\wm8978-esp32\WM8978.cpp.o
Archiving .pio\build\esp32dev\lib3e6\libAsyncMqttClient.a
Compiling .pio\build\esp32dev\lib720\FS\FS.cpp.o
Archiving .pio\build\esp32dev\lib002\libPubSubClient.a
Archiving .pio\build\esp32dev\libc5e\libAC101.a
In file included from C:/Users/tstil/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:76,
                 from C:/Users/tstil/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,
                 from .pio/libdeps/esp32dev/wm8978-esp32/src/WM8978.cpp:2:
Compiling .pio\build\esp32dev\lib720\FS\vfs_api.cpp.o
.pio/libdeps/esp32dev/wm8978-esp32/src/WM8978.cpp: In member function 'bool WM8978::begin()':
.pio/libdeps/esp32dev/wm8978-esp32/src/WM8978.cpp:318:14: error: 'TAG' was not declared in this scope
     ESP_LOGE(TAG, "No WM8978 dac @ i2c address: 0x%X", WM8978_ADDR);
              ^~~
C:/Users/tstil/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-log.h:159:72: note: in definition of macro 'log_e'
 #define log_e(format, ...) log_printf(ARDUHAL_LOG_FORMAT(E, format), ##__VA_ARGS__)
                                                                        ^~~~~~~~~~~
.pio/libdeps/esp32dev/wm8978-esp32/src/WM8978.cpp:318:5: note: in expansion of macro 'ESP_LOGE'
     ESP_LOGE(TAG, "No WM8978 dac @ i2c address: 0x%X", WM8978_ADDR);
Compiling .pio\build\esp32dev\libf82\WiFi\WiFi.cpp.o
     ^~~~~~~~
.pio/libdeps/esp32dev/wm8978-esp32/src/WM8978.cpp:323:14: error: 'TAG' was not declared in this scope
     ESP_LOGE(TAG, "WM8978 init err: 0x%X", err);
              ^~~
C:/Users/tstil/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-log.h:159:72: note: in definition of macro 'log_e'
 #define log_e(format, ...) log_printf(ARDUHAL_LOG_FORMAT(E, format), ##__VA_ARGS__)
                                                                        ^~~~~~~~~~~
.pio/libdeps/esp32dev/wm8978-esp32/src/WM8978.cpp:323:5: note: in expansion of macro 'ESP_LOGE'
     ESP_LOGE(TAG, "WM8978 init err: 0x%X", err);
     ^~~~~~~~
.pio/libdeps/esp32dev/wm8978-esp32/src/WM8978.cpp: In member function 'bool WM8978::begin(uint8_t, uint8_t, uint32_t)':
.pio/libdeps/esp32dev/wm8978-esp32/src/WM8978.cpp:346:38: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second:
   if (!Wire.begin(sda, scl, frequency)) {
                                      ^
In file included from .pio/libdeps/esp32dev/wm8978-esp32/src/WM8978.cpp:3:
C:/Users/tstil/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:79:10: note: candidate 1: 'bool TwoWire::begin(int, int, uint32_t)'
     bool begin(int sda=-1, int scl=-1, uint32_t frequency=0); // returns true, if successful init of i2c bus
          ^~~~~
C:/Users/tstil/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.h:80:10: note: candidate 2: 'bool TwoWire::begin(uint8_t, int, int, uint32_t)'
     bool begin(uint8_t slaveAddr, int sda=-1, int scl=-1, uint32_t frequency=0);
          ^~~~~
In file included from C:/Users/tstil/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:76,
                 from C:/Users/tstil/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,
                 from .pio/libdeps/esp32dev/wm8978-esp32/src/WM8978.cpp:2:
.pio/libdeps/esp32dev/wm8978-esp32/src/WM8978.cpp:347:14: error: 'TAG' was not declared in this scope
     ESP_LOGE(TAG, "Wire setup error sda=%i scl=%i frequency=%i", sda, scl, frequency);
              ^~~
C:/Users/tstil/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-log.h:159:72: note: in definition of macro 'log_e'
[... cut ...]
     ESP_LOGE(TAG, "Wire setup error sda=%i scl=%i frequency=%i", sda, scl, frequency);
     ^~~~~~~~
*** [.pio\build\esp32dev\lib84c\wm8978-esp32\WM8978.cpp.o] Error 1
================================================================================================= [FAILED] Took 148.54 seconds =================================================================================================

Environment    Status    Duration
-------------  --------  ------------
esp32dev       FAILED    00:02:28.541`

Romkabouter, 2022-08-08:
I do not know which device you want to build, but you can remove these lib_deps.

   https://github.com/matrix-io/matrixio_hal_esp32.git
   m5stack/M5Atom
   fastled/FastLED
   yveaux/AC101 @ ^0.0.1
   celliesprojects/wm8978-esp32
   makuna/NeoPixelBus

All of those are device specific, so just keeps the ones your device needs.
I do not have windows, so I cannot reproduce this

tstill88, 2022-08-08:
Im building for an M5 Atom Echo.

Im wondering if im getting confused on the library locations? I was copying your photo above within the esp32dev folder, but since im using the M5Atom should it be under that? I see there is another ArduinoJson folder in there, and i dont know if both are being updated to 6.19.3 when i add it to the project folder.

At the moment im only trying to build the code, ive noticed after i deleted the AC101 folder from the esp32dev folder, it reappears when i hit build.

Thanks for your patience....

tstill88, 2022-08-08:
[image]

Should i be removing all the folders apart from the M5Atmoecho folder?

Romkabouter, 2022-08-08:
> Should i be removing all the folders apart from the M5Atmoecho folder?

First remove these from the libs_deps in platformio.ini

https://github.com/matrix-io/matrixio_hal_esp32.git
fastled/FastLED
yveaux/AC101 @ ^0.0.1
celliesprojects/wm8978-esp32
makuna/NeoPixelBus

And then remove the folders, otherwise they will be downloaded again.

tstill88, 2022-08-08:
Success! It was due to the extra libraries. Ive kept FastLED as it wants that but removing the other libraries from the esp32dev and m5athom folder has let me compile the code.

A small victory as now i have to learn Rhasspy, but thanks for your help!

closed 2022-08-15 by Romkabouter