Romkabouter/ESP32-Rhasspy-Satellite #107: [INMP44 / FN-M16P combo] New to ESP32, require some starting help

Alexwijn, 2022-08-19:
Hello, I just purchased some ESP32 NodeMCUs to eventually replace my Google Home Mini. I also purchased a DollaTek INMP44 to test with and a speaker that appears to be based on FN-M16P. 

Now I found that there is support for an INMP44 but I'm wondering how I can configure the speaker. I do not have the hardware yet, it should arrive this Sunday. But I like to do some preparation.

The speaker has JST-PH2.0mm 2-pins which I assume I need to change to some normal jumper cables in order to connect to the ESP32?

I already pulled the project and changed the General, WiFi and MQTT settings.

Alexwijn, 2022-08-20:
After some more research, I think I forgot to buy ampliers to make the speaker work. So I came to conclusion to buy MAX98357 amplifiers, and this project also support that combo (yah!).

The ESP32 will arrive tomorrow, so I can already get started with the microphone. The amplifier will be delivered this Tuesday. I'll update the issue by then, if I got it solved.

Alexwijn, 2022-08-23:
Okay, I got everything now. And everything is working so far.

But when using Rhasspy Satellite the sound is very distorted. Is there any way I can tweak it?

I tested before with a radio station and the sound was perfect.

Romkabouter, 2022-08-23:
I do not own such a board, so I have no way of telling how to fix it.

Maybe there is an issue with the hardware pins? Since you mention the recording is also distorted

Alexwijn, 2022-08-23:
Should not be. When using this audio driver the speaker works just fine when playing a radio station:
https://github.com/schreibfaul1/ESP32-audioI2S

So it has to be a configuration issue. That audio driver also supports volume control which the current driver does not, what maybe also helps.

Alexwijn, 2022-08-24:
Maybe related to: https://github.com/Romkabouter/ESP32-Rhasspy-Satellite/issues/53 ? Only different device.

Alexwijn, 2022-08-26:
Not sure how but I fixed the microphone. I got everything on a new breadboard now. But the speaker remains the same, and is perfect when playing radio station.

Romkabouter, 2022-08-28:
Well, I do not have such a device so I cannot help to reproduce the issue sadly. 
What are your settings in rhasspy for TTS?

Alexwijn, 2023-01-08:
I think I want to try this project again. At the time I bought every component I needed for this project in the hope it works, but now it just collects dust. So let's hope it works with the latest code base.

But to answer your question: I tried all sort of TTS settings: difference voices, different sample sizes, etc.
But everything resulted in exactly the same problem. If you have suggestions on what setting to try, I'm all ears.

Romkabouter, 2023-01-09:
> But everything resulted in exactly the same problem.

That is a pity, but like I said before I do not own such a board and the device is added by another contributer. So at the moment I will not be able to investigate or solve this particular issue. I will leave this defect open for anyone else having this issue.

carkom, 2023-05-24:
@Alexwijn can you share the code you have for the microphone? I am using external speakers so broken speakers is not an issue for me. Would love to duplicate your project with just mic and board. Cheers!

Alexwijn, 2023-05-24:
I don't have the code anymore, I also didn't change much.

But I moved on to ESPHome to get everything working with Home Assistant. They recently added support for it and the speaker sounds fine now.

carkom, 2023-05-24:
Can you share the .yaml you use to connect to Rhasspy as a satellite?

Alexwijn, 2023-05-24:
I do not use Rhasspy anymore. I have the ESP32 connected to Home Assistant using ESPHome as mentioned before.

I imagine Home Assistant get more and more features from Rhasspy since the companies are working together now.

closed 2023-11-06 by Romkabouter