Romkabouter/ESP32-Rhasspy-Satellite #109: Wake word detection not working

ProrokWielki, 2022-08-28:
Hi
I'm trying to set up ESP32 as a satellite for rhasspy. 
My setup:
* ESP32 with inmp441 and button for hardware wake up. (SiteId: Test)
* Raspberry Pi 3B hosting mqtt broker and  rhasspy 2.5.11(in docker container). (SiteId: Master)
What works:
* Audio from - hermes/audioServer/Test/audioFrame - loud and clear
* triggering hardware wake up works perfectly - sesion starrted and intent detected
What does not work:
* Detecting wake up word - I tried 2 different models wit porcupine (jarvis and bumblebee) with every sensitivity level. i also tried pocketsphinx with different key phrases. No session is started no intent detected.

rhasspy config :
[image]
```json
{
    "dialogue": {
        "satellite_site_ids": "Test",
        "system": "rhasspy"
    },
    "handle": {
        "satellite_site_ids": "Test"
    },
    "intent": {
        "satellite_site_ids": "Test",
        "system": "fsticuffs"
    },
    "microphone": {
        "system": "hermes"
    },
    "mqtt": {
        "enabled": "true",
        "host": "192.168.40.2",
        "site_id": "Master"
    },
    "sounds": {
        "system": "hermes"
    },
    "speech_to_text": {
        "satellite_site_ids": "Test",
        "system": "kaldi"
    },
    "text_to_speech": {
        "satellite_site_ids": "Test",
        "system": "nanotts"
    },
    "wake": {
        "porcupine": {
            "keyword_path": "bumblebee_raspberry-pi.ppn"
        },
        "satellite_site_ids": "Test",
        "system": "porcupine"
    }
}
```

What am i doing wrong?

Romkabouter, 2022-08-28:
Since the hardware button and intent recognition seems to work, this is an issue with Rhasspy and not the satellite.
What software is running on the pi? Maybe try another wakeword model, I believe there is also bumblebee_linux

ProrokWielki, 2022-08-28:
Thanks for quick replay. The software on the Pi:
* Raspbian buster - kernel  5.10.103-v7+
* docker 18.09.1, build 4c52b90
* rhasspy 2.5.11

i tried differnet models and methods with the same resoult - no luck 

I setup a rhasspy on the x86 machine just to check if helps, but it is exactly the same.
I also tried rhasppy 2.5.10 and behawior is exactly the same.

So the problem is either rhasppy config or the satellite config. If you don't see anything fundamentally wrong with my setup I will create a post on rhasspy community forum.

Romkabouter, 2022-08-29:
I do not see an issue with your config. You also mentioned that hardware button and intent recognition works, so it does not seem to be an error on this software.
You state audio is loud and clear, basically that is all this software does. I cannot tell why the wakeword does not work, I am using snowboy engine and that works fine.

ProrokWielki, 2022-08-29:
Thanks for ensuring me there is nothing wrong with the config, I created a [topic ](https://community.rhasspy.org/t/wake-word-detection-with-esp32-satellite-not-working/3879) on rhasspy community forum. If someone encounters the same problem in future maybe solution will be there.

closed 2022-08-29 by ProrokWielki

Romkabouter, 2022-08-29:
ok cool, just to be sure: the wakeword setting on the satellite is set to Remote yes?

ProrokWielki, 2022-08-30:
It is

[image]
[image]