Romkabouter/ESP32-Rhasspy-Satellite #110: no audio or response on wakeword

NopeFinder, 2022-09-01:
I'm new to MQTT devices, but i feel like it should be working. Yet it doesn't for me.
I've connected it to the mosquitto broker I'm using on homeassistant and using mosquito_sub I'm listening in on hermes/# and see a stream of data coming as expected starting with: 
hermes/audioServer/speaker/audioFrame RIFF,☻WAVEfmt ►☺☺Ç>}☻►data☻ 
and then some random characters resembling the data. however it does not seem to get processed by rhasspy? (nothing in the logs and no response to wake word) then when I try to play something using the tts for instance i see: 
hermes/tts/say {"text": "test", "siteId": "default", "lang": null, "id": "id-of-my-rhasspy-in-mqtt", "sessionId": "", "volume": 1.0}
hermes/audioServer/default/playBytes/id-of-my-rhasspy-in-mqtt RIFF$P  WAVEfmt ►   ☺ ☺ Ç>   }  ☻ ► data 
and then the .wav data i guess. yet nothing is playing anything? might be too little info to go on but any idea what i might be doing wrong? feel like i just set something up wrong in rhasspy, but can't figure out what exactly...

Romkabouter, 2022-09-01:
please post your Rhasspy profile, otherwise there is no way to tell if there is an error in it.
The stream of data is correct, so most likely the satellite is working correctly.

> try to play something using the tts for instance i see:
> hermes/tts/say {"text": "test", "siteId": "default", "lang": null, "id": "id-of-my-rhasspy-in-mqtt", "sessionId": "", "volume": 1.0}
> hermes/audioServer/default/playBytes/id-of-my-rhasspy-in-m

You are posting this to default as it seems, but you should use the siteId of your satellite (I assume speaker in your case)

NopeFinder, 2022-09-01:
Thanks for responding.
My current profile is:
{
    "dialogue": {
        "satellite_site_ids": "speaker",
        "system": "rhasspy"
    },
    "intent": {
        "system": "fsticuffs"
    },
    "microphone": {
        "pyaudio": {
            "siteId": "speaker",
            "udp_audio_host": "192.168.0.41",
            "udp_audio_port": "12202"
        }
    },
    "mqtt": {
        "enabled": "true",
        "host": "192.168.0.36",
        "password": "mypassword",
        "username": "myusername"
    },
    "speech_to_text": {
        "system": "kaldi"
    },
    "text_to_speech": {
        "satellite_site_ids": "speaker",
        "system": "espeak"
    },
    "wake": {
        "porcupine": {
            "udp_audio": "192.168.0.41:12202"
        },
        "satellite_site_ids": "speaker"
    }
}
these are only the things that differ from the default thought, so I'm guessing the variable for SiteId is not one of these. which setting would have to be changed to puiblish to the right SiteId?

NopeFinder, 2022-09-02:
managed to look over the very first option, even above the mqtt tab. read shield instead of siteId. rookie mistake

Romkabouter, 2022-09-02:
Ah ok, so does it work now?

NopeFinder, 2022-09-02:
Yeah it works now

closed 2022-09-02 by Romkabouter