Romkabouter/ESP32-Rhasspy-Satellite #112: Device Support - Raspiaudio ESP Muse (Proto|Luxe)

rufinus, 2022-09-25:
Would this Device work as Rhasspy Satellite?
https://raspiaudio.com/produit/muse-proto

Which device_type should I use?

TIA

Romkabouter, 2022-09-25:
Hi,

It is an ESP32 so yes that should be possible. You should connect your speaker to the external speaker connection
You can create a new device for it, use this file as a reference to set the correct pins:

https://github.com/RASPIAUDIO/museProto_speaker/blob/main/museProto_speaker/museProto_speaker.ino

For testing you can edit an existing device_type, the Inmp441 might work if you change the pin numbers.

rufinus, 2022-09-26:
thank you, will test and report back.

Romkabouter, 2022-09-26:
ok, good luck and thanks for the effort!

closed 2022-09-26 by Romkabouter

Wolverine-14, 2022-10-14:
@rufinus, did you ever get this working? I'm looking into doing the same thing for the ESP Muse Luxe.

rufinus, 2022-10-17:
@Wolverine-14 didnt have the time yet. but want to look into it, cant say when unfortunatly.

Wolverine-14, 2022-11-12:
@rufinus dis you ever get the time to teat this out?

rufinus, 2022-11-12:
@Wolverine-14 nope, sorry. :(