Romkabouter/ESP32-Rhasspy-Satellite #113: AudioKit ES8388 Automatic Level Control

uncarvedblock78, 2022-10-15:
I'm running this on an an audio kit (ES8388 variant) and have found that unless I am very close to the microphones, the speech to text engine has difficulty hearing commands (although the wake words seem to be okay even at a fair distance)

In doing some digging into the ES8388 datasheets, it would appear that the chip can perform ALC, and some of the older documentation does describe some recommended presets for voice, music, and general configurations. [https://dl.radxa.com/rock2/docs/hw/ds/ES8388%20user%20Guide.pdf](https://dl.radxa.com/rock2/docs/hw/ds/ES8388%20user%20Guide.pdf)

Referring to the driver code found here: [https://github.com/vanbwodonk/es8388arduino](https://github.com/vanbwodonk/es8388arduino) which is very similar to the ES8388 driver you've written, I've managed to enable ALC  in "voice" mode and the speech to text seems to be greatly improved.

my edits were the following:
added `setALCmode()` and `alcmodesel_t` to  ES8388Control.cpp and ES8388Control.h,
modify `InitI2SSpeakerOrMic()` in AudioKit.hpp, calling `setALCmode(VOICE)` if mode is set to mic, otherwise `setALCmode(DISABLED)`

Perhaps you can take a look, thanks.

Romkabouter, 2022-12-13:
Ok, maybe you can make a pull request for it and than I can check. I do not have such a device myself.

romakrau, 2023-01-03:
Hi, I have tested your hint and add the code described in your post. It works and now the Rhassyp recognized the spoken sentence very well. Please could you be so kind and describe the correct integration of the code in the function InitI2SSpeakerOrMic() . I'm not shure if I do it right, but it's works.
Greetings Roman

uncarvedblock78, 2023-01-03:
I'll see if I can make a pull request.

uncarvedblock78, 2023-01-19:
Sorry for the delay, I created pull request #122

Romkabouter, 2023-01-20:
https://github.com/Romkabouter/ESP32-Rhasspy-Satellite/releases/tag/v7.9.1

Thanks for the effort!

closed 2023-01-20 by Romkabouter