Romkabouter/ESP32-Rhasspy-Satellite #114: M5 Atom Echo not always working

asetGem, 2022-10-26:
Hi, I am trying to use a Atom M5 echo as a satellite for rhasspy. However, it does not work properly. Here are the behaviors I have (without changing any parameter):

# What it does when it seems to work
1. the atom echo led is blue
2. when I pronounce the hot word, or click on the button, it turns red
3. I say the command 
4. I can hear a little sound, probably signaling the end of recording
5. in rhasspy web UI, I can see some text in the field "text to recognize", and, when it has recognized something, I have the corresponding intents.

# What it does most of the time

## Nothing sent

Steps 1 to 4 are as previously, but then, nothing seems to be received on rhasspy side

## Nothing happens

The device stays in step 1. 
Neither pushing the button or pronouncing the hot word does something. 

When this happens, I have to unplug and replug the device. 
This usually happens after 2 or 3 times working, so, very often!

# What is my configuration

## Rhasspy base
```
{
    "dialogue": {
        "satellite_site_ids": "AtomEcho",
        "system": "rhasspy"
    },
    "handle": {
        "satellite_site_ids": "AtomEcho",
        "system": "hass"
    },
    "home_assistant": {
        "access_token": "XXX",
        "handle_type": "event",
        "url": "http://192.168.XXX"
    },
    "intent": {
        "satellite_site_ids": "AtomEcho",
        "system": "fuzzywuzzy"
    },
    "microphone": {
        "pyaudio": {
            "siteId": "AtomEcho"
        },
        "system": "hermes"
    },
    "mqtt": {
        "enabled": "true",
        "host": "XXX",
        "password": "XXX",
        "site_id": "base",
        "username": "XXX"
    },
    "sounds": {
        "system": "hermes"
    },
    "speech_to_text": {
        "satellite_site_ids": "AtomEcho",
        "system": "kaldi"
    },
    "text_to_speech": {
        "satellite_site_ids": "AtomEcho",
        "system": "nanotts"
    },
    "wake": {
        "raven": {
            "keywords": {
                "custom_wakeword": {
                    "enabled": true
                }
            }
        },
        "satellite_site_ids": "AtomEcho",
        "system": "raven"
    }
}
```

## Flashing Atom Echo

setting.ini : 
```
[General]
hostname="rhasspy base ip"
deployhost="rhasspy base ip"
siteId=AtomEcho
device_type=0
network_type=0

[Wifi]
ssid="XXX"
password="XXX"

[OTA]
method=upload
password=OTApassword
port=3232

[MQTT]
hostname="same as in base config"
port=1883
username = "same as base config"
password=same as base config
```

Then, I plug it to my computer via USB, and flash it using `pio run --target upload`. Everything works well, it is transmitted until the end.

## Satellite config

Once it has been flashed, the Atom Echo gets its own IP. I open it in my browser, and I have the following config:
 
[image]

Do not hesitate to ask me if you need more details.

Romkabouter, 2022-12-13:
Hi @asetGem, thanks for the details.
The M5 turning red at point 2 is not OK, that means it crashes and reboots, did you use the latest code?
The sound at 4 is probably the small crackling you hear when booting. I expect you hear the same sound if you pull the plug and insert it again.
Your point 5 will send the TTS to the base, not the M5 so that is expected behaviour.
Your settings seem fine to me.

asetGem, 2023-05-08:
Hi again,

I did many tests and brought another Atom echo to check if it was not an issue with the first one...but still the same problem.

I still have the led turning red when I press the button, but this seems to be ok, as when it happens, audio is taken into account (although intent not always recognized but that's another problem).

But sometimes, 
- the led does not turn red (remains blue), and in this case nothing happens. 
- once it turns red, recognizes and sents the intent via mqtt, it remains red, and I can't do anything else than unplug/replug to reboot the system.

When I boot the system (=insert plug), the led is first green and then turns blue. Usually, the first try works as expected.

I used the main branch, cloned yesterday.

aperrin, 2023-12-02:
Nop, still nothing :(

jmorris644, 2023-12-02:
So, a little more playing around. I have been watching the serial console in VSstudio. 

Here is the boot info:
```
Booting
M5Atom initializing...OK
Loading configuration
{
  "siteid": "AtomEcho",
  "mqtt_host": "192.168.x.x",
  "mqtt_port": 1883,
  "mqtt_user": "",
  "mqtt_pass": "",
  "mute_input": false,
  "mute_output": false,
  "amp_output": 0,
  "brightness": 25,
  "hotword_brightness": 80,
  "hotword_detection": 1,
  "volume": 100,
  "gain": 5,
  "animation": 0
}
Creating I2Stask
Enter WifiDisconnected
Enter WifiConnected
Connected to Wifi with IP: 192.168.x.x, SSID: xxxx, BSSID: FA..., RSSI: -48
Enter MQTTDisconnected
Connecting MQTT: 192.168.x.x, 1883
Enter MQTTConnected
Connected as AtomEcho
Connected to asynch MQTT!
Enter Idle
E (2482) I2S: i2s_driver_uninstall(1999): I2S port 0 has not installed
```
Which all looks fine.

Then I hit the button and did a command and got this:

```
Send ListeningEvent from toggleOff (dialogueSession)
Enter Listening
Send ListeningEvent from toggleOff (playAudio)
Samplerate: 44100, Channels: 2, Format: 1, Bits per Sample: 16, Start: 44
Send PlayBytesEvent
Send BeginPlayAudioEvent in PlayBytesEvent
BeginPlayAudioEvent in Listening
Enter ListeningPlay
Done
Send StreamAudioEvent
Send EndPlayAudioEvent in StreamAudioEvent
EndPlayAudioEvent in ListeningPlay
Enter Listening
Send IdleEvent from toggleOn (playAudio)
IdleEvent in Listening
Enter Idle
Send IdleEvent from toggleOn (dialogueSession)
```

Which as far as I can tell, looks fine as well.

If I let it error out without saying anything I get this:

```
Enter Listening
Send IdleEvent from toggleOn (playAudio)
IdleEvent in Listening
Enter Idle
Send IdleEvent from toggleOn (dialogueSession)
Send ErrorEvent from errorTopic
Enter Error
Send ListeningEvent from toggleOff (playAudio)
Samplerate: 44100, Channels: 2, Format: 1, Bits per Sample: 16, Start: 44
Send PlayBytesEvent
Send BeginPlayAudioEvent in PlayBytesEvent
BeginPlayAudioEvent in Error
Enter ErrorPlay
Done
Send StreamAudioEvent
Send EndPlayAudioEvent in StreamAudioEvent
EndPlayAudioEvent in ErrorPlay
Enter Error
Send IdleEvent from toggleOn (playAudio)
IdleEvent in Error
Enter Idle
Send IdleEvent from toggleOn (dialogueSession)
```

And I do hear the error sound.

At one point I was getting buffer underrun errors but I did not capture them and cannot reproduce now.
I still am getting the red led and am not seeing a reboot. So I think the red LED is ok.