Romkabouter/ESP32-Rhasspy-Satellite #115: ESP32-LyraT V4.3 with ES8388 support

Omirax, 2022-12-12:
Hello!

Unfortunately, only headphones work for ESP32-Lyrath V4.3 with ES8388.
The microphone and speaker do not work, although all pinouts are the same: 
https://docs.espressif.com/projects/esp-adf/en/latest/design-guide/dev-boards/board-esp32-lyrat-v4.3.html#allocation-of-esp32-pins

Could you take a look at what the problem might be?

Regards,

Romkabouter, 2022-12-13:
Hi @Omirax, I do not have such a device so I cannot test it myself. The code seems to work because the logging shows no error.

Omirax, 2022-12-13:
Hello!
Unfortunately, as I wrote earlier, only the headphone output works. It is possible that this is due to the fact that both ESP32-Lyrath microphones are connected to the ES8388 pin 23/24. While contacts 21/22/23/24 are used on AI

Send ListeningEvent from toggleOff (playAudio)
Samplerate: 16000, Channels: 1, Format: 1, Bits per Sample: 16, Start: 44
Send PlayBytesEvent
Send BeginPlayAudioEvent in PlayBytesEvent
BeginPlayAudioEvent in Error
**Enter ErrorPlay**

Send IdleEvent from toggleOn (dialogueSession)
Send ErrorEvent from errorTopic
**Enter Error**

This is ok?

Romkabouter, 2022-12-14:
Yes, that is ok. That is the state of the state machine.
Your intent was not recognised (no mic is working) and then Rhasspy sends an error (+sound)
That is the logging you see.

It is possible that there is a problem with the pinouts difference, your link at the top is not for the board with ES8388, is that correct?

Omirax, 2022-12-14:
ESP32-Lyrat  V4.3 - board with ES8388
 https://dl.espressif.com/dl/schematics/ESP32-LYRAT_V4.3-20220119.pdf

Omirax, 2022-12-22:
I found all the little problems and shortcomings with the volume and playback quality. 
But I still couldn't figure out why the microphones aren't activated. 

@Romkabouter Maybe you have some ideas?

Omirax, 2022-12-24:
@Romkabouter 
Work with es8388_write_reg(ES8388_ADDR, ES8388_ADCCONTROL3, 0x02);

Romkabouter, 2022-12-26:
Well, PR's are welcome but I do not have such a device so I cannot test it and therefore will not change code myself

Omirax, 2022-12-26:
`#define ES8388_ADCCONTROL3 0x0b`
...
ES8388 User Guide: `es8388_write( 0x0B 0x02); // Reg 0x0B = 0x02 (Select LIN1and RIN1 as differential input pairs)`
ESP32-LyraT Use only LIN1and RIN1 for mic input. LIN2and RIN2 - for AUX 
https://dl.espressif.com/dl/schematics/ESP32-LYRAT_V4.3-20220119.pdf

Romkabouter, 2023-11-06:
@Omirax Are you still trying to find a solution?