Romkabouter/ESP32-Rhasspy-Satellite #117: Can't follow instructions (M5Stack Atom Echo)

wryandginger, 2022-12-22:
Could someone please write a step-by-step tutorial for getting this working on the M5Stack Atom Echo? --- Clearly, I need a tutorial written as if I was a complete moron.  

 I can't follow the "getting started" directions -- There are far too many assumptions. I tried to cobble together something by semi following the tutorial for the matrix voice, but this is incredibly frustrating and I'm totally lost. 

I cloned this project in Visual Studio Code. Then I modified the example settings file the best I could. When I ran "pio run --target upload" the compile fails:

Compiling .pio\build\esp32dev\lib5f0\FastLED\platforms.cpp.o
xtensa-esp32-elf-g++: fatal error: no input files
compilation terminated.
xtensa-esp32-elf-g++: fatal error: no input files
xtensa-esp32-elf-g++: fatal error: no input files
compilation terminated.
compilation terminated.

I've purged the platformio cache and tried this several times. Obviously, I'm doing something wrong. Could someone PLEASE just WALK me through this?

sajov, 2022-12-22:
That would be helpful. If someone can write a short tutorial , thank you very much!

wryandginger, 2022-12-22:
For the life of me I cannot get this to compile. My attempt above was done on Windows 11 with VisualCode.
I just tried to compile on my Mac with no luck. I hope I'm not coming across as rude; I'm just not sure what I'm doing wrong.

Here's what I did on my Mac:

1. I installed homebrew, then installed platformio.
2. I cloned this repo, went to the platformio folder and made a settings.ini file. I did my best to follow the instructions.
3. I then tried: pio run --target upload

The compile failed:
Compiling .pio/build/esp32dev/lib035/FastLED/bitswap.cpp.o
sh: -c: line 0: unexpected EOF while looking for matching `''
sh: -c: line 1: syntax error: unexpected end of file
Compiling .pio/build/esp32dev/lib035/FastLED/colorpalettes.cpp.o
sh: -c: line 0: unexpected EOF while looking for matching `''
sh: -c: line 1: syntax error: unexpected end of file
Compiling .pio/build/esp32dev/lib035/FastLED/colorutils.cpp.o
sh: -c: line 0: unexpected EOF while looking for matching `''
sh: -c: line 1: syntax error: unexpected end of file
Compiling .pio/build/esp32dev/lib035/FastLED/hsv2rgb.cpp.o
sh: -c: line 0: unexpected EOF while looking for matching `''
sh: -c: line 1: syntax error: unexpected end of file
Compiling .pio/build/esp32dev/lib035/FastLED/lib8tion.cpp.o
sh: -c: line 0: unexpected EOF while looking for matching `''
sh: -c: line 1: syntax error: unexpected end of file
Compiling .pio/build/esp32dev/lib035/FastLED/noise.cpp.o
sh: -c: line 0: unexpected EOF while looking for matching `''
sh: -c: line 1: syntax error: unexpected end of file
Compiling .pio/build/esp32dev/lib035/FastLED/platforms.cpp.o
sh: -c: line 0: unexpected EOF while looking for matching `''
sh: -c: line 1: syntax error: unexpected end of file
*** [.pio/build/esp32dev/lib035/FastLED/bitswap.cpp.o] Error 2
*** [.pio/build/esp32dev/lib035/FastLED/FastLED.cpp.o] Error 2
*** [.pio/build/esp32dev/lib035/FastLED/colorutils.cpp.o] Error 2
*** [.pio/build/esp32dev/lib035/FastLED/noise.cpp.o] Error 2
*** [.pio/build/esp32dev/lib035/FastLED/hsv2rgb.cpp.o] Error 2
*** [.pio/build/esp32dev/lib035/FastLED/lib8tion.cpp.o] Error 2
*** [.pio/build/esp32dev/lib035/FastLED/colorpalettes.cpp.o] Error 2
sh: -c: line 0: unexpected EOF while looking for matching `''
sh: -c: line 1: syntax error: unexpected end of file
*** [.pio/build/esp32dev/lib035/FastLED/platforms.cpp.o] Error 2
========================= [FAILED] Took 51.34 seconds =========================

Environment    Status    Duration
-------------  --------  ------------
esp32dev       FAILED    00:00:51.341
==================== 1 failed, 0 succeeded in 00:00:51.341 ====================

sajov, 2022-12-22:
Great, I will try the same today and report then

sajov, 2022-12-22:
@wryandginger my bad, sorry i do not own a M5Stack, my device is a ESP32-S3-BOX, wich is not supported yet

wryandginger, 2022-12-22:
Turns out it's my wifi password causing the compile error.

FWIW, this won't compile if you have the ' character anywhere in your SSID or wifi password.

closed 2022-12-22 by wryandginger