Romkabouter/ESP32-Rhasspy-Satellite #118: ESP AudioKit I2S port 0 has not installed

romakrau, 2022-12-23:
Hi all,

I will use this software as a satellite for Rhasspy on a FHEM smart home server.
I had compiled and installed the software with PlatformIO and got this output:

`rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:12776
load:0x40080400,len:3032
entry 0x400805e4
Booting
Identifying Codec... Checking for ES8388 Pinout Variant 1...
Checking for ES8388 Pinout Variant 2...
Found audio controller ES8388 Pinout Variant 2 at SDA: 33 / SCL: 32
Loading configuration
Failed to read file, using default configuration
Creating I2Stask
Enter WifiDisconnected
Enter WifiConnected
Connected to Wifi with IP: 192.168.xxx.xxx, SSID: xxxx BSSID: xxxxxxxxxxxxxxxxxxxx, RSSI: -60
Enter MQTTDisconnected
Connecting MQTT: 192.168.xxx.xxx, 1883
Enter MQTTConnected
Connected as ESPAudiokitFlur
Connected to asynch MQTT!
Enter Idle
InitI2SSpeakerOrMic -> Mic
E (2409) I2S: i2s_driver_uninstall(1999): I2S port 0 has not installed
`
What does it mean " I2S port 0 has not installed"?
What about the settings?

hostname=192.168.xxx.xxx                  ; The Adress by what?
deployhost=192.168.187.76               ; Only used by matrix?

Thank you for helping me

Greetings Roman

Omirax, 2022-12-25:
Hello!
`(2409) I2S: i2s_driver_uninstall(1999): I2S port 0 has not installed`
It's not a problem. This is an informational message. After that, the device waits for the command via the microphone.

romakrau, 2022-12-26:
Hallo! Trank you for the answer. How can I check if the speaker output works correct.Greetings Roman

Romkabouter, 2022-12-26:
> hostname=192.168.xxx.xxx ; The Adress by what?

this will  be the ip of the device if you set te network to fixed

> deployhost=192.168.187.76 ; Only used by matrix?

yes, that is the ip of the pi on which the matrix is attached (only required on first flash)

romakrau, 2023-01-03:
I improved the code with the hint mentioned in post #114. It works now. Thank you for the reply.

closed 2023-01-03 by romakrau