Romkabouter/ESP32-Rhasspy-Satellite #119: Help for flash M5 Atom Echo

Gouldile, 2022-12-31:
Hello, I bought an M5 Atom Echo and I would like to install rhasspy on it.  I usually use the arduino IDE to flash the esp32 but there I don't understand how to do it with the procedure described in github... I searched on the internet but I couldn't find a tutorial explaining  how to flash this code.  Could someone please explain to me in more detail how to do this?  :) Thank you

Romkabouter, 2022-12-31:
Is it best to use VS Code with PlatformIO and then follow https://github.com/Romkabouter/ESP32-Rhasspy-Satellite/blob/master/m5atomecho.md

What part did you not understand?

closed 2023-01-24 by Romkabouter

Gouldile, 2023-01-25:
hello, 

sorry I forgot to answer, thank you for your help you have to use PlatformIO which I had never used.

Have a good day