Romkabouter/ESP32-Rhasspy-Satellite #121: Help needed with AudioKit

rcjkierkels, 2023-01-08:
Hi,

I previously got this working with an Atom Echo M5 but because of the poor recognition I am now trying an [ESP32-AudioKit V2.2 (ES8388)](https://www.tinytronics.nl/shop/nl/development-boards/microcontroller-boards/met-wi-fi/ai-thinker-esp32-audio-kit-met-wi-fi-en-bluetooth)

Unfortunately I am not getting it to work. I successfully flashed the device using `device_type=2`. After rebooting the device (siteId: satellite2) the LED D4 and D5 is permanently ON. Is that correct?

I am using MQTT Explorer to check the MQTT traffic and see constantly audio frames coming in:
[image]

This is my satellite config:
[image]

This is what I configured in Rhasspy:

```
{
    "dialogue": {
        "satellite_site_ids": "satellite2",
        "system": "rhasspy"
    },
    "handle": {
        "satellite_site_ids": "satellite2",
        "system": "hass"
    },
    "home_assistant": {
        "access_token": "--secret--",
        "handle_type": "event",
        "url": "--secret--"
    },
    "intent": {
        "satellite_site_ids": "satellite2",
        "system": "fsticuffs"
    },
    "mqtt": {
        "enabled": "true",
        "host": "--secret--",
        "password": "--secret--",
        "username": "--secret--"
    },
    "sounds": {
        "system": "hermes"
    },
    "speech_to_text": {
        "satellite_site_ids": "satellite2",
        "system": "kaldi"
    },
    "wake": {
        "porcupine": {
            "keyword_path": "computer_raspberry-pi.ppn"
        },
        "satellite_site_ids": "satellite2",
        "system": "porcupine"
    }
}
```

I does not matter if I press KEY1 or I just try to say the wakeword. Nothing happends and the D4/D5 leds never change. They are always ON (red).

Does anybody know what could be wrong?

Romkabouter, 2023-01-08:
> After rebooting the device (siteId: satellite2) the LED D4 and D5 is permanently ON. Is that correct?

D4 and D5 are for stream and wifi, so yes :)

> I am using MQTT Explorer to check the MQTT traffic and see constantly audio frames coming in:

This is correct, the software cannot do local hotword detection, so streams the audio to Rhasspy server over MQTT

> They are always ON (red).

In general I think red is not good, but the audiostream seems to work. Is there anything in the log of Rhasspy?

rcjkierkels, 2023-01-09:
Hi, thanks for the quick response. This is in my rhasspy logs:

```
[WARNING:2023-01-09 12:13:35,406] rhasspywake_porcupine_hermes: Missing keyword dir: /share/rhasspy/profiles/nl/porcupine
[DEBUG:2023-01-09 12:13:35,425] rhasspywake_porcupine_hermes: -> Hotwords(models=[Hotword(model_id='jarvis_raspberry-pi.ppn', model_words='jarvis', model_version='', model_type='personal'), Hotword(model_id='terminator_raspberry-pi.ppn', model_words='terminator', model_version='', model_type='personal'), Hotword(model_id='hey google_raspberry-pi.ppn', model_words='hey google', model_version='', model_type='personal'), Hotword(model_id='ok google_raspberry-pi.ppn', model_words='ok google', model_version='', model_type='personal'), Hotword(model_id='americano_raspberry-pi.ppn', model_words='americano', model_version='', model_type='personal'), Hotword(model_id='alexa_raspberry-pi.ppn', model_words='alexa', model_version='', model_type='personal'), Hotword(model_id='grasshopper_raspberry-pi.ppn', model_words='grasshopper', model_version='', model_type='personal'), Hotword(model_id='computer_raspberry-pi.ppn', model_words='computer', model_version='', model_type='personal'), Hotword(model_id='pico clock_raspberry-pi.ppn', model_words='pico clock', model_version='', model_type='personal'), Hotword(model_id='blueberry_raspberry-pi.ppn', model_words='blueberry', model_version='', model_type='personal'), Hotword(model_id='picovoice_raspberry-pi.ppn', model_words='picovoice', model_version='', model_type='personal'), Hotword(model_id='bumblebee_raspberry-pi.ppn', model_words='bumblebee', model_version='', model_type='personal'), Hotword(model_id='porcupine_raspberry-pi.ppn', model_words='porcupine', model_version='', model_type='personal'), Hotword(model_id='hey siri_raspberry-pi.ppn', model_words='hey siri', model_version='', model_type='personal'), Hotword(model_id='grapefruit_raspberry-pi.ppn', model_words='grapefruit', model_version='', model_type='personal')], site_id='default', id='86bd5f0e-e215-4d89-8341-5962b28e4a6d')
[DEBUG:2023-01-09 12:13:35,425] rhasspywake_porcupine_hermes: Publishing 1798 bytes(s) to rhasspy/hotword/hotwords
[DEBUG:2023-01-09 12:13:35,461] rhasspyserver_hermes: Handling Hotwords (topic=rhasspy/hotword/hotwords, id=b2679690-e0da-4bb2-a97b-590f9695c7cc)
[DEBUG:2023-01-09 12:13:35,631] rhasspyprofile.download: speech_to_text.system pocketsphinx kaldi = False
[DEBUG:2023-01-09 12:13:35,632] rhasspyprofile.download: speech_to_text.system kaldi kaldi = True
[DEBUG:2023-01-09 12:13:35,633] rhasspyprofile.download: Skipping kaldi/base_dictionary.txt (/share/rhasspy/profiles/nl/kaldi/base_dictionary.txt)
[DEBUG:2023-01-09 12:13:35,633] rhasspyprofile.download: Skipping kaldi/g2p.fst (/share/rhasspy/profiles/nl/kaldi/g2p.fst)
[DEBUG:2023-01-09 12:13:35,634] rhasspyprofile.download: Skipping kaldi/g2p.corpus (/share/rhasspy/profiles/nl/kaldi/g2p.corpus)
[DEBUG:2023-01-09 12:13:35,635] rhasspyprofile.download: Skipping kaldi/model/conf/mfcc_hires.conf (/share/rhasspy/profiles/nl/kaldi/model/conf/mfcc_hires.conf)
[DEBUG:2023-01-09 12:13:35,636] rhasspyprofile.download: Skipping kaldi/model/conf/online_cmvn.conf (/share/rhasspy/profiles/nl/kaldi/model/conf/online_cmvn.conf)
[DEBUG:2023-01-09 12:13:35,637] rhasspyprofile.download: Skipping kaldi/model/extractor/final.dubm (/share/rhasspy/profiles/nl/kaldi/model/extractor/final.dubm)
[DEBUG:2023-01-09 12:13:35,637] rhasspyprofile.download: Skipping kaldi/model/extractor/final.ie (/share/rhasspy/profiles/nl/kaldi/model/extractor/final.ie)
[DEBUG:2023-01-09 12:13:35,638] rhasspyprofile.download: Skipping kaldi/model/extractor/final.mat (/share/rhasspy/profiles/nl/kaldi/model/extractor/final.mat)
[DEBUG:2023-01-09 12:13:35,639] rhasspyprofile.download: Skipping kaldi/model/extractor/global_cmvn.stats (/share/rhasspy/profiles/nl/kaldi/model/extractor/global_cmvn.stats)
[DEBUG:2023-01-09 12:13:35,640] rhasspyprofile.download: Skipping kaldi/model/extractor/online_cmvn.conf (/share/rhasspy/profiles/nl/kaldi/model/extractor/online_cmvn.conf)
[DEBUG:2023-01-09 12:13:35,641] rhasspyprofile.download: Skipping kaldi/model/extractor/splice_opts (/share/rhasspy/profiles/nl/kaldi/model/extractor/splice_opts)
[DEBUG:2023-01-09 12:13:35,642] rhasspyprofile.download: Skipping kaldi/model/model/cmvn_opts (/share/rhasspy/profiles/nl/kaldi/model/model/cmvn_opts)
[DEBUG:2023-01-09 12:13:35,643] rhasspyprofile.download: Skipping kaldi/model/model/final.mdl (/share/rhasspy/profiles/nl/kaldi/model/model/final.mdl)
[DEBUG:2023-01-09 12:13:35,645] rhasspyprofile.download: Skipping kaldi/model/model/frame_subsampling_factor (/share/rhasspy/profiles/nl/kaldi/model/model/frame_subsampling_factor)
[DEBUG:2023-01-09 12:13:35,646] rhasspyprofile.download: Skipping kaldi/model/model/tree (/share/rhasspy/profiles/nl/kaldi/model/model/tree)
[DEBUG:2023-01-09 12:13:35,648] rhasspyprofile.download: Skipping kaldi/model/phones/extra_questions.txt (/share/rhasspy/profiles/nl/kaldi/model/phones/extra_questions.txt)
[DEBUG:2023-01-09 12:13:35,649] rhasspyprofile.download: Skipping kaldi/model/phones/nonsilence_phones.txt (/share/rhasspy/profiles/nl/kaldi/model/phones/nonsilence_phones.txt)
[DEBUG:2023-01-09 12:13:35,649] rhasspyprofile.download: Skipping kaldi/model/phones/optional_silence.txt (/share/rhasspy/profiles/nl/kaldi/model/phones/optional_silence.txt)
[... cut ...]
[DEBUG:2023-01-09 12:15:18,905] rhasspyprofile.download: (and text_to_speech.system text_to_speech.larynx.vocoder) (larynx universal_large) ['dummy', 'universal_large'] = False
[DEBUG:2023-01-09 12:15:18,905] rhasspyprofile.download: (and text_to_speech.system text_to_speech.larynx.vocoder) (larynx vctk_medium) ['dummy', 'universal_large'] = False
[DEBUG:2023-01-09 12:15:18,905] rhasspyprofile.download: (and text_to_speech.system text_to_speech.larynx.vocoder) (larynx vctk_small) ['dummy', 'universal_large'] = False
```

Could this be the issue?
```
Missing keyword dir: /share/rhasspy/profiles/nl/porcupine
```

rcjkierkels, 2023-01-09:
I have just added my Satellite1 (Atom Echo M5) again and that one still works. I also see much more MQTT traffic now:

[image]

Still the AudioKit is doing nothing.

rcjkierkels, 2023-01-10:
I also noticed that the option 'Activation' 'Key press only' does not work. If I use that option I do notice that the satellite stops sending audioframes over MQTT. But I  would expect to se audioframes again if I press the KEY1 button. I do not see anything happen if I hold that button down or press it only once and start talking. Only 'Remote Hotword detection' seems to send audioframes.

The log which is streamed live from my Rhasspy server is also showing nothing. I only see debug messages when I am using the Atom Echo M5.

rcjkierkels, 2023-01-11:
@Romkabouter 
Taking a look at your code I do see that there is a detection algorithm for detecting which Audiokit version I am using.
[https://github.com/Romkabouter/ESP32-Rhasspy-Satellite/blob/master/PlatformIO/src/devices/AudioKit.hpp#L18](https://github.com/Romkabouter/ESP32-Rhasspy-Satellite/blob/master/PlatformIO/src/devices/AudioKit.hpp#L18)

I have a ES8388_V2. Could it be that it detected the wrong version and that this could be the reason that my KEY1 is also doing nothing?

Romkabouter, 2023-01-11:
@rcjkierkels I did not create that device nor the algorithm, but if I check the code KEY1 will always be GPIO_NUM_36:

```
#define KEY1_GPIO GPIO_NUM_36
#define KEY_LISTEN KEY1_GPIO
#define ES_KEY_LISTEN KEY1_GPIO
key_listen = is_es ? ES_KEY_LISTEN : KEY_LISTEN;
```
So key_listen will always be GPIO_NUM_36.
You can attach the device to the usb port and check the output of the serial. There should be some logging about the detected chip

rcjkierkels, 2023-01-14:
@Romkabouter I investigated the audio transmitted by the AudioKit and it makes sense that Rhasspy does not detect any wakewords because its rubbish what it transmits. I only hear some noise.  I used the code here to capture the audio:
https://github.com/Romkabouter/ESP32-Rhasspy-Satellite/blob/voco/record.py

When I test this with the atom m5 I do clearly hear me speak.

Could there be something else wrong?

rcjkierkels, 2023-01-14:
Ok there is something really weird going on. I flashed it again, did not change anything and now it works when I use KEY1. Rhasspy recognises my command. But it ONLY works using KEY1. It does never detect my wakewords? Even when I choose 'activation' = remote hotword detection it only works when I press KEY1. Any thoughts on this?

uncarvedblock78, 2023-01-19:
I had the same issue until I hacked in Automatic Level Control, see issue #113, and my PR #122

Romkabouter, 2023-01-20:
I have create a new release
https://github.com/Romkabouter/ESP32-Rhasspy-Satellite/releases/tag/v7.9.1

Please test and reopen if not fixed :)

closed 2023-01-20 by Romkabouter