Romkabouter/ESP32-Rhasspy-Satellite #123: Sending extra buttons, and direct audio wiring

Wolf220, 2023-01-24:
I've actually got 2 questions here, but i figured I'd only make one issue. First of all, is it possible to directly wire a microphone to a nodemcu, along with wiring a 3.5mm trs jack to the same for audio out (for externally powered speaker). Second of all, is it possible to run esphome and rhasspy on one board? I want to use a couple of analog pins for audio in and out for rhasspy, but also connect a couple of buttons for triggering smart home actions. (I'm fine with using something different like mqtt, i just have a basic knowledge of esphome already). If anyone has some advice or guidance, it would be greatly appreciated.

Romkabouter, 2023-01-24:
Hi there, thanks for the interest.

To answer the questions, I am not a hardware expert and therefore not qualified to answer the nodemcu question.
Also, you would have to figure out the correct pintout the run this ESP32 software.
Which brings me to you second question.

Rhasspy does not run on a ESP32 board, the software in this repo is for a Rhasspy satellite. 
This software flashes the whole chip, so it cannot run two different firmwares.

Wolf220, 2023-01-24:
Alright, thank you. I did mean rhasspy sattelite, not rhasspy, sorry for not specifying that.

sehraf, 2023-01-31:
FWIW:
Apparently you seem to be able to play sound without a DAC using ESP8266Audios `AudioOutputI2SNoDAC`
https://www.instructables.com/MQTT-Audio-Notifier-for-ESP8266-Play-MP3-TTS-RTTL/