Romkabouter/ESP32-Rhasspy-Satellite #125: ESP with INMP441 freezes all the time

DivanX10, 2023-03-01:
What could it be that the microphone freezes all the time? If you restart the ESP32, the microphone works, but not for long. In the topics, I see how the microphone reacts to sounds, and then it stops and nothing arrives in the topics
 
[image]

I drew a diagram of how I connected INMP441 and MAX98357a to ESP 32 bathroom
-----
**Option 1. INMP441 microphone + ESP32 board**

[image]
-----
**Option 2. INMP441 microphone + MAX98357a sound amplifier + ESP 32 board.**

[image]

и еще, как правильно подключить микрофон?

mikey60, 2023-05-13:
I am having the same issue.  Did you find a solution?

DivanX10, 2023-05-13:
> I am having the same issue. Did you find a solution?

I bought a ReSpeaker Core V2 that has 6 microphones, output to speakers and I forgot about this horror esp + microphone + speakers

mikey60, 2023-05-13:
Thanks for the reply.

ProrokWielki, 2023-05-14:
How long are you wires between ESP and mic (If they are too long, or run close to some high frequency source, it might mess up the mic communication)?
Could you also show logs from the esp? (data on the COM port)

DivanX10, 2023-05-14:
> How long are your wires between ESP and mic (If they are too long, or run close to some high frequency source, it might mess up the mic communication)? Could you also show logs from the esp? (data on the COM port)

I used dupont wires 20 cm long. I am currently using ReSpeaker Core V2 with Rhasspy and unfortunately I will not be able to show you the logs, only if someone interested in this project provides it to you

mikey60, 2023-05-14:
@ProrokWielki 
> How long are your wires between ESP and mic (If they are too long, or run close to some high frequency source, it might mess up the mic communication)? Could you also show logs from the esp? (data on the COM port)

I am using 100 mm wires for my breadboard.  At first the project was connected thorough a longer USB cable to a passive USB hub connected through  another USB cable to my desktop computer. With that configuration the microphone worked once and then I would have to restart the ESP32 to work again for one time.  I now have the project connected with a short USB cable to a 2.4 amp USB wall charger and the breadboard has been working reliably for several hours with out re-booting the ESP32.  Some of the wires of the bread board possibly moved some when connecting to the wall power source and eliminated an interference, but I am thinking the issue was due to a 5 volt voltage drop (I*R)  or noise at the ESP32 board when the speaker and LED were energized.  Thanks to Romkabouter and you for a great project!

Romkabouter, 2023-05-15:
Ok great find. So this issue can be closed?

mikey60, 2023-05-15:
I am okay with closing the issue, but I am not the original author.

closed 2023-11-06 by Romkabouter