Romkabouter/ESP32-Rhasspy-Satellite #126: Ota update

Dave2526, 2023-03-15:
Hello,
Whats the was to Update a Satellite via ota? Have i Set the Sattelite into an Update Mode? And where i have to Insert the IP of it? Is it the deploy Host in settings.ini or what is this variable? Thanks!

Romkabouter, 2023-03-15:
Hi there, did you read the get started doc of the device you use?

Dave2526, 2023-03-16:
Yes, the Sattelite is working with inmp441, max and fastled, i want to change the amount of the Leds, but when i try to Update it via ota There is a an Error with "cant Fond 0.0.0.0:... "

Dave2526, 2023-03-16:
ok i found it. 
I have set the hostname in the settings.ini to "rhasspy-wz" and(i think so) my Adguard block this type of hostnames. I set it to the IP address and now its uploading.

closed 2023-03-16 by Dave2526