Romkabouter/ESP32-Rhasspy-Satellite #130: Install guide missing?

carkom, 2023-05-24:
Hi @Romkabouter, is there an installation guide? I see some references to running a deploy.sh file and flashing via pio but I don't see any step-by-step process. Sorry if this is a simplistic request or if I should already know how to do this. I have all the ESP-IDF installed but not sure how to build the project so I can flash it to my ESP32. Thanks!

Romkabouter, 2023-05-24:
Hi there, you can check the Getting Started guide of the device you have.
You can find those here https://github.com/Romkabouter/ESP32-Rhasspy-Satellite#getting-started

There is no installation or setup guide for the platformIO part however, but you already have got that right?
Then you can clone this repo and then do "pio run --target upload" from the commandline

Since you have ESP-IDF installed, where are you stuck?

carkom, 2023-05-25:
I got it sorted.

Also, all those guides just assume that you have pio already installed and your USB-UART controller already configured. Would be nice to have a "wifi softap" project integrated. There are loads of them out there could easily copy and paste most of the code. It would allow people to flash the firmware without having to go through the process of building the entire project. Cheers!

At the very least it would be better if there was a more comprehensive guide to installing PIO and then building the project. Something like [this](https://lindevs.com/install-platformio-core-on-ubuntu/) would help.

It's awesome code and I'd love for it to be more accessible. Thanks for your hard work!

Romkabouter, 2023-05-26:
> Also, all those guides just assume that you have pio already installed and your USB-UART controller already configured.

Yeah, but otherwise a guide must be made for every platform (win/mac/linux) and I feel that is a bit too much here. If you are willing to write or link them, I am always open for pull requests :)

A wifi softap would be nice indeed, but I am currently a bit low on my time for this project.
Thanks for the feedback though!

closed 2023-05-26 by Romkabouter