Romkabouter/ESP32-Rhasspy-Satellite #131: Cant Compile for ESP32 Audio Kit

NobodyO815, 2023-07-07:
Hi,

I can't compile due to these errors.
Would be great to get any help.
`[{
	"resource": "/c:/Users/Me/ESP32Rhasspy/ESP32-Rhasspy-Satellite/PlatformIO/.pio/libdeps/esp32dev/FastLED/src/noise.cpp",
	"owner": "C/C++: IntelliSense",
	"code": "350",
	"severity": 8,
	"message": "more than one operator \"*\" matches these operands:",
	"source": "C/C++",
	"startLineNumber": 623,
	"startColumn": 100,
	"endLineNumber": 623,
	"endColumn": 100,
	"relatedInformation": [
		{
			"startLineNumber": 623,
			"startColumn": 100,
			"endLineNumber": 623,
			"endColumn": 100,
			"message": "function \"q<T, F, I>::operator*(uint32_t v) [with T=uint8_t, F=4, I=4]\" (declared at line 909 of \"C:\\Users\\Me\\ESP32Rhasspy\\ESP32-Rhasspy-Satellite\\PlatformIO\\.pio\\libdeps\\esp32dev\\FastLED\\src\\lib8tion.h\")",
			"resource": "/C:/Users/Me/ESP32Rhasspy/ESP32-Rhasspy-Satellite/PlatformIO/.pio/libdeps/esp32dev/FastLED/src/noise.cpp"
		},
		{
			"startLineNumber": 623,
			"startColumn": 100,
			"endLineNumber": 623,
			"endColumn": 100,
			"message": "function \"q<T, F, I>::operator*(uint16_t v) [with T=uint8_t, F=4, I=4]\" (declared at line 911 of \"C:\\Users\\Me\\ESP32Rhasspy\\ESP32-Rhasspy-Satellite\\PlatformIO\\.pio\\libdeps\\esp32dev\\FastLED\\src\\lib8tion.h\")",
			"resource": "/C:/Users/Me/ESP32Rhasspy/ESP32-Rhasspy-Satellite/PlatformIO/.pio/libdeps/esp32dev/FastLED/src/noise.cpp"
		},
		{
			"startLineNumber": 623,
			"startColumn": 100,
			"endLineNumber": 623,
			"endColumn": 100,
			"message": "function \"q<T, F, I>::operator*(int32_t v) [with T=uint8_t, F=4, I=4]\" (declared at line 913 of \"C:\\Users\\Me\\ESP32Rhasspy\\ESP32-Rhasspy-Satellite\\PlatformIO\\.pio\\libdeps\\esp32dev\\FastLED\\src\\lib8tion.h\")",
			"resource": "/C:/Users/Me/ESP32Rhasspy/ESP32-Rhasspy-Satellite/PlatformIO/.pio/libdeps/esp32dev/FastLED/src/noise.cpp"
		},
		{
			"startLineNumber": 623,
			"startColumn": 100,
			"endLineNumber": 623,
			"endColumn": 100,
			"message": "function \"q<T, F, I>::operator*(int16_t v) [with T=uint8_t, F=4, I=4]\" (declared at line 915 of \"C:\\Users\\Me\\ESP32Rhasspy\\ESP32-Rhasspy-Satellite\\PlatformIO\\.pio\\libdeps\\esp32dev\\FastLED\\src\\lib8tion.h\")",
			"resource": "/C:/Users/Me/ESP32Rhasspy/ESP32-Rhasspy-Satellite/PlatformIO/.pio/libdeps/esp32dev/FastLED/src/noise.cpp"
		},
		{
			"startLineNumber": 623,
			"startColumn": 100,
			"endLineNumber": 623,
			"endColumn": 100,
			"message": "operand types are: q44 * int",
			"resource": "/C:/Users/Me/ESP32Rhasspy/ESP32-Rhasspy-Satellite/PlatformIO/.pio/libdeps/esp32dev/FastLED/src/noise.cpp"
		}
	]
},{
	"resource": "/c:/Users/Me/ESP32Rhasspy/ESP32-Rhasspy-Satellite/PlatformIO/.pio/libdeps/esp32dev/FastLED/src/noise.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "ambiguous overload for 'operator*' (operand types are 'q44' {aka 'q<unsigned char, 4, 4>'} and 'int')",
	"startLineNumber": 623,
	"startColumn": 100,
	"endLineNumber": 623,
	"endColumn": 100
},{
	"resource": "/c:/Users/Me/ESP32Rhasspy/ESP32-Rhasspy-Satellite/PlatformIO/.pio/libdeps/esp32dev/FastLED/src/noise.cpp",
	"owner": "C/C++: IntelliSense",
	"code": "350",
[... 376 lines cut ...]

;uncomment next 4 lines and fill with your own values if you want to use a static ip address
;ip=192.168.xxx.xxx
;gateway=192.168.xxx.xxx
;subnet=255.255.255.0
;dns1=192.168.xxx.xxx
;optional: second dns server
;dns2=192.168.xxx.xxx

[OTA]
;supported: upload, ota, matrix
;-upload: device should be attached to computer via usb
;-ota: will use espota
;-matrix: matrix voice should be attached to a raspberry pi with matrix software.
;         deployhost should be set to ip of the pi
method=upload
password=save
port=3232

[MQTT]
hostname=10....
port=1883
username=RhasspyFlur
password=123123123
`

NobodyO815, 2023-07-08:
Hi, Thanks for quick reply and sorry for wrong paste.

first problem is the FastLED/src/noise.cpp 
and the declaration in \toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\12.2.0\ctime:71:11: error: 'localtime'
I hat an older toolchain-xtensa-esp32 first. Same there. I've also checking FastLED for another version.

`
.pio/libdeps/esp32dev/FastLED/src/noise.cpp: In function 'void fill_raw_2dnoise8(uint8_t*, int, int, uint8_t, q44, fract8, int, uint16_t, int, uint16_t, int, uint16_t)':
.pio/libdeps/esp32dev/FastLED/src/noise.cpp:623:100: error: ambiguous overload for 'operator*' (operand types are 'q44' {aka 'q<unsigned char, 4, 4>'} and 'int')
  623 |     fill_raw_2dnoise8(pData, width, height, octaves-1, freq44, amplitude, skip+1, x*freq44, freq44 * scalex, y*freq44, freq44 * scaley, time);
      |                                                                                             ~~~~~~ ^ ~~~~~~
      |                                                                                             |        |
      |                                                                                             |        int
      |                                                                                             q44 {aka q<unsigned char, 4, 4>}
In file included from .pio/libdeps/esp32dev/FastLED/src/pixeltypes.h:7,
                 from .pio/libdeps/esp32dev/FastLED/src/controller.h:9,
                 from .pio/libdeps/esp32dev/FastLED/src/FastLED.h:57:
.pio/libdeps/esp32dev/FastLED/src/lib8tion.h:909:14: note: candidate: 'uint32_t q<T, F, I>::operator*(uint32_t) [with T = unsigned char; int F = 4; int I = 4; uint32_t = long unsigned int]'
  909 |     uint32_t operator*(uint32_t v) { return (v*i) + ((v*f)>>F); }
      |              ^~~~~~~~
.pio/libdeps/esp32dev/FastLED/src/lib8tion.h:911:14: note: candidate: 'uint16_t q<T, F, I>::operator*(uint16_t) [with T = unsigned char; int F = 4; int I = 4; uint16_t = short unsigned int]'
  911 |     uint16_t operator*(uint16_t v) { return (v*i) + ((v*f)>>F); }
      |              ^~~~~~~~
.pio/libdeps/esp32dev/FastLED/src/lib8tion.h:913:13: note: candidate: 'int32_t q<T, F, I>::operator*(int32_t) [with T = unsigned char; int F = 4; int I = 4; int32_t = long int]'
  913 |     int32_t operator*(int32_t v) { return (v*i) + ((v*f)>>F); }
      |             ^~~~~~~~
.pio/libdeps/esp32dev/FastLED/src/lib8tion.h:915:13: note: candidate: 'int16_t q<T, F, I>::operator*(int16_t) [with T = unsigned char; int F = 4; int I = 4; int16_t = short int]'
  915 |     int16_t operator*(int16_t v) { return (v*i) + ((v*f)>>F); }
      |             ^~~~~~~~
.pio/libdeps/esp32dev/FastLED/src/noise.cpp:623:127: error: ambiguous overload for 'operator*' (operand types are 'q44' {aka 'q<unsigned char, 4, 4>'} and 'int')
...
c:\users\Me\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\12.2.0\ctime: At global scope:
c:\users\Me\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\12.2.0\ctime:64:11: error: 'clock' has not been declared in '::'
   64 |   using ::clock;
      |           ^~~~~
c:\users\Me\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\12.2.0\ctime:65:11: error: 'difftime' has not been declared in '::'
   65 |   using ::difftime;
      |           ^~~~~~~~
c:\users\Me\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\12.2.0\ctime:66:11: error: 'mktime' has not been declared in '::'
   66 |   using ::mktime;
      |           ^~~~~~
c:\users\Me\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\12.2.0\ctime:67:11: error: 'time' has not been declared in '::'
   67 |   using ::time;
      |           ^~~~
c:\users\Me\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\12.2.0\ctime:68:11: error: 'asctime' has not been declared in '::'
   68 |   using ::asctime;
      |           ^~~~~~~
c:\users\Me\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\12.2.0\ctime:69:11: error: 'ctime' has not been declared in '::'
   69 |   using ::ctime;
      |           ^~~~~
c:\users\Me\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\12.2.0\ctime:70:11: error: 'gmtime' has not been declared in '::'
   70 |   using ::gmtime;
      |           ^~~~~~
c:\users\Me\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\12.2.0\ctime:71:11: error: 'localtime' has not been declared in '::'
   71 |   using ::localtime;
      |           ^~~~~~~~~
c:\users\Me\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\12.2.0\ctime:72:11: error: 'strftime' has not been declared in '::'
   72 |   using ::strftime;
      |           ^~~~~~~~
*** [.pio\build\esp32dev\src\Satellite.cpp.o] Error 1`

Romkabouter, 2023-07-08:
Looks to my like there is some sort of version mismatch. I also do not use windows, so I want be able to help with this particular problem. Sorry about that.

Romkabouter, 2023-11-06:
@NobodyO815 did you find a way to fix this?

closed 2023-12-16 by Romkabouter